UtilitechAS/amsreader-firmware #175: Kamstrup, Norway, Invalid HDLC retured with -2 error

candycab, 2021-12-27:
Hi
I have a Pow-K (kamstrup) board.

After updating to v 2.0.2 from v1.5.7 everything works fine except that the status indicator on the web page occasionally goes from green to yellow and red.

When I log in to telnet I get this:

(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (228b):
(D) 7E A0 E2 2B  21 13 23 9A  E6 E7 00 0F  00 00 00 00
(D) 0C 07 E5 0C  1B 01 14 38  1E FF 80 00  00 02 19 0A
(D) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(D) 06 01 01 00  00 05 FF 0A  10 35 37 30  36 35 36 37
(D) 32 37 32 37  38 39 32 35  32 09 06 01  01 60 01 01
(D) FF 0A 12 36  38 34 31 31  33 31 42 4E  32 34 33 31
(D) 30 31 30 34  30 09 06 03  01 01 07 00  FF 06 00 00
(D) 0C FA 09 06  01 01 02 07  00 FF 06 00  00 00 00 09
(D) 06 01 01 03  07 00 FF 06  00 00 00 00  09 06 01 01
(D) 04 07 00 FF  06 00 00 00  5A 09 06 01  01 1F 07 00
(D) FF 06 00 00  03 40 09 06  01 01 33 07  00 FF 06 00
(D) 00 02 5D 09  06 01 01 47  07 00 FF 06  00 00 00 29
(D) 09 06 01 01  20 07 00 FF  12 00 EA 09  06 01 01 34
(D) 07 00 FF 12  00 EA 09 06  01 01 48 07  00 FF 12 00
(D) ED E0 9F 7E
(W) (readHanPort) Invalid HDLC, returned with -2

When everything works I get:

(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (228b):
(D) 7E A0 E2 2B  21 13 23 9A  E6 E7 00 0F  00 00 00 00
(D) 0C 07 E5 0C  1B 01 14 35  1E FF 80 00  00 02 19 0A
(D) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(D) 06 01 01 00  00 05 FF 0A  10 35 37 30  36 35 36 37
(D) 32 37 32 37  38 39 32 35  32 09 06 01  01 60 01 01
(D) FF 0A 12 36  38 34 31 31  33 31 42 4E  32 34 33 31
(D) 30 31 30 34  30 09 06 01  01 01 07 00  FF 06 00 00
(D) 0D 09 09 06  01 01 02 07  00 FF 06 00  00 00 00 09
(D) 06 01 01 03  07 00 FF 06  00 00 00 00  09 06 01 01
(D) 04 07 00 FF  06 00 00 00  5F 09 06 01  01 1F 07 00
(D) FF 06 00 00  03 42 09 06  01 01 33 07  00 FF 06 00
(D) 00 02 5F 09  06 01 01 47  07 00 FF 06  00 00 00 29
(D) 09 06 01 01  20 07 00 FF  12 00 EA 09  06 01 01 34
(D) 07 00 FF 12  00 EB 09 06  01 01 48 07  00 FF 12 00
(D) ED 24 55 7E
(I) (readHanPort) Valid HDLC, start at 29

**Hardware information:**
 - Meter: Kamstrup
 - AMS reader: Pow-K

**Relevant firmware information:**
 - Version: 2.2
 - MQTT: yes
 - HAN GPIO: GPIO3
 - HAN baud and parity: 2400 8N1

gskjold, 2021-12-28:
Try this firmware:
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7785581/esp8266.zip)

candycab, 2021-12-28:
Thanks but it is exactly the same.

(I) (readHanPort) Valid HDLC, start at 29
(D) (AmsDataStorage) Time is: 1640711281
(D) (AmsDataStorage) Last day update: 1640703600
(D) (AmsDataStorage) Clearing hour: 15
(D) (AmsDataStorage) Clearing hour: 16
(D) (AmsDataStorage) Last month update: -1014618496
(D) (AmsDataStorage) Clearing day: 13
(D) (AmsDataStorage) Clearing day: 14
(D) (AmsDataStorage) Clearing day: 15
(D) (AmsDataStorage) Clearing day: 16
(D) (AmsDataStorage) Clearing day: 17
(D) (AmsDataStorage) Clearing day: 18
(D) (AmsDataStorage) Clearing day: 19
(D) (AmsDataStorage) Clearing day: 20
(D) (AmsDataStorage) Clearing day: 21
(D) (AmsDataStorage) Clearing day: 22
(D) (AmsDataStorage) Clearing day: 23
(D) (AmsDataStorage) Clearing day: 24
(D) (AmsDataStorage) Clearing day: 25
(D) (AmsDataStorage) Clearing day: 26
(D) (AmsDataStorage) Clearing day: 27
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (228b):
(D) 7E A0 E2 2B  21 13 23 9A  E6 E7 00 0F  00 00 00 00
(D) 0C 07 E5 0C  1C 02 12 08  0A FF 80 00  00 02 19 0A
(D) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(D) 06 01 01 00  00 05 FF 0A  10 35 37 30  36 35 36 37
(D) 32 37 32 37  38 39 32 35  32 09 06 01  01 60 01 01
(D) FF 0A 12 36  38 34 31 31  33 31 4E 4E  32 34 33 31
(D) 30 31 30 34  30 09 06 01  01 01 07 00  FF 06 00 00
(D) 15 D6 09 06  01 01 02 07  00 FF 06 00  00 00 00 09
(D) 06 01 01 03  07 00 FF 06  00 00 00 00  09 06 01 01
(D) 04 07 00 FF  06 00 00 00  0A 09 06 01  01 1F 07 00
(D) FF 06 00 00  02 D2 09 06  01 01 33 07  00 FF 06 00
(D) 00 02 5F 09  06 01 01 47  07 00 FF 06  00 00 04 45
(D) 09 06 01 01  20 07 00 FF  12 00 EA 09  06 01 01 34
(D) 07 00 FF 12  00 EC 09 06  01 01 48 07  00 FF 12 00
(D) EB 0E D3 7E
(W) (readHanPort) Invalid HDLC, returned with -2
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 1 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...

gskjold, 2021-12-28:
-2 is checksum error, so then I guess there is some issue with the data, hard to tell why that is. Tried re-seating the module?

candycab, 2021-12-28:
I'm so embarrassed now.
Yep, it seems like that did the trick..

Thank you!!

gskjold, 2021-12-28:
Hehe no worries, good to hear it got sorted!

closed 2021-12-28 by gskjold

ArnieO, 2021-12-29:
On Pow-K the RESET button is not made available when the card is plugged in the meter. 
- Hardware reset is done by pulling out the module, press reset (or wait for the supercap to discharge) then re-insert it. 
- Software reset is done via the menu (System/restart).