UtilitechAS/amsreader-firmware #179: "New" Dark mode in Chrome

hanseinar, 2022-01-01:
The color scheme is not ideal for dark mode. The text in the gauge is hardly readable. 

Perhaps it is possible to find a color combination that handle both light and dark mode in a good way
[image]
.

gskjold, 2022-01-05:
Question, how do I enable dark mode in chrome?

hanseinar, 2022-01-05:
Paste:
chrome://flags/#enable-force-dark
in addr field and hit enter

gskjold, 2022-01-05:
I am not so sure that a forced stylesheet is something we should support at all, but I can have a look at some point if there is a simple solution to it.

gskjold, 2022-01-08:
Added transparent background on graphs helps a lot. Added for 2.0.3

closed 2022-01-08 by gskjold