UtilitechAS/amsreader-firmware #180: Timestamp from Kaifa meter is not correct

gskjold, 2022-01-02:
Probably same issue as #43. Need to investigate if Kaifa reports summertime.

ThomasEdvardsen, 2022-01-02:
Hi. 
If at any help - here is a debug dump showing the output at 14:37:15 norwegian time:

(D) (AmsDataStorage) Time is: 1641130635
(D) (AmsDataStorage) It is only 2225 seconds since last update, ignoring
(D) (readHanPort) Frame dump (41b):
(D) 7E A0 27 01  02 01 10 5A  87 E6 E7 00  0F 40 00 00
(D) 00 09 0C 07  E6 01 02 07  0E 25 12 FF  80 00 00 02
(D) 01 06 00 00  10 15 F5 FF  7E
(I) (readHanPort) Valid HDLC, start at 31
(D) (AmsDataStorage) Time is: 1641130637
(D) (AmsDataStorage) It is only 2227 seconds since last update, ignoring
(D) (readHanPort) Frame dump (123b):
(D) 7E A0 79 01  02 01 10 80  93 E6 E7 00  0F 40 00 00
(D) 00 09 0C 07  E6 01 02 07  0E 25 14 FF  80 00 00 02
(D) 0D 09 07 4B  46 4D 5F 30  30 31 09 10  36 39 37 30
(D) 36 33 31 34  30 33 32 37  35 39 32 38  09 08 4D 41
(D) 33 30 34 48  33 45 06 00  00 10 01 06  00 00 00 00
(D) 06 00 00 00  C3 06 00 00  00 00 06 00  00 2F 18 06
(D) 00 00 16 7A  06 00 00 34  64 06 00 00  09 58 06 00
(D) 00 00 00 06  00 00 09 5D  5A 4F 7E
(I) (readHanPort) Valid HDLC, start at 31
(D) (AmsDataStorage) Time is: 1641130640
(D) (AmsDataStorage) It is only 2230 seconds since last update, ignoring

gskjold, 2022-01-03:
Try this: 
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7800841/esp8266.zip)

gskjold, 2022-01-03:
Ignore that one, forgot to save before build, here we go again: 
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7800861/esp8266.zip)

ThomasEdvardsen, 2022-01-03:
Uploaded the last one, but the problem still exists.

Got this message at 11:13:06:

{
"id":"60:01:94:5E:3B:5F",
"name":"ams-reader",
"up":266,
"t":1641208386,
"vcc":3.304,
"rssi":-66,
"temp":-127.00,
"data":{
  "P":1624
}
}

ThomasEdvardsen, 2022-01-03:
Got the hourly message logged now at 13:00:10, and the rtc/meterTimestamp is correct. The timestamp in the "t" field is incorrect:

{"id":"60:01:94:5E:3B:5F","name":"ams-reader","up":675,"t":1641214810,"vcc":3.304,"rssi":-67,"temp":-127.00,"data":{"lv":"KFM_001","id":"6970631403275928","type":"MA304H3E","P":7344,"Q":199,"PO":0,"QO":0,"I1":17.16,"I2":14.12,"I3":24.54,"U1":237.40,"U2":0.00,"U3":237.60,"tPI":104513.211,"tPO":0.000,"tQI":3652.118,"tQO":3469.453,"rtc":1641211210}}

gskjold, 2022-01-03:
Added same fix for package timestamp
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7801884/esp8266.zip)

ThomasEdvardsen, 2022-01-03:
Thanks, but still the same @gskjold.

At 13:34:16:

{
  "id": "60:01:94:5E:3B:5F",
  "name": "ams-reader",
  "up": 104,
  "t": 1641216856,
  "vcc": 3.317,
  "rssi": -69,
  "temp": -127,
  "data": {
    "P": 5664
  }
}

gskjold, 2022-01-03:
Weird, can you confirm your version is 220103.6 ?

ThomasEdvardsen, 2022-01-03:
Yes. Shows up in the UI.

Just uploaded the new firmware and waited for the restart. Anything else I need to do?

gskjold, 2022-01-03:
I see why now, try this: 
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7801927/esp8266.zip)

ThomasEdvardsen, 2022-01-03:
The last one seams to fix the error!

Thanks!

closed 2022-01-03 by gskjold

ThomasEdvardsen, 2022-01-03:
After I installed this version, the ESP start rebooting at random intervals. 

Any clues for debugging? I tried logging with telnet, but the session was closed due to inactivity. Is it possible to extend the inactivity interval for telnet?

gskjold, 2022-01-04:
Doesn't look like there is an option to keep telnet alive. There is a few changes from 2.0.2 in the master branch, but I can't see any reason for random reboots. https://github.com/gskjold/AmsToMqttBridge/compare/v2.0.2...master 
I should get this version running on my test setup at home and see if I have the same problem.

gskjold, 2022-01-04:
By the way, what kind of hardware are you using? 
Serial debugging is probably the only way to catch this. Also, if you have debugger enable, try to disable it for a while to see if that helps.

ThomasEdvardsen, 2022-01-04:
Ok. I will disable debugging for now.

I am using a NodeMCU + TSS721 M-BUS module board. My setup is quite temporarily connected, so the cause may also be there. But I do not think so since all the wires are soldered. I have to 3D-print a box for the reader, so we'll see how it goes.

gskjold, 2022-01-04:
I agree it should work. If it doesn't improve without debugging, I will get my test setup going when I have time.