UtilitechAS/amsreader-firmware #181: Static IP string length too short?

dagbdagb, 2022-01-02:
**Describe the bug**
Configuring with a static IP results in wrong IP being applied

**To Reproduce**
1. Configure IP/mask/GW with 192.168.240.240/255.255.255.0/192.168.240.1
2. Observe that wireless controller sees device trying to communicate with address 192.168.240.0

**Expected behavior**
I'd expect the device to use 192.168.240.240

**Hardware information:**
 - Meter: Kaifa
 - AMS reader: Pow-U

**Relevant firmware information:**
 - Version: 2.0.2
 - MQTT: yes

**Additional context**
The same address works OK if applied as a static allocation from the DHCP server

gskjold, 2022-01-03:
Try this one: 
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7800767/esp8266.zip)

dagbdagb, 2022-01-03:
Looks like it is working with this version. Thanks.
And now it becomes very clear that separating the NTP config from the IP config is unwise. Because now you absolutely need to configure NTP manually.

gskjold, 2022-01-03:
Thanks, I will look into that

closed 2022-01-03 by gskjold