UtilitechAS/amsreader-firmware #182: One phase voltage missing?

dagbdagb, 2022-01-02:
**Describe the bug**
The web UI shows three phase currents, but only phase voltage L1 and L3.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to web UI

**Expected behavior**
I'd expect to see an equal amount of currents and voltages.
Network is 3-phase 230V (TT)

**Hardware information:**
 - Meter: Kaifa
 - AMS reader: Pow-U

**Relevant firmware information:**
 - Version: 2.0.2
 
**Additional context**
Network is 3-phase 230V (TT)

dagbdagb, 2022-01-02:
So, this appears to be a result of the exact meter model not providing this voltage directly. MA304H3E will always report 0 for the L2 phase voltage. L2 must be calculated from L1 and L3. This URL spells it out:

https://www.hjemmeautomasjon.no/forums/topic/7543-ams-han-port-hvilke-verdier-m%C3%A5ler-de-foreskjellige-str%C3%B8mm%C3%A5lerene/

IIUIC:
L2 = sqrt( (L1 - 0.5*L3)^2  + (0.866*L3)^2 ) 

If I have ever learnt this, it is most definitely forgotten now. Hence, no clue about conditions for this calculation to be true.
But results do check out as 'sane'.

gskjold, 2022-01-03:
Try this: 
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7800935/esp8266.zip)

dagbdagb, 2022-01-03:
Either my formula is wrong, or your implementation. L2 is currently reported as 463.29V.

gskjold, 2022-01-03:
I kind of forgot how cos and sin works, lets try another one: 
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7800980/esp8266.zip)

dagbdagb, 2022-01-03:
The value for L2 looks sane now. Thanks.

gskjold, 2022-01-03:
Awesome!

closed 2022-01-03 by gskjold