UtilitechAS/amsreader-firmware #183: UI suggestions

dagbdagb, 2022-01-02:
**Is your feature request related to a problem? Please describe.**
These are feature requests.
I'd suggest putting the NTP config together with the rest of the network config.
Timezone goes with the Webserver config.
Under 'Static IP', the field currently named Subnet (Ex. 192.168.100.32/27) actually expects a Netmask (Ex. 255.255.255.224).

**Describe the solution you'd like**
Under 'Static IP', add a 'NTP 1' and 'NTP 2' field. If there is  a need to permit static NTP even when DHCP is enabled, another checkbox is needed for this.
I believe there is some other work being considered for timezone handling. Being able to use 'Europe/Oslo' should reduce the need for two fields for timezone information? Unsure if timezone is used for web ui only? If so, maybe add this field under 'Web'?
Please rename Subnet to Netmask

**Additional context**
These are mostly cosmetical. Being able to verify these settings in the UI would be a nice touch. That is, a status page for DHCP config *as interpreted by the client*, NTP status (unsure if used for oneshot corrections or continuous sync?), Namserver/GW/Internet reachability and so on could turn out handy.

bardahlm, 2022-01-30:
Type of distribution system can be deduced from the model number of at least the Kamstrup meter.

I have a meter with the following model number: 6841131BN243101040
This matches 648-11-31x-Nxx-x101-040, ie 3 phase 4 wire aka 400V TN-S

gskjold, 2022-02-19:
@bardahlm Do you have any reference for decoding the model number on Kamstrup?

gskjold, 2022-11-21:
Changed alot of this in upcoming v2.2

closed 2022-11-21 by gskjold