UtilitechAS/amsreader-firmware #184: "Energy use last month" is all 0's

mikkel75, 2022-01-03:
I've been running 2.xx (now 2.0.2) since release and never seen "Energy use last month" been anything but 0's - do I have to set something up to enable this? Other graphs works just fine :)

Don't mind the ESP voltage - it's measuring the voltage of the superCap setting before a DC/DC to the ESP...

**Screenshots**
[image]

**Hardware information:**
 - Meter: Kamstrup Omnjipower
 - AMS reader: ESP8266

**Relevant firmware information:**
 - Version: 2.0.2
 - MQTT: yes
 - HAN GPIO: UART2 GPIO13

hanseinar, 2022-01-03:
Same in my installation. Last month is all 0. Online for one week.

sfk2611, 2022-01-03:
I have exactly same issue. But the strange thing is that I have two readers, one on each of my two Aidon meters.
Both readers installed at same time and running 2.02.
One reader is showing energy used last month. The other one is showing only zeros.

AMS reader: ESP8266
GPIO12
No MQTT

sfk2611, 2022-01-03:
Pictures to my comment above

[image]
[image]

gskjold, 2022-01-04:
Interesting. Which country and meter? Question goes for all of you.

mikkel75, 2022-01-04:
Denmark
Kamstrup Omnipower
ESP8266

TilSjoes, 2022-01-04:
Same here..
[image]
Norway
Aidon 6525
TN 400V

sfk2611, 2022-01-04:
Norway
Aidon 6525
ESP8266
GPIO12

ArnieO, 2022-01-04:
A Pow-U (GPIO12) user reports a similar issue. 
Norway, unknown meter type.

Could it be that the algorithm is not robust enough to recover if one midnight reading is lost?

[image]

mikkel75, 2022-01-04:
I guess it could be, but in my 'backend' Domoticz I have valid log for many days provided by the ams2mqtt...also the "enegy use last 24h" also have valid data for 00.00

servswitch, 2022-01-04:
> A Pow-U (GPIO12) user reports a similar issue. Norway, unknown meter type.
> 
> Could it be that the algorithm is not robust enough to recover if one midnight reading is lost?
> 
> [image]

Information about my meter:
Kaifa -> model: MA105H2E
RSSI: -81 dBm
Uptime: 4 days

Will move WiFi router closer to AMS reader to get better signal.

I did upgrade the firmware to version 2.0.2 a couple of days ago. I cannot tell if it was on the same day as daily energy usage stopped showing up, but it is close.

ArnieO, 2022-01-04:
> Will move WiFi router closer to AMS reader to get better signal.

I don't think -80 dBm RSSI is the issue here, ref also the [interesting input from @mikkel75](https://github.com/gskjold/AmsToMqttBridge/issues/184#issuecomment-1004749496).
We'll have to wait for @gskjold to find time to look into this!

Strange issue... My Pow-K with v2.0.2 on a Kamstrup meter has not so far (14 days) failed in this way.

madsbg, 2022-01-04:
I know it doesn't help you guys with problems, but like @ArnieO my Pow-K with v.2.0.2 in Kamstrup Omnipower / Denmark has also been very stable.
[image]

ArnieO, 2022-01-04:
@madsbg All information could help to pinpoint the problem, so thank you!
Apparently you must have started on v2.0.0, then upgraded from there:
[image]

I also note that your device has been up 15 days, some of us experience irregular reboots on v2.0.x, see https://github.com/gskjold/AmsToMqttBridge/issues/174. Mine has been up 3 days now. 

PS: Impressively low consumption! 😮

hanseinar, 2022-01-04:
Norway
Aidon 6525
ESP8266 D1 mini Lite
GPIO13
And this TTL to MBUS:  https://www.aliexpress.com/item/32834331647.html

ArnieO, 2022-01-05:
And now it is my turn.
Symptoms:
- Zero in the month-graph.
- The hourly value also seems to be stuck since 22 yesterday.
- The instantaneous power presentation works as normal.

Setup:
- Current uptime: 4 days
- Kamstrup 3-phase TN / Norway
- Pow-K (ESP8266), UART0
- v2.0.2

@gskjold: Note that the hourly values seems to be shifted 1 hour in dayplot.json:
[image]
[image]
[image]

gskjold, 2022-01-05:
Very interesting, and thank you for all the input. I think this may be connected to when and how the code does the calculations for month. I have some ideas on where to start looking. Will see if I can get to look at it this weekend.

ArnieO, 2022-01-05:
Huh, strange. Now the hourly reading started moving again:
[image]

I find it very improbable that nine subsequent identical hourly consumptions is correct.

gskjold, 2022-01-05:
Looks like the last hourly reading it managed to get was at 22:00, then missed all of then until 07:00. It will spread the use over all hours in between. I've made some changes for v2.0.3 that may fix this issue. Attaching build.

[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7813763/esp8266.zip)
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7813765/esp32.zip)

ArnieO, 2022-01-05:
Understood: If it misses a List 3 reading it will average over the missing period(s) when a new valid List 3 is read. 
But it sounds very strange that it should suddenly miss all List 3 readings for such a long period... 
Thank you for update! I will upgrade later today then we'll just have to wait and see.

sfk2611, 2022-01-05:
Thank you for a prompt solving action. 
Vennlig hilsenJon Taby

mikkel75, 2022-01-05:
I get frame errors with debug image:

(D) (readHanPort) Authentication tag:
(D) 61 AD DC BD  47 72 9F 60  97 2D E2 EE
(W) (readHanPort) Frame checksum error
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (491b):
(D) 7E A1 E9 41  03 13 C6 37  E6 E7 00 DB  08 4B 41 4D
(D) 45 01 AE 90  F6 82 01 D0  30 00 14 30  C2 0F 00 00
(D) 00 00 0C 07  E6 01 05 03  0B 29 14 FF  80 00 00 02
(D) 41 0A 0E 4B  61 6D 73 74  72 75 70 5F  56 30 30 30
(D) 31 09 06 01  01 01 08 00  FF 06 00 23  16 99 09 06
(D) 01 01 02 08  00 FF 06 00  00 00 00 09  06 01 01 03
(D) 08 00 FF 06  00 00 4C 55  09 06 01 01  04 08 00 FF
(D) 06 00 05 2D  51 09 06 01  01 00 00 01  FF 06 01 44
(D) 7F 24 09 06  01 01 01 07  00 FF 06 00  00 01 83 09
(D) 06 01 01 02  07 00 FF 06  00 00 00 00  09 06 01 01
(D) 03 07 00 FF  06 00 00 00  00 09 06 01  01 04 07 00
(D) FF 06 00 00  00 2A 09 06  00 01 01 00  00 FF 09 0C
(D) 07 E6 01 05  03 0B 29 14  FF 80 00 00  09 06 01 01
(D) 20 07 00 FF  12 00 DF 09  06 01 01 34  07 00 FF 12
(D) 00 E4 09 06  01 01 48 07  00 FF 12 00  E3 09 06 01
(D) 01 1F 07 00  FF 06 00 00  00 6C 09 06  01 01 33 07
(D) 00 FF 06 00  00 00 51 09  06 01 01 47  07 00 FF 06
(D) 00 00 00 20  09 06 01 01  15 07 00 FF  06 00 00 00
(D) D1 09 06 01  01 29 07 00  FF 06 00 00  00 A0 09 06
(D) 01 01 3D 07  00 FF 06 00  00 00 12 09  06 01 01 21
(D) 07 00 FF 12  00 63 09 06  01 01 35 07  00 FF 12 00
(D) 63 09 06 01  01 49 07 00  FF 12 00 1B  09 06 01 01
(D) 0D 07 00 FF  12 00 63 09  06 01 01 16  07 00 FF 06
(D) 00 00 00 00  09 06 01 01  2A 07 00 FF  06 00 00 00
(D) 00 09 06 01  01 3E 07 00  FF 06 00 00  00 00 09 06
(D) 01 01 16 08  00 FF 06 00  00 00 00 09  06 01 01 2A
(D) 08 00 FF 06  00 00 00 00  09 06 01 01  3E 08 00 FF
(D) 06 00 00 00  00 09 06 01  01 15 08 00  FF 06 00 0E
(D) 48 04 09 06  01 01 29 08  00 FF 06 00  0C 75 F9 09
(D) 06 01 01 3D  08 00 FF 06  00 08 58 9B  F1 EF 9D 53
(D) 55 01 5C 46  CE 38 13 52  8C 17 7E
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 AE 90 F6
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 AE 90 F6  00 14 30 C2
(D) (readHanPort) Additional authenticated data:
(D) 30 DF 15 04  2E DB 4A E4  AB A1 C6 5E  4F 69 2E B9
(D) 60
(D) (readHanPort) Authentication tag:
(D) F1 EF 9D 53  55 01 5C 46  CE 38 13 52
(W) (readHanPort) Frame checksum error
(D) (loop) Used 1 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...

AND YES - I have checked keys :)

gskjold, 2022-01-05:
You get checksum error on all frames? No change in the code up to this point where it throws checksum error, so there is no reason why that should be different from 2.0.2. Have you tried to re-seat the module? (take out and put back in)

mikkel75, 2022-01-05:
Hmm switching back to 2.0.2 (which I used before trying debug build) and it went to factury defaults - forgot all settings even wifi :(
Now - after setting correct keys - I get

(W) (readHanPort) Invalid HDLC, returned with -9

How do I specify it's a Kamstrup reader - it detected that right before?

gskjold, 2022-01-05:
Yes, reverting will clear config.
2400 baud 8N1 ?
You cannot specify meter type, it will be detected.

mikkel75, 2022-01-05:
Ah yes you are right 2.0.2 works after setting back to 8N1 - but debug image still fails with checksum error on every frame.
Have also tried software serial, same result.

mikkel75, 2022-01-05:
I have tried debug image again while double checking settings - I still get checksum errors....2.0.2 works as before...

gskjold, 2022-01-05:
ok, will look at it next time I have room to set up test with encrypted payloads

gskjold, 2022-01-05:
Hang on, I see it now :) New image attached

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7814600/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7814601/esp8266.zip)

mikkel75, 2022-01-05:
Much better - now I have valid data...Will report back wrt "Energy use..."

ArnieO, 2022-01-06:
Yesterday I saw this:
[image]

Now, with 220105.2, this:
[image]

28 is clearly wrong, and yesterday shows zero.

gskjold, 2022-01-06:
Agree, but given the nature of the algorithm I say we give it one more midnight to see if it stabilize.

mikkel75, 2022-01-06:
Data from latest beta image:
Still 0's unfortunately...

[image]

gskjold, 2022-01-06:
I've managed to reproduce this at home now, should be easier to debug.

gskjold, 2022-01-06:
I found a few problems in that code actually. Attached is a new firmware to fix it. May not produce any (or maybe incorrect) value on first midnight. There is permanent logging in the month plot code, so check serial or telnet at midnight. Debug level is ignored, so it will output on any level. Will probably generate message "Invalid future timestamp, resetting" and not produce any value over first midnight.

If you are like me and do not want to be up at midnight, but have a linux box at hand, add this to crontab:
59 23 * * * telnet 192.168.0.0 >> amslog.txt 2>&1
Remember to replace IP.

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7823649/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7823651/esp8266.zip)

mikkel75, 2022-01-07:
I didn't update to latest beta but running 220105.02 - BUT I have my first bar ever in the month status :)

[image]

servswitch, 2022-01-07:
I upgraded to 220106.1 last night and lost the few bars I had and did not get one during this first night.
[image]

ArnieO, 2022-01-07:
Upgraded to 220106.1 this morning, and unfortunately; no.
Zero on the hourly graph since this morning.
Would wipe and reflash help?

gskjold, 2022-01-07:
Seriously? Have no real recommendations for you unfortunately. I'm starting to regret this feature...

ArnieO, 2022-01-07:
It changed after next hour passing, showed averaging (10 identical hour values) over previous 10 hours.

Strange all this, I really understand the frustration!

TilSjoes, 2022-01-07:
> Seriously? Have no real recommendations for you unfortunately. I'm starting to regret this feature...

Please don't :) Your efforts are highly appreciated!!
Frode

ArnieO, 2022-01-07:
@gskjold: All the feedback in this thread is a strong indication that this is a strong and highly appreciated feature. So please don’t let the current minor bumps discourage you!

gskjold, 2022-01-08:
Thanks, appreciate the support. This has stolen way more time from me than any of the other features, so I really hope I'm able to make it stable at some point.

Got the first bar back on the graph now on my test setup, I hope all of you guys have the same positive result.

ArnieO, 2022-01-08:
Too early here for any bar on the bottom graph, as I installed 220106.1 yesterday morning.
The rest is just fine! 👍

servswitch, 2022-01-08:
I got two days/bars at the same time after two nights on 220106.1. The value listed is not correct, so I will give it a third night to see if it starts showing correct values. 
[image]

gskjold, 2022-01-08:
To be expected, It split the value over two days. It should give a correct result on next calculation

hanseinar, 2022-01-08:
Sorry to say, but no bars at my installation. There have never been bars in the month graph.
[image]

gskjold, 2022-01-08:
Just checking, have you passed at least two midnights with that build version? In any case, crossing fingers it will give you something tomorrow

hanseinar, 2022-01-08:
Yes, midnight passed two times.

sfk2611, 2022-01-08:
I am back on 2.02 on both of my readers. The first trial version did not help.

I thrust on you and install the last trial version when confirmed to work.

 

As mentioned before. Version 2.02 did correct on one of the readers in both diagrams, but the other did not sum in the month diagram. Same meter for both and GPIO12.

hanseinar, 2022-01-09:
Good morning. 
Progress! For the first time there is bars on the month graph. Strange results, bur results. Perhaps they "calibrate" in the future,
[image]

hanseinar, 2022-01-09:
Not sure if it's the right time or the right topic, but I give it a try.

Nordpool has a free API to retrieve hourly rates for the next day. Node-RED has even a node "nordpool-api-plus" which retrieves these values. You choose in the node from drop down power areas from all over Scandinavia. I am more curios for the daily and monthly cost than the consumption.
Since this feature is a bit bumpy, might it be an idea to look at it from a cost perspective rather than a consumption perspective even if they are related? Maybe we already at the limit of what esp8266/32 can handle? Perhaps instead of bars, just cost today, and cost month? Just thoughts.

gskjold, 2022-01-09:
This statement that Nordpool has a free API has come up before. Last time I checked this is not true, and the plugins out there (for node-red and alike) uses the same endpoint that nordpoolgroup.com website uses, where it is clearly states:

"Nord Pool does not permit automatic extraction of data or other usage that reduces the performance of the website. Any such extraction or usage will lead to the user being blocked from the website without further notice"
https://www.nordpoolgroup.com/About-us/Terms-and-conditions-for-use/

If this turns out not to be true, please refer me to the correct API endpoints together with terms and conditions.

However, I have implemented support for ENTSO-E API, but this is only possible with ESP32 since there is not enough available RAM with the ESP8266 to perform a HTTPS request.

[image]

hanseinar, 2022-01-09:
Ok, not aware of that. I reed the statement slightly different.  "extraction of data or other usage that _reduces the performance_ of the website". Perhaps one extraction pr. day not reduse the performance of the website, and the statement say nothing about private use, but that discussion is OT. I've asked them in a mail. Awaiting answer.

gskjold, 2022-01-09:
In my experience you should read a legal statement like this:
Nord Pool does not permit:
1. Automatic extraction of data 
or 
2. Other usage that reduces the performance of the website

In any case I refuse to implement any API which does not have public terms and conditions that states that it is free to use. 
Also, the api endpoints at nordpoolgroup.com are clearly tailored to the views at their website, which suggests that they will change the implementation without notice when they change the design of their site.

ArnieO, 2022-01-09:
I have already had dialogues with both NordPool and ENTSO-E, and confirm what @gskjold says. 

The big difference is:
- ENTSO-E is a EU site, its main mission is to inform the public.
- NordPool is an enterprise, its main mission is to generate revenue.

bardahlm, 2022-01-09:
Regarding cost calculation: This could be calculated client side instead of server side, ie by javascript that fetches the needed prices from whatever sources are available for the user. There are quite a lot of parameters that make up the price of electricity so calculating real price is complex. Capacity on the ESP should be used to deliver a steady stream of metrics to other systems that can make use of those metrics to make decisions on what to do.

TilSjoes, 2022-01-12:
Just a quick feedback. I don't know if it is relevant anymore, but for me the monthly summary seems to be working and adjusting :)
Thanks for all the help!
[image]
The last readings pr. day are also correct!
Frode

tronde-ams, 2022-01-12:
Mine is OK for energy, but has skipped energy prices for some reason. Unfortunately no active logging at the moment, but it was OK last midnight. No prices as MQTT or json either. I have two modules running 2.0.2 which shows prices.

[image]

tronde-ams, 2022-01-12:
At the momet it gives some older prices as MQTT, but "null" as json.
[image]

Logging data appears to be corrupt. I have not changed or re-started anything except the logging since yesterday where logging appeared OK.

Current log

<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...

tronde-ams, 2022-01-12:
New day, and energy prices came back. Also correct on json and MQTT.
Energy use is still correct.
No changes to log.

[image]

[2 more replies]

closed 2022-01-13 by gskjold