UtilitechAS/amsreader-firmware #185: Option to change "ESP Volt" to allow 'remote sense'

mikkel75, 2022-01-04:
**Is your feature request related to a problem? Please describe.**

I use the ESP8266 ADC to monitor the voltage over a superCap sourced from the Omnipower Vout. This voltage is ~4.4V.
When I monitor this in the GUI the "ESP" turns red as 4.4V is above recommended operating range.

**Describe the solution you'd like**
Would be nice with a more generic option to use the ADC for tracking a voltage not directly the ESP Vin.

ArnieO, 2022-01-04:
I assume then you are feeding the Kamstrup Omnipower Vout (4,15 ± 0,25 V) to a GPIO pin on your ESP8266, and select the pin by System/GIPO/Vcc. 

If I understand you correctly, you get the right voltage reading - but the "flag" in the GUI is marked "ESP" and goes red.
So your request is to add options on the System/GPIO page to change the text for this flag, and to set user-selected limits for green/yellow/red indication?

My point of view is that this would be an option very few users will need. I am also concerned that users could be encouraged to inject too high voltage (the ESP is a 3,3V device). Your setup works because there is a voltage divider on your board somewhere that divides down the 4,4V to a voltage that does not kill the ESP. 

Sorry, but I do not recommend further complexification of the firmware by adding such an option.

mikkel75, 2022-01-04:
The ESP8266 that I use only has one ADC, but yes I provide a divided (0-5.6V -> 0-1V) to the ADC pin.
Yes by using the present multiplier option, I get the right voltage - but thats not supposed to be a valid ESP 3.3V rail.

If it's only me requesting this feature I understand you point of not implementing it :)

ArnieO, 2022-01-04:
Thank you for understanding! 😄

gskjold, 2022-01-05:
I have actually looked into this before, but since we are defining ADC_MODE(ADC_VCC) for internal voltage sensing, the ADC pin cannot be used in parallell. Unfortunately one has to choose between ADC and VCC on build time, so supporting ADC input would mean a new build file. I cannot see a bigger benefit of making the effort to support both for ESP8266.

closed 2022-01-05 by gskjold

ArnieO, 2022-01-05:
But... @mikkel75 seems to be reading his external voltage with the current build? 

I find the Vcc/ADC issue is a bit confusing. In current System/GPIO you can set:
- Vcc (I assume ADC GPIO input)
- GND resistor
- Vcc resistor

I have not experimented with these settings myself, but my understanding is that @mikkel75 has used "Vcc" to select ADC input pin to read his 4,4V voltage.

Or maybe the above System/GPIO settings are not used with the current build, and @mikkel75 is in fact still reading internal Vcc but with adjusted multiplier scaling it up to 4,4V?

gskjold, 2022-01-05:
Some information on how to use ESP8266 ADC here: https://arduino-esp8266.readthedocs.io/en/latest/reference.html#analog-input

where it is clearly stated: "To read VCC voltage, use ESP.getVcc() and **ADC pin must be kept unconnected**. Additionally, the following line has to be added to the sketch: ADC_MODE(ADC_VCC);"
Because of this statement, I will not speculate on what reading you get from ADC pin if that is connected in this configuration or how valid it is.

As for GPIO configuration when using **ESP8266**:
If no pin is specified for Vcc, ESP.getVcc() is used. Resistor config is technically not applicable, but will still be applied to calculation. Vcc offset and multiplier should be used instead. 

Note that NodeMCU and D1 mini (and probably others) have a voltage divider attached to the ADC pin which taints the internal voltage reading. Configuring GND/Vcc resistor value in config will not work, since the voltage reading is not defined by this divider, but rather tainted by it since it is connected to ADC pin (when it shouldn't be in ADC_VCC mode).

If a pin is specified for Vcc, analogRead(pin) value will be used for reading voltage. This will work, but is not really accurate reading. Voltage divider config applies.

When using **ESP32**:
No internal Vcc sensor is available and Vcc pin must be specified and a voltage divider must be attached to the pin and configured in GUI. (Or rather should be, you can also use offset/multiplier instead). If pin supports ADC, adc[1,2]_get_raw(channel) will be used. If ADC is not supported by pin, analogRead(pin) is used. Voltage divider config applies in both cases.

I am open for suggestion for user experience in config when it comes to Vcc.

mikkel75, 2022-01-05:
Hmm I was not aware of this limitation in the ESP8266/ESP-12 :(
Just to understand - what is the purpose of forcing the ADC_MODE(ADC_VCC) in the code? Seems like only ESP-12 users with unconnected ADC-pin can trust the readings.
I would then VERY much like a build without this ADC_MODE(ADC_VCC) set to use the external ADC pin - as would many nodeMCU and other boards users I guess :)

ArnieO, 2022-01-05:
@mikkel75 :
If you select a pin, ADC_MODE(ADC_VCC) is **not** added to the code. The ADC reading will then depend on a resistor voltage divider outside of the ESP module (existing on NodeMCU etc), you will have enter the resistor values implemented on your board. I believe (but please double check) they are selected so that 3,3V provides 1V to the ADC pin on the ESP12. 1V is the maximum voltage for the ADC pin, as 1V will give the maximum digital value reading. This means you cannot with a NodeMCU measure higher voltages than Vcc (which is what you want to do here; measure the 4,4V). 

Please understand that this is a hardware limitation on the NodeMCU. If you work directly with an ESP12 you could adjust the resistor voltage divider so that it gives an other scaling of the digital reading vs voltage.

mikkel75, 2022-01-05:
My understanding was that ADC_MODE(ADC_VCC) is a compile-time selection and thus not possible to switch on/off in an image - or what am I missing?
I use the ESP-12 mounted on my own PCB with external dividers allowing up to 5.6V readings.

gskjold, 2022-01-05:
ADC_MODE is compile-time, cannot be switched off in code, has to be built as separate image.

ArnieO, 2022-01-05:
> My understanding was that ADC_MODE(ADC_VCC) is a compile-time selection and thus not possible to switch on/off in an image - or what am I missing?

Indeed... I think you are right and I am wrong. 
Maybe @gskjold can clarify: Will voltage reading on ESP8266 always use ESP.getVcc() (meaning that pin selection for ESP8266 serves no purpose)?

gskjold, 2022-01-05:
I guess that Vcc pin on ESP8266 has no real effect. While it does allow the code to do analogRead() from any pin on the chip, I guess the reading is either 0 or 1024 ? Needs investigating.