UtilitechAS/amsreader-firmware #190: Watchdog timer to solve sporadic hangs

kng, 2022-01-05:
**Is your feature request related to a problem? Please describe.**
After some days running the ESP32 stops responding and getting slightly warm. It is running cool normally.

**Describe the solution you'd like**
There's at least WDT for the ESP8266, it should also be possible on ESP32.

**Describe alternatives you've considered**
Not sure, it seems to be stuck either in a loop somewhere or just crashed.

**Additional context**
Probably no need to keep the reset time low, a minute or so should be ok.

gskjold, 2022-01-08:
Will be enabled in next version

closed 2022-01-08 by gskjold

gskjold, 2022-01-14:
I specifically tested firmware upgrade, and that did work for me at least.