UtilitechAS/amsreader-firmware #191: AMS reader reboots every few minutes

angainor, 2022-01-06:
This is more of a question than a bug report. I have recently installed the Pow-U in my Aidon elmeter. The problem is that Pow-U reboots every few minutes. It collects the data correctly, sends them to my RPi mqtt broker, but only lives for a few minutes. Since the reboots are very quick, I didn't notice it for a while. You can also see that the daily integrated energy usage is wrong (or 0) starting from a few days back:

[image]

The values before 3rd of January seem fine comparing to the Elvia numbers.
I also noticed that the ESP sign in the www interface is turning red quite often:

[image]

Do you have any ideas what is going wrong here? Is the voltage too low?

ThomasEdvardsen, 2022-01-06:
There is already an issue on this problem: #174

gskjold, 2022-01-06:
It goes red because of voltage. I think your WiFi is too busy, keeping the modem on the ESP in Pow-U awake and draining more power than the meter can provide. I guess @ArnieO can say something about how this have been solved in the past, but I guess the only two options are to move Pow-U to a separate WiFi with less traffic or provide the device with USB power.

angainor, 2022-01-06:
@gskjold I can certainly move it to a different network, if that is what you mean. Coud the voltage be a reason for reboots?

gskjold, 2022-01-06:
Yes, move to a different network. Voltage is definitely low enough to trigger reboot.

angainor, 2022-01-06:
@gskjold looks better, it's up now for 22 minutes. I'll monitor and let you know. 

I don't know this protocol, but would it not help to send the data over MQTT less often?

To be fair, for me it's not such a big issue right now - I send the data to Home Assistant, and the reboots are so fast that they are not noticeable. And I integrate the power usage there anyway.

gskjold, 2022-01-06:
Have looked into sending MQTT less often, but doesn't have much impact.

ArnieO, 2022-01-06:
Please see paragraph 7.1 in the Pow-U usermanual. You find it on the product page at amsleser.no.

angainor, 2022-01-07:
@ArnieO Thanks. Looks like the device works better on the new guest network, although it still reboots. Not every 2 minutes, but every few hours.

closed 2022-01-07 by angainor

ArnieO, 2022-01-07:
@angainor If you have power outlet available, and the reboots are a concern - you can avoid this by powering via USB port. The intention of the design is however to avoid that - as finding power outlet close to the meter is in many cases difficult. 

Which meter and firmware version are you on? @gskjold is currently looking into a reboot issue on v2.x.x that is unrelated to the power issue.

angainor, 2022-01-07:
@ArnieO I'm on 2.0.2, bought a Pow-U just before Christmas, that's how it came. I will follow the other thread and upgrade when it gets solved.