UtilitechAS/amsreader-firmware #194: Kamstrup meter, Denmark, incorrect Current readout

GGBMLC, 2022-01-09:
Hi 
I happy that my setup now works. But I found one mysterious with the current data. 

Step to reproduce the issue:

Note down the total power “watt” and  Phase current for L1, L2 and L3. 
I read the data from the web browser. 
Be sure not any other change of the consumption. 

I use a “about” 2000 watt heater for testing. 

In my case:
373 watt Import
L1: 1,74A
L2: 0,6 A
L3: 0,4 A

Then connect a consumption of a load of 2000 watt “10 A” on L1. 
2,3 kW Import
L1: 9,3 A
L2: 6,8 A (the current rises here ?)
L3: 0,4 A
This is total 3795 watt?  But the meter shows 2,3kW?

Then connect a consumption of a load of 2000 watt “10 A” on L2. 

Phase L2 is fine. 

2,3 kW Import2
L1: 1,74A
L2: 8,3 A
L3: 0,4 A
Total watt ok.

Then connect a consumption of a load of 2000 watt “10 A” on L3. 
2,3kw Import
L1: 1,74 A
L2: 7,02 A (the current rises here ?)
L3: 8,3 A 
This is total 3923 watt?  But the meter shows 2,3kW?

Bug: 
1:
The total power does not match the total power delivered from each Phase. 
W=(V1 X L1A)+ (V2 X L2A)+ (V3 X L3A)

2:
Some of load on one single phase (L1 pr L3) is also showed on another phase “L2”. 

Note:
Load on L2 is ok. I can also not guarantee, that the phase can be switched in my installation. "L1 and L3"

Sorry but I don't know if that’s an error in the Kamstrup sw, or in the AmstToMqttBridge sw.

I tried the cases several time to be sure about the issue. 

Hardware information:
•	Meter: Kamstrup, 3F, Denmark
•	AMS reader: Wireless module ESP8266-NodeMcu v3 CH340.
•	M-bus adapter (if applicable):

Relevant firmware information:
•	Version: v2.02
•	MQTT: [yes/no] , Yes
•	HAN GPIO: GPIO13
•	Temperature sensors, Not used/tested

ArnieO, 2022-01-09:
The short answer is that you cannot calculate 3-phase like you indicate, as the voltages and currents are phase shifted.
As far as I can see, what you experience is "as expected". 

Please google 3-phase systems to learn more.

GGBMLC, 2022-01-09:
Why is the current so measured ok on only one of the phases?
I know how the meter works,, and how the 3 phase system work (120 deg between phase).

tronde-ams, 2022-01-09:
Have you in some way configured your meter as 230V IT? Config - meter - distribution system. Denmark is as far as I know only using 400V TN. Or do you have a combination of both 400V and 230V loads active at the same time? 

Does the Danish meter have a menu showing different values? Most (all?) Norwegian meters have a user accessible menu by means of a push-button.

GGBMLC, 2022-01-09:
The meter is set to TN (400V).
The only 3 fase is the oven. All other are single phase.
Yes the meter has a menu there can be selected by a button.

tronde-ams, 2022-01-09:
Compare the values from the meter and the WEB-interface. It might give a clue about the problem. 

We Norweginans must be very aware of the differences between IT and TN, since we ar most familiar with the "wacky" IT system. Single phase 230V load in TN is a real single phase load, as opposed to "our" 230V IT-system.

gskjold, 2022-01-09:
Selection of IT or TN in the menu has no effect on values other than maximum value for the graph.

The software applies a calculation for L2 current for IT/TT meters that does not report this value on the HAN port, but it should not be applied for your meter considering you are in Denmark and all values are reported on HAN port.

Compare values from meter with values in web interface as suggested above and let me know if it does not match.

GGBMLC, 2022-01-11:
I do some more investigation.
Tried to measure the current with my Fluke meter and compare data from the AMS reader.

10A load on one phase. (230V).
The current load increases exact on each phase. But load on L1 and L3 give
[image]
s an error load extra on L2 with about 7.4A
Load on L2 there is no error.
[image]

I also check the data with Telnet. For mee is look fine.
My meter does not support the function for check current in the display.

GGBMLC, 2022-01-11:
I do some more investigation.
Tried to measure the current with my Fluke meter and compare data from the AMS reader.

10A load on one phase. (230V).
The current load increases exact on each phase. 
But load on L1 and L3 gives an error load extra on L2 with about 7.4A
Load on L2 there is no error.

I also check the data with Telnet. For mee is look fine.
My meter does not support the function for check current in the display.

[image]

gskjold, 2022-01-11:
Reading L1=5.29 L2=12.44 L3=1.16 from your screenshot. Are these the same values shown in gui?

gskjold, 2022-01-11:
Same error with v2.0.3?

GGBMLC, 2022-01-13:
HI
Version 2.0.3 
Only 2 phase ?

[image]

gskjold, 2022-01-13:
Interesting, your payload has value for all phases.. But this explains why you had incorrect value for L2 current earlier, because in 2.0.2 it would calculate 0-value, but in 2.0.3 you have to select IT/TT system in meter config to enable calculation for L2.

Cant really tell why L2 current would disappear like that. Could you enable verbose level debug and send me a few frames?

gskjold, 2022-01-14:
If you are already using GPIO13, you should have HAN pin set to UART2 which is a hardware serial pin swap from pin 3 to pin 13.

GGBMLC, 2022-01-14:
system setting to GPIO13 then AMS reader is not connected to the meter.

Error with boundry flag missing.

The data package also not end with "7E
[UART2F.txt](https://github.com/gskjold/AmsToMqttBridge/files/7871751/UART2F.txt)
"

ArnieO, 2022-01-14:
Did you try using UART2 (not GPIO13) in the setting?
[image]

GGBMLC, 2022-01-14:
Hi 
Last log is with settings UART2 (GPIO13) setting. 
But then I do not get any data from the meter.

GGBMLC, 2022-01-14:
Note. The wiki do not have any information about HW setup for the ESP8266-NodeMcu v3 CH340 board. 
Suggest to add a "whitelist" of ESP boards there works with this nice setup. 
I start with a D1 Mini but never get this board to work with my meter. 
See Issue "https://github.com/gskjold/AmsToMqttBridge/issues/173"

Then order a new ESP8266 board. 

Now I use this one:
https://www.myarduino.dk/produkt/wireless-module-esp8266-nodemcu-v3-ch340/

ArnieO, 2022-01-18:
This looks like read error, I suspect a hardware issue. 
How do you connect to the meter? 
Photo could be helpful.
Others have reported connectivity issues using Dupont-type wires. 
Could also be power issues on the NodeMCU (potential voltage instability), if I'm allowed to speculate.
Are you powering your board from the 6-pin connector in the meter? If so, the voltage regulator (in the NodeMCU?) needs to have low enough dropout to handle 4,15V.

GGBMLC, 2022-01-18:
I now tried with new Dupont wire.
Setting, Pin
GPRIO 13, D7,. no data
GPRIO 16, D0, no data
GPIO 03, RX, no data
GPIO 04, D2, no data
GPIO 13, D7,data but then missing data from one phase
GPIO UART2, D7 no data

GGBMLC, 2022-01-18:
[image]
![20220118_211419](https://user-images.githubusercontent.com/9222104
[image]
0/150011988-f211a04f-2a9d-4246-bcd5-897cfe8f1d3f.jpg)

ArnieO, 2022-01-18:
Thanks! 

1. I would suspect that power supply for this application. Try a beefier one, or a power bank.

2. Try adding large, low-ESR capacitor to 3,3V on the board - as close as possible to the ESP.

GGBMLC, 2022-01-22:
The power supply is a 5V, 2A. The board do not use 2A.  (10w). 

I assume the ESP8266 PCB/schematic include DC capacitor. 

But of course this kind of high speed electronics require stable DC supply.

ArnieO, 2022-01-24:
The photo is blurry, but... are you on the bottom two pins here?
[image]
Data_Out is on the middle top pin, see https://github.com/ArnieO/SmartMeterDocumentation#han-interface-description

GGBMLC, 2022-01-25:
[image]
connected to GND and DATA out.

ArnieO, 2022-01-26:
OK, I have read the thread and looked at your logs again.

To summarize (please correct me if this is wrong!):
- You receive **valid** data with HAN set to GPIO13 - but L2 current is missing
- You receive **invalid** data with HAN set to UART2(GPIO13)

@gskjold --> Any idea after looking at the logs (https://github.com/gskjold/AmsToMqttBridge/issues/194#issuecomment-1015523454)?

gskjold, 2022-01-28:
Not sure why UART2 would be worse than GPIO13... And L2 should not be missing on danish kamstrup. Try the attached firmware and see if there is any difference

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7960555/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7960556/esp8266.zip)

GGBMLC, 2022-01-28:
[image]
[image]
Hi
I tried with my ESP8266 board. GPIO setting: UART2 (GPIO13).
The meter show data but still the L2 is missing. 
I was not able to set my Telnet to save the logfile. Will try later.

gskjold, 2022-02-04:
I have made some adjustments that I hope will work, try the attached firmware:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8000761/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8000762/esp8266.zip)

GGBMLC, 2022-02-04:
[image]
GPIO setting : UART2(GPIO13)
Firmware 220204.1
Then I not getting any data.

ArnieO, 2022-02-04:
Check that you parity is set correctly (Configuration/Meter)

GGBMLC, 2022-02-04:
Hi. I check the meter setting. Sorry, but it was not set to 8N1.
Great news. All 3 phases is now pressent. 

[image]

gskjold, 2022-02-04:
Excellent!

closed 2022-02-04 by gskjold