UtilitechAS/amsreader-firmware #195: Missing data?

stigvi, 2022-01-11:
**Describe your problem**
I didn't received data from meter/import/active/accumulated at midnight today with the 2.0.3 version. All the other 23 hours are fine. Is it only a coincidence or have someone else experienced the same?

**Hardware information:**
 - Meter: Aidon
 - AMS reader: ESP32

**Relevant firmware information:**
 - Version: 2.0.3
 - MQTT: yes

Isaksson, 2022-01-11:
I am using version 2.0.3 for ESP8266 and I do not have that issue.
[image]

ujhede, 2022-01-12:
I am using an ESP32 and have data for all hours in the chart.

You guys are burning some serious kWh's,  does not look like EV charging - electric heating ?

Isaksson, 2022-01-12:
Nope, no heating just several computers and  electrical equipment (and an old electrical ventilation system that i will replace) 🙂

gskjold, 2022-01-13:
Sometimes there is a checksum error and data is lost, there isn't really much to do about that. How have the last couple of days been @stigvi ?

closed 2022-01-15 by gskjold