UtilitechAS/amsreader-firmware #196: warning "Unknown data format..."

MikOsle, 2022-01-12:
**Describe your problem**
I am getting the warning "Unknown data format..." and no data is displayed on the dashboard. I have received the enencryption_key and authentication_key from the supplier. The initial setup was running without problem.

*** Remote debug - over telnet
(W t:674255ms) (readHanPort) Unknown data format 7E
(D t:677488ms) (loop) Used 0 ms to update temperature
(D t:692490ms) (loop) Used 0 ms to update temperature
(W t:694197ms) (readHanPort) Unknown data format FF
(D t:707490ms) (loop) Used 0 ms to update temperature
(W t:714213ms) (readHanPort) Unknown data format 7E
(D t:722492ms) (loop) Used 0 ms to update temperature
(W t:734197ms) (readHanPort) Unknown data format FF
(D t:737493ms) (loop) Used 0 ms to update temperature
(W t:744202ms) (readHanPort) Unknown data format FF
(D t:752495ms) (loop) Used 0 ms to update temperature
(W t:764258ms) (readHanPort) Unknown data format 7E
(D t:767495ms) (loop) Used 0 ms to update temperature
(D t:782496ms) (loop) Used 0 ms to update temperature
(W t:795078ms) (readHanPort) Boundry flag missing
(W t:795085ms) (readHanPort) Unknown data format FD
(D t:797498ms) (loop) Used 1 ms to update temperature
(D t:812498ms) (loop) Used 0 ms to update temperature
(W t:824984ms) (readHanPort) Boundry flag missing

**Meter configuration:**
Baud rate: 2400
Parity: 8N1
Distribution system: TN (400V)
Main fuse: 32A
Production capacity: 0
Encryption key: 0x7C2BBCD...
Authentication key: 0x1F93912...

**Hardware information:**
 - Meter: Kamstrup OMNIPower 
 - AMS reader: D1 mini

**Relevant firmware information:**
 - Version: 2.0.3
 - MQTT: Not yet
 - HAN GPIO: GPIO5
 - Temperature sensors: non

gskjold, 2022-01-12:
Try connecting meter to GPIO 13 and switch HAN pin in System->GPIO to UART2

MikOsle, 2022-01-12:
Thanks for a quick reply. I just tried but unfortunately I am getting the same warning.

ujhede, 2022-01-12:
Are you aure the parity is correct ?  I got the same error when i had 8E1 on a Danish Kamstrup.  changing to 8N1 made everything work.

MikOsle, 2022-01-12:
Yes, the parity is configured to 8N1. I have have also tried GPIO 16. Did not do it. But thanks

ujhede, 2022-01-12:
And you have triplechecked the connection to the HAN header ?  I was using dupont wires to begin with and suspected that the connection was not reliable.

MikOsle, 2022-01-13:
When I disconnect it the warning stops. This indicates to me that when connected it receives some kind of data.

Can it be the encryption key that is not correct? or would that give a different error?

gskjold, 2022-01-13:
You would not get the same error. It does not even try to decrypt this since the format is unknown. This is either a baud/parity problem or incorrect/poor connection with your meter.

MikOsle, 2022-01-14:
I have tried all combinations of baud and parity. Did not work.
And what do you use as connector? I have tried various but does not help.

ArnieO, 2022-01-15:
A photo could be helpful.

MikOsle, 2022-01-15:
[image]

ujhede, 2022-01-16:
I felt that using those dupont wires (the ones you have in the picture) did not feel like a proper connection so I swirched to using one that loooks like this an a flat cable, which has worked well for about a week now. 
[image]

ArnieO, 2022-01-16:
There are many versions and clones of the D1 mini around. Yours looks like this one: 
[image]

This application uses the wifi quite heavily, and this D1 layout does not seem to have much capacitance on the 3,3V, so your setup relays on having both a performant voltage regulator and a power supply capable of delivering enough current to handle the peak consumption. 

I would try to add capacitance to the D1 3,3V power rail. You should use a low-ESR capacitor. Ceramic is a good choice, or a low-ESR super capacitor.

gskjold, 2022-01-28:
How did this turn out? Seen so many examples lately where dupont wires are cause for problems, so that is most likely the case.

MikOsle, 2022-01-29:
Sorry for not getting back. I ended up buying a 2x3 dupont female connector soldering the wires to the pins and to the D1 board. This works but playing to much with the plug it easily fail.
Next step is to find a dc-dc module which can supple the D1 from the Vcc pin of the meter.
[image]

gskjold, 2022-01-29:
Thanks for the feedback! Apparently very important to have proper connection with these Kamstrup meters..

closed 2022-01-29 by gskjold

ArnieO, 2022-01-31:
> Next step is to find a dc-dc module which can supply the D1 from the Vcc pin of the meter.

Any decent low-dropout voltage regulator will do nicely. 
The Vcc pin of the meter is specified to deliver 4,15±0.25V. So you need a voltage regulator with a dropout voltage lower than 4.9-3.3=**1.7V**
Maximum current draw is 75 mA.
Remember to add input/output capacitors to the voltage regulator according to its data sheet.

Tip: Check which voltage regulator is used on the D1 itself. If dropout voltage is adequate you can just connect meter Vcc to "5V" terminal of the D1.

MikOsle, 2022-02-01:
For info, I unfortunately still have issue with unknown data. It falls out quite a few times per day. Have to look for at other solution.

e-bits, 2022-02-11:
Hi @MikOsle,

I have pretty mutch the same issue and also the same setup with WEMOS D1 mini. The only difference I have powered my WEMOS D1 mini with an external 5V Power-Supply and used Input D1, which translates to GPIO5. 

I ordered now a POW-K at https://amsleser.no/ to be sure the electronics Stuff is ok.

What I realize that my HAN Icon on AMS reader web interface is changing from green, to yellow to red and also back to green again. On telnet I receive the following message, which let the HAN icon go yellow:
```
(W) (readHanPort) Boundry flag missing
(D) (readHanPort) Frame dump (1b):
(D) 13
(W) (readHanPort) Unknown data format 13
```

and then this message which let the HAN icon go red:
```
(W) (readHanPort) Frame checksum error
(D) (readHanPort) Frame dump (1b):
(D) E1
(W) (readHanPort) Unknown data format E1
```

After these failed messages and without interaction it will kind of self-heal and the HAN Icon goes green again.

So I don't know if it helps to shorthen these dupont cables or change to a different GPIO port? I have the feeling that something is completly wrong on my side because the `Energy use last 24 hours` chart is not updating properly.

[image]

Btw i checked with my grid operator who delivered my keys and I can confirm, HAN Push is activated and encryption is disabled. So no need for any keys on my side if I understood everything correctly.

Best Regards
e-bits

e-bits, 2022-02-12:
Hi @ArnieO, 
thanks for your detailed feedback. I tried different things since last night.

- directly soldered the pins (GND, DATA) to the WEMOS D1 mini, keep dupont Connectors on Meter Side
- twisted the cables for GND and DATA as you suggested
- changed to WEMOS Input D7=GPIO13
- changed Power Supply, currently powering via Micro USB

All these changes didn't help and I'm still facing the same issues. I even realized that the WEMOS D1 mini is rebooting some times, indicated by the Uptime.

This brings me to my question now. Is it possible, that the meters in Switzerland are sending some slightly different messages which can't be decode correctly by AMS Reader and let the WEMOS D1 mini crash? Would it help if I send message payloads to someone to analyse? Probably @gskjold can have a look an that?

Many thanks for your help. 

Here a photo of my current setup:
[image]

e-bits, 2022-02-13:
At the moment it is also hours on my setup, but i'm still missing the total usage from last 24 hours.
What's new at the moment, I have a read-out for my total Meter count which was different in comment https://github.com/gskjold/AmsToMqttBridge/issues/196#issuecomment-1036351310

[image]

Here is a **valid** payload:
```
(D) (readHanPort) Frame dump (228b):
(D) 7E A0 E2 2B  21 13 23 9A  E6 E7 00 0F  00 00 00 00
(D) 0C 07 E6 02  0D 07 10 1C  0A FF 80 00  00 02 19 0A
(D) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(D) 06 01 01 00  00 05 FF 0A  10 35 37 30  36 35 36 37
(D) 33 32 39 37  34 37 30 38  32 09 06 01  01 60 01 01
(D) FF 0A 12 36  38 34 31 31  33 31 42 4E  32 34 30 31
(D) 30 31 30 36  33 09 06 01  01 01 07 00  FF 06 00 00
(D) 12 15 09 06  01 01 02 07  00 FF 06 00  00 00 00 09
(D) 06 01 01 03  07 00 FF 06  00 00 00 00  09 06 01 01
(D) 04 07 00 FF  06 00 00 00  53 09 06 01  01 1F 07 00
(D) FF 06 00 00  03 87 09 06  01 01 33 07  00 FF 06 00
(D) 00 03 DC 09  06 01 01 47  07 00 FF 06  00 00 00 8F
(D) 09 06 01 01  20 07 00 FF  12 00 EA 09  06 01 01 34
(D) 07 00 FF 12  00 E9 09 06  01 01 48 07  00 FF 12 00
(D) EB 06 25 7E
(D) (readHanPort) Valid data, start at byte 29
(V) (AmsDataStorage) Time is: 1644766092
(D) (AmsDataStorage) Last day update: 1644753607
(V) (AmsDataStorage) Clearing hour: 12
(V) (AmsDataStorage) Clearing hour: 13
(V) (AmsDataStorage) Clearing hour: 14
```

Checksum Error:
```
(D) (readHanPort) Frame dump (228b):
(D) 7E A0 E2 2B  21 13 23 9A  E6 E7 00 0F  00 00 00 00
(D) 0C 07 E6 02  0D 07 10 1B  0A FF 80 00  00 02 19 0A
(D) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(D) 06 01 01 00  00 05 FF 0A  10 35 37 36  35 36 37 33
(D) 32 39 37 34  37 30 38 32  09 06 01 01  60 01 01 FF
(D) 0A 12 36 38  34 31 31 33  31 42 4E 32  34 30 31 30
(D) 31 30 36 33  09 06 01 01  01 07 00 FF  06 00 00 11
(D) 62 09 06 01  01 02 07 00  FF 06 00 00  00 00 09 06
(D) 01 01 03 07  00 FF 06 00  00 00 00 09  06 01 01 04
(D) 07 00 FF 06  00 00 00 60  09 06 01 01  1F 07 00 FF
(D) 06 00 00 03  63 09 06 01  01 33 07 00  FF 06 00 00
(D) 03 DE 09 06  01 01 47 07  00 FF 06 00  00 00 8F 09
(D) 06 01 01 20  07 00 FF 12  00 EB 09 06  01 01 34 07
(D) 00 FF 12 00  E9 09 06 01  01 48 07 00  FF 12 00 EB
(D) 03 D2 7E 7E
(W) (readHanPort) Frame checksum error
(D) (readHanPort) Frame dump (1b):
(D) 13
(W) (readHanPort) Unknown data format 13
```

Boundry flag missing:
```
(D) (readHanPort) Frame dump (228b):
(D) 7E A0 E2 2B  21 13 23 9A  E6 E7 00 0F  00 00 00 00
(D) 0C 07 E6 02  0D 07 10 1C  14 FF 80 00  00 02 19 0A
(D) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(D) 06 01 01 00  00 05 FF 0A  10 35 37 30  36 35 36 37
(D) 33 32 39 37  34 37 30 38  32 09 06 01  01 60 01 01
(D) FF 0A 12 36  38 34 31 31  33 31 42 4E  32 34 30 31
(D) 30 31 30 36  33 09 06 01  01 01 07 00  FF 06 00 00
(D) 11 C8 09 06  01 01 02 07  00 FF 06 00  00 00 00 09
(D) 06 01 01 03  07 00 FF 06  00 00 00 00  09 06 01 01
(D) 04 07 00 FF  00 00 68 09  06 01 01 1F  07 00 FF 06
(D) 00 00 03 62  09 06 01 01  33 07 00 FF  06 00 00 03
(D) DC 09 06 01  01 47 07 00  FF 06 00 00  00 91 09 06
(D) 01 01 20 07  00 FF 12 00  EA 09 06 01  01 34 07 00
(D) FF 12 00 E9  09 06 01 01  48 07 00 FF  12 00 EB FD
(D) 09 7E 7E A0
(W) (readHanPort) Boundry flag missing
(D) (readHanPort) Frame dump (1b):
(D) F3
```
Please let me know if you need additional data. I'm pretty new to this project and happy to help but I don't know if these are the correct dumps.

Many Thanks.

ArnieO, 2022-02-14:
These are the correct dumps - thank you!
You seem to have frequent errors in reading the payloads from the meter. 
If you look at the content towards the end, you find the same bytes shifted in the last two messages. 
The end byte should be 7E like the start byte. So you get checksum error end "Boundry flag missing". 
[image]
[image]
[image]

This is not the cause for reboots, it "only" leads to reject of messages, and red "HAN" flag.

This still looks like a hardware/wiring related issue to me, not related to the firmware.

MikOsle, 2022-02-14:
After I updated from 2.0.8 to 2.0.10 and twisted the wires I have had only every few error.

ArnieO, 2022-02-14:
> After I updated from 2.0.8 to 2.0.10 and twisted the wires I have had only every few error.

Thank you for sharing your experience!

e-bits, 2022-02-14:
> After I updated from 2.0.8 to 2.0.10 and twisted the wires I have had only every few error.

Hi @MikOsle,
Do you have values for the total usage in the last 24 hours? Are the bar's shown correctly?

ArnieO, 2022-02-14:
The bars use data from the message arriving at whole hour, containing total accumulated kWh for the meter (the value indicated by red rectangle by @e-bits . If you miss those messages, the bars will not appear. If you miss many messages it is unlikely that the bars will appear.

MikOsle, 2022-02-14:
Yes, I have all the information. Though, I still have issues with unknown data occasionally and involuntary reboot the moment I access the web interface.
What I forgot to mention is I have also added two capacitors to the 3,3V. 

[image]

ArnieO, 2022-02-14:
> What I forgot to mention is I have also added two capacitors to the 3,3V.

I have in several threads here recommended for such development boards to add low-ESR capacitor(s) (ceramic, tantalum or Low-ESR supercap) on 3,3V as close as possible to the ESP. 

There is no guarantee that this was the trick that improved your situation, but this application is using the Wifi quite heavily, with modem sleep in between. It is well known that the ESPs pull strong current spikes each time it reconnects to Wifi - meaning improving decoupling can only be a good thing.

Thank you for sharing!

e-bits, 2022-02-22:
@ArnieO more and more I have the feeling that missing hour values are related to a timing issue. First I had the impression it was related to `enabled` or `disabled` debugging.
In meantime I upgrade to `v2.0.11` and let it run without debugging and without MQTT support. After 12 hours I checked the hourly values and nothing showed up.
Then I enabled MQTT and ét voila, I can see hourly values since MQTT is active.

Can I ask someone to do some tests without enabling MQTT to verify if this behavior is only on my side?

Probably I need to describe what I'm trying to do. I would like to use AmsToMqttBridge and pull the /data.json API to get actual consuption data into Home-Assistant via REST Sensor.

With enabled MQTT, which delivers data into a sinkhole, it looks like this. The values shown in Home-Assistant are coming via REST. At the moment it looks like the counter during an hour is not as exact or many messages from HAN Interface are missing.
[image]

ArnieO, 2022-02-22:
Interesting analysis!
It is indeed strange that the hourly updates are read only when MQTT is active - and agree that this is a reason to suspect a timing issue. What are your thoughts, @gskjold ?

Your graph: I read it as a confirmation that many payloads are lost between the hourly accumulated value from the meter.  The algorithm adds (P * delta_t) to previous accumulated value from the meter. If many P values are lost you would get such a curve. 

I have run with and without MQTT, and never seen any loss of hourly values. 

I have not studied the live energy accumulation like you have - so I cannot comment on that. I just never see yellow or red HAN indicator - so "no" payloads seem to be lost.