UtilitechAS/amsreader-firmware #198: Unstable retrieval of energy prices

gskjold, 2022-01-13:
- Sometimes does not refresh data
- Sometimes uses data from wrong day after midnight

ArnieO, 2022-01-13:
It is probably already taken into account, but a potential source of issues is non-response or wrong response from the ENTSO-E API. I have seen that it is not quite stable.

gskjold, 2022-01-13:
Yes, I have seen that as well. I have a stand alone code to retrieve prices for my home automation, so I can run a comparison with it for a while to see if it is API or our code.

tronde-ams, 2022-01-13:
Is "verbose" the preferred logging to catch errors, or should I choose something else? I will try to log until you close this topic.

Regarding the API. 
I understand you only retrieve prices once a day. Is it a problem to retrieve them every hour? The limit for the API seems to be 400 polls per minute.

ArnieO, 2022-01-13:
I have seen periods (not often) where the API failed systematically over several days. The code should maybe have a way to signal «failed to retreive prices».

gskjold, 2022-01-13:
> Is "verbose" the preferred logging to catch errors, or should I choose something else? I will try to log until you close this topic.

Only INFO and DEBUG level messages logged from ENTSO-E code, so DEBUG is enough.

> I understand you only retrieve prices once a day. Is it a problem to retrieve them every hour? The limit for the API seems to be 400 polls per minute.

It is possible to get prices every hour, but I see no reason to do so. A lot of resources are used to establish SSL connection with remote server, which could impact other (more important) functions of the software. In fact,I've seen several times that the device reboots when establishing connection with the server.

tronde-ams, 2022-01-13:
This is what is logged as "debug".
It is the same as yesterday when the prices re-apperared at midnight. Does not seem to give any useful infrmation about changes? I deleted the old log, but it was the same information shown.

<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18

gskjold, 2022-01-13:
There will only be logging about prices when they are fetched, right after boot and at 13.00. Also some logging at midnight when prices are rotated.

tronde-ams, 2022-01-13:
Ok, Is it possible to see from the verbose log that (or why) prices disappear? I have no personal interest in logging. Only to assist in finding the bug.

gskjold, 2022-01-14:
Setting to debug and catching output from 13.00 to 13.15 and from 23.59 to 00.01 is the only way to see if it is doing what it should. I am not very hopeful though, this bug could be a logic issue that we could not catch by watching logs.

ArnieO, 2022-01-14:
Could it be an idea to avoid reading prices shortly after 13:00?

- Yes, prices should be published at 13:00 but I have seen days with delays, so in an automation I implemented some years back I decided to read ENTSO-E prices later in the evening.
- Traffic on the ENTSO-E server is probably higher shortly after prices are published. Could be an argument for waiting. 
- Maybe even consider randomizing the time of reading to avoid that (in a not too distant future! 😉 ) thousands of AmsToMqttBridge devices poll the server at exactly the same time.

gskjold, 2022-01-16:
I think these are all very good points! Implementing random time between 13:30 and 14:00 for now, we can move and/or expand this in future releases when number of devices increase.

closed 2022-01-28 by gskjold