UtilitechAS/amsreader-firmware #199: Still bug with the Pow-U

sfk2611, 2022-01-13:
I had the problem that there was no values in the "Energy use last Month"
I tried the intermediate firmware versions (6/1), but no help. i have two readres, one to each of mine two meters.
One of the readers was working very slow and reported the 24 h use about three days late and the HAN mark was red most of the time, for both. The caracteristic red confirmation blink was not observed.

Then I installed version 2.03
Then one of the phase current readings disappeared; On both readers.

There is still no red acknowledge blink. The only red thing is the HAN.

I performed a factory reset on one of the readers, but still missing phase current L2 and the other troubles.

[image]
This is the image of the one without a factory reset. See the time and the lag for the statistic.

[image]
This is the image of the reset one. This has obviously been on to short for any statistics, but still the "HAN" is red and L2 missing.

Hardware information (both meters and all settings are identical:
 - Meter: [e.g. Aidon 6525
 - AMS reader: ESP8266  (Pow-U ?)

Relevant firmware information
 - Version: 2.03
 - MQTT: no
 - HAN GPIO: GPIO12]
 - HAN baud and parity: 2400 8E1

gskjold, 2022-01-13:
IT/TT system maybe? If so, it must be set in Meter config to have L2 calculated.
As for the graphs and the red HAN indicator, it looks like you have a lot of lost frames. (Judging from lots of distributed values in the day graph)
Are you able to use debugging? If so, set debug level and telnet to the device and show me what it looks like. I would assume a lot of errors.

sfk2611, 2022-01-13:
It system yes.
At least L2 was present under version 2.02.
Which debug level you want?
I am have started the debugger in command window, but how to get the result on file?

gskjold, 2022-01-14:
L2 current is not present in data from IT/TT meters using Norwegian data format. In v2.0.0 -> v2.0.2 it was automatically calculated regardless of distribution system. In v2.0.3 distribution system have to be set to IT/TT to enable calculation of L2.

"Debug" level is sufficient for now. I'm assuming you are on windows? I do not have a windows machine in front of me, but if I remember correctly you can just click and drag to select text inside cmd. Then hit enter to copy.

In any case, what we are looking for is that you get a "Valid data, start at byte 18" every 2.5s. If you get errors instead or the data is not received every 2.5s, there is some kind of communication issue.

gskjold, 2022-01-14:
CTRL+M to mark text in cmd I am told now. And hit enter to copy

sfk2611, 2022-01-14:
I get valid data on byte 18.

 

Vennlig hilsenJon Taby

sfk2611, 2022-01-14:
I will be ready to send you something within an hour. 

IT system: sorry, forgot to fill in again after update. 

When it is IT, the voltage is Line Voltage and should be given as L1-L2, L1-L3, L2-L3.

Vennlig hilsenJon Taby

sfk2611, 2022-01-14:
The problem may now have been solved. It is obvious of importance to have the system right.
But the red HAn may be a concern. 

Debug output from green HAN (one of the readers):
[image]

[image]

And here is thesecond one with red HAN. You see an unknown data format in first line and a frame checksum error 6 line below.
[image]

[image]

gskjold, 2022-01-14:
It looks like you have a small amount of data corruption on the second unit. Could be related to cable length or quality maybe? Not sure.. It looks like you have data every 2.5s though, so HAN indicator should not have been red, not sure what is going on there to be honest..

sfk2611, 2022-01-14:
The data cable is the 200mm flat cable following the reader. 

HAN is green some times. Strange. 

Is there any reason to take a factory reset to flush registers? 

I will try and give you status within two days. 

Sorry for the IT bother. 

Vennlig hilsenJon Taby

ArnieO, 2022-01-14:
What you experience (sporadic HAN read error) could be a hardware marginality on the Pow-U when used on your meter.

A comparator is used to detect "0" vs "1". The input level to the comparator is set by a resistor voltage divider. On some meters the maximum voltage for '0' is quite a bit higher than the nominal 12V. On others the minimum for '1' is quite a bit lower than the nominal 24V. This makes it challenging to set the right values for the voltage divider. If setting is marginal you can experience the kind of issue you now see. The issue is rare, you are the first (of around 100 Pow-U users) on a non-Kamstrup meter. Kamstrup is more prone to variations in output voltages due to the design with a plug-in HAN adapter (which is why I recommend Pow-K for Kamstrup meters, it receives directly a 0-3.3V signal from the meter).

This is an identified improvement point on the Pow-U design: a more robust level detection circuitry.

If you have access to an oscilloscope, I could guide you on how to measure the board, please contact me on post@amsleser.no. If a scope measurement confirms the issue on your board/meter, you could return it to me for resistor adjustment based on your measurements. Unfortunately difficult to do this without measuring on your meter.

sfk2611, 2022-01-14:
It is the same issue on both readers. 

It is also the case that the ESP voltage varies significantly. Adjusting the multipliers to fine, and then next hour it is out of range. 

Unfortunately, I have no oscilloscope. May I use a FLUKE 289?

Vennlig hilsenJon Taby

ArnieO, 2022-01-14:
> Unfortunately, I have no oscilloscope. May I use a FLUKE 289?

No, sorry - that will not work. We need to see the pulses on the input of the comparator.

Aha... the voltage varies - that is a clue. Could you please read paragraph 7.1 of the user manual and see if the phenomena (and proposed solution) described there helps you? If this is the issue, you will have frequent restarts (see "uptime"). You can test this by powering over micro USB plug.

Adjusting multiplier only scales the displayed value for Vcc. If you want to adjust it you can use your Fluke to measure Vcc while powering the board over micro USB, then adjust multiplier until displayed value == measured value. I do an approximate adjustment as part of testing before shipping the boards.

sfk2611, 2022-01-14:
I will try to feed them by the USB.

If OK after 48 h, I will report back. 
Vennlig hilsenJon Taby

sfk2611, 2022-01-18:
One meeter fed by USB. The feeding didn't change much. HAN is mainy red, but at some occations it is green.
The accumulated energy calculations are not correct.
Can you indicate which resistor(s) to alter and approximately how much? It should be either up or down. I can try both and find out which is correct.

[image]

Here is the other one: HAN is mainly green, but changes to red sometimes.
The energy accummulation may be close to correct.

Any how, the "Phase voltage" should be changed to "Line voltage" for IT systems. 

[image]

gskjold, 2022-01-21:
I have a fix here for labels and possibly month graph values:

[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7915559/esp8266.zip)

sfk2611, 2022-01-21:
Good! 

I appreciate your excellent service. 

I will give it a try. 

I have recognised the change to line voltage. The given current is also OK, but we are normally better off with the current in the phase wires? A bit more troublesome effect calculation though. 

Vennlig hilsenJon Taby

sfk2611, 2022-01-22:
Labels is now ok.
But the HAN is still getting red, with or without external power.
Since HAN is red, the accumulated energy will not necessary be correct.

The strange thing is that HAN is getting green when I start debugging. After stopping debugging it goes back to red after a while.

Earlier, there was a red blink acknowledging the read. This red blink is no more present. Is that an indication?

sfk2611, 2022-01-22:
The accumulation issues seems to work ok now for both Pow-Ups, except that data may be corrupted by the red HAN.

sfk2611, 2022-01-22:
Hmm. A value recordet at 14 o clock on one of the readers suddenly disappeared. Strange.

gskjold, 2022-02-04:
@sfk2611 Do you still have that issue where HAN turns red?

sfk2611, 2022-02-04:
Yes. HAN shifts from green to red more or less randomly on both units. One of the units are not able to sum pr hour or pr day. The other seems to sum up better. 
Vennlig hilsenJon Taby

sfk2611, 2022-02-04:
I use the unofficial version you sent. 
Vennlig hilsenJon Taby

ArnieO, 2022-02-07:
(Sorry, my mistake - the third one was another J. Taby; not you!)

From your postal address I assume you are on an Aidon meter - which should be fine (some Pow-U units have difficulties on Kamstrup meters, as the signal voltage level on some of those are slightly different). 

I will be happy to replace your boards - but would appreciate if you could return yours to me for diagnosis - which could potentially help me understand where this assumed marginality lies. 
Please email me at post@amsleser.no to arrange!