UtilitechAS/amsreader-firmware #201: Voltage indication IT vs TN

ArnieO, 2022-01-14:
**Is your feature request related to a problem? Please describe.**
In TN systems, line voltage is phase vs neutral, so current indication is correct: voltages refer to L1, L2, L3. 

In IT/TT systems, voltages are measured between phases, as mentioned in comment here: https://github.com/gskjold/AmsToMqttBridge/issues/199#issuecomment-1012866481

**Describe the solution you'd like**
For IT/TT systems, change the labels on the graph to indicate the voltages are between phases:
L1-L2, L2-L3, L3-L1
[image]

gskjold, 2022-01-14:
Any change to graph title? ( line voltage vs phase voltage )
I'm not sure of the correct terminology on stuff like this..

ArnieO, 2022-01-14:
I would leave the title as is, unless [@sfk2611 ](https://github.com/sfk2611) has a strong opinion. 😄 

I am not sure which label belongs to which data from the meter. Maybe identifiable in the documentation for one of the meters but don't have time to dig right now - can do later.

Nothing urgent, this is just a possibility to do a minor "beautification" of the code. 😉

tronde-ams, 2022-01-14:
It seems like all three manufacturers agree upon the L1-L2, L1-L3, L2-L3 sequence for U1, U2 and U3. Note the difference from what ArnieO wrote.

"Line voltage" is more correct than "Phase voltage.  And "Line current" is more correct than "Phase current" in IT/TT. If you change one, you should change both.

Another way to title the voltage readings is "Voltage between". This is technically correct (aka what you will read with a voltmeter), and probably more descriptive to most people.

closed 2022-01-16 by gskjold

tronde-ams, 2022-01-16:
I know this is clised, but I tryy.

You got it right in 2.0.5 regarding the voltage, but wrong with the currents for IT/TT.

Label for currents should read "Line current", and L1, L2 and L3 should be used just like before.

gskjold, 2022-01-16:
Got it