UtilitechAS/amsreader-firmware #203: Nothing happen when using in plce upgrade

ujhede, 2022-01-16:
Nothing happen when uploading new firmware.

I have tried both uploading from the Firmware menu and choosing the "upgrade" link on the front page.  I am just left on the same version 2.0.3.

**Hardware information:**
 - Meter: Kamstrup Danish
 - AMS reader: ESP32 D1 clone
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.0.3
 - MQTT: yes
 - HAN GPIO: 16
 - Temperature sensors none.

gskjold, 2022-01-16:
Yes, 2.0.3 has known OTA upgrade issues with ESP32. Seems like v2.0.4 is also suffering from same issue

gskjold, 2022-01-16:
Final issue with OTA update on ESP32 will be fixed in v2.0.5. This version needs to be flashed by cable.

closed 2022-01-16 by gskjold