UtilitechAS/amsreader-firmware #204: ESP32 AP button

tronde-ams, 2022-01-16:
AP button is default GPIO 0, at least for generic ESP32, while FAQ says GPIO 4. 

GPIO 0 is not routed to a pin on some development boards making it more difficult to use. I suggest to use GPIO 4 as default, or update wiki.

gskjold, 2022-01-16:
Thanks. Removing AP pin config from generic ESP32 to avoid unwanted side effects. If user actually has button on a pin, it should be configured manually.

closed 2022-01-16 by gskjold

tronde-ams, 2022-01-16:
I don't think it should be configured manually, because it will probably only be needed if something goes wrong (eg. WiFi configuration) during first set up. If it has to be configured, it will not be available when actually needed most.

I see nothing wrong if it is defaulted to one specific pin, but it should me made clear which pin is used.

I have flashed several ESPs for friends, but they have no way to re-flash if something goes wrong, so an AP button makes sense.

gskjold, 2022-01-16:
Considering that different development boards have different layout, having default AP pin can cause problems. The generic profile should be able to boot on any board. It was not intended to have a default for generic ESP32 boards in the first place. 

When you flash boards for friends, preconfigure GPIO with AP pin 4 and factory reset before you ship. GPIO is not cleared on factory reset.

I can also add a profile for the board you are using. Note thtat specified AP pin for the profile must also work unconnected.

tronde-ams, 2022-01-16:
I am fine with the solution you suggest. No need to have a lot of profiles for unknown Chinese hardware.

Maybe write somthing about this in the wiki?

gskjold, 2022-01-16:
Will do!