UtilitechAS/amsreader-firmware #205: Energy prices seems to be EUR only

tronde-ams, 2022-01-16:
2.0.5
Energy prices seems to be EUR only, regardless if choosen currency.

Same for WEB, json and MQTT.

gskjold, 2022-01-16:
Working fine on my device. Swapping between currencies updates price. NOK prices are matching price in my home automation.

tronde-ams, 2022-01-16:
I have two old and this freshly flashed unit. All of them seems to show EUR. No changes when I change currency.

I wil leave them alone, and report back tomorrow when new prices are fetched.

[image]
[image]

[image]
[image]

tronde-ams, 2022-01-16:
There is something buggy here.

I have log on one unit. At midnight it refused to connect to Entso-e.

I have deleted my token in this log, but it vas the exact same as I have used before.

```
> <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
> <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
> <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201162300&p<ESC>[0meriodEnd=202201172300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
> <ESC>[0m(D) (EntsoeApi)Connection established
> <ESC>[0m(E) (EntsoeApi)Communication error: 
> <ESC>[0m(E) (EntsoeApi)connection refused
> <ESC>[0m(D) (EntsoeApi)
> <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
> 

```

Then I deleted my token in the adapter, saved and inserted my token again and saved. Then the conection was accepted, and some different prices appeared. They are different from what is shown on the Entso-e web.

**More log**

```
<ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
<ESC>[0m(I) (EntsoeApi) Setting midnight millis 86204003
<ESC>[0m(I) (EntsoeApi) Fetching prices for today
<ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201152300&p<ESC>[0meriodEnd=202201162300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
<ESC>[0m(D) (EntsoeApi)Connection established
<ESC>[0m(D) (EntsoeApi)Receiving data
<ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
<ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
<ESC>[0m
assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace:0x40083bd9:0x3ffb1d900x4008d129:0x3ffb1db0 0x40092685:0x3ffb1dd0 0x400922cb:0x3ffb1f00 0x4008408d:0x3ffb1f20 0x4008606d:0x3ffb1f40 0x4014dc6e:0x3ffb1f60 0x40105de6:0x3ffb1f80 0x400f7b37:0x3ffb1fa0 0x400f6d49:0x3ffb1fc0 0x400f8155:0x3ffb1fe0 0x400f838d:0x3ffb2000 0x400f8491:0x3ffb2060 0x400dbd73:0x3ffb20b0 0x400dc357:0x3ffb20f0 0x400dc546:0x3ffb2270 0x400debc1:0x3ffb22b0 0x400d820e:0x3ffb2410 0x4010013d:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57
```

**Then something wrong at norges-bank.no**

```
<ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (EntsoeApi) Setting midnight millis 86192009
<ESC>[0m(I) (EntsoeApi) Fetching prices for today
<ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"a&documentType=A44&periodStart=202201152300&p<ESC>[0meriodEnd=202201162300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
<ESC>[0m(D) (EntsoeApi)Connection established
<ESC>[0m(D) (EntsoeApi)Receiving data
<ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
<ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
<ESC>[0m(D) (EntsoeApi)Connection established
<ESC>[0m(E) (EntsoeApi)Communication error: 
<ESC>[0m(E) (EntsoeApi)connection refused
<ESC>[0m(D) (EntsoeApi)
<ESC>[0m(D) (EntsoeApi) Resulting currency multiplier: 1.0000
<ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
```

Prices shown

[image]

Does not agree

[image]

[image]

Prices from the module i flashed fresh yesterday is different. Same NTP setting on that unit. I have not changed anything on this unit.
[image]

hanseinar, 2022-01-17:
I had to multiply by 10, then the amounts show correctly in NOK
[image]
[image]

tronde-ams, 2022-01-17:
I logged until now, but no new prices and still connection refused at both Ensto-e and Norges Bank.

Then I flashed one of my units back to 2.0.4, and that unit got all new prices correctly in EUR, but not in NOK. Unfortunately no screenshot of the first prices in NOK.

The log says nothing about norges-bank.no.

```
(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (EntsoeApi) Setting midnight millis 33405006
<ESC>[0m(I) (EntsoeApi) Fetching prices for today
<ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201162300&p<ESC>[0meriodEnd=202201172300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
<ESC>[0m(D) (EntsoeApi)Connection established
<ESC>[0m(D) (EntsoeApi)Receiving data
<ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
<ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201172300&p<ESC>[0meriodEnd=202201182300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
<ESC>[0m(D) (EntsoeApi)Connection established
<ESC>[0m(D) (EntsoeApi)Receiving data
<ESC>[0m(D) (EntsoeApi)Connection established
<ESC>[0m(D) (EntsoeApi)Receiving data
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(D) (AmsWebServer)Serving /energyprice.json over http...
<ESC>[0m(W) (readHanPort)(C1) Unknown data format 00
<ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
<ESC>[0m(D) (AmsWebServer)Serving /github.svg over http...
<ESC>[0m(D) (AmsWebServer)Serving /application.js over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(D) (AmsWebServer)Serving /dayplot.json over http...
<ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
<ESC>[0m(D) (AmsWebServer)Serving /energyprice.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
<ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
<ESC>[0m(I) (readHanPort)(C1) Valid data, start at byte 18
```

The graphics is EUR x 10 to compare with graphics from ensto-e which shows the same. 
[image]

Then I switched back to NOK x 1, and it seems like the values are correct now, but it can also be because the exchange rate is close to 10 at the moment.

Regardless of currency conversion, it seems to be a bug in 2.0.5 when all three units was unable to fetch new prices, and 2.0.4 still does.

ujhede, 2022-01-19:
I am also struggeling a bit with the price feed in 2.0.5.  Since prices are only fetched once a day would it be an idea to add the result from the last fetch (prices and rates)  somwhere in the GUI.  Getting serial debug output at the exact time the fetch happens is a little difficult.

gskjold, 2022-01-21:
Have made some changes I think will work, try this firmware:

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7915524/esp32.zip)

tronde-ams, 2022-01-21:
Testing with log on one unit now.  It started with "error" as debug level.  Should I use "error", or "debug"?

tronde-ams, 2022-01-21:
v 220121.3
Energy prices disappered from WEB at midnight. Was OK before. 
/data.json says p = null

I keep on logging.

```
(23:59:30.971) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:32.358) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(23:59:33.665) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:34.860) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:37.367) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(23:59:39.732) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:40.849) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:42.332) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:44.838) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(23:59:44.877) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(23:59:45.669) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:47.336) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:50.847) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(23:59:51.215) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:52.373) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:54.859) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(23:59:56.537) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:57.338) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:00.847) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(00:00:00.925) <ESC>[0m(I) (EntsoeApi) Rotating price objects at 1642806002
(00:00:01.495) <ESC>[0m(I) (EntsoeApi) Setting midnight millis 101986052
(00:00:01.495) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(00:00:02.078) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:02.125) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:02.330) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:04.860) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:07.399) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:07.434) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:11.338) <ESC>[0m(D) (AmsDataStorage) Last day update: 1642802413
(00:00:11.404) <ESC>[0m(D) (AmsDataStorage) Last month update: 1642719609
(00:00:11.404) <ESC>[0m(I) (AmsDataStorage) Usage for hour 22: 2655
(00:00:11.433) <ESC>[0m(I) (AmsDataStorage) Usage for day 22: 61359
(00:00:11.433) <ESC>[0m(I) (readHanPort)(C1) Saving day plot
(00:00:11.433) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:12.366) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:12.726) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:14.856) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:17.368) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(00:00:17.368) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:18.013) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
(00:00:18.099) <ESC>[0m(D) (AmsWebServer)Serving /application.js over http...
(00:00:19.655) <ESC>[0m(D) (AmsWebServer)Serving /github.svg over http...
(00:00:20.291) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:20.885) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:22.374) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:22.501) <ESC>[0m(D) (AmsWebServer)Serving /dayplot.json over http...
(00:00:22.532) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(00:00:22.626) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:24.830) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:27.336) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:28.043) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:30.846) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
(00:00:31.985) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:32.373) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(00:00:32.373) <ESC>[0m(D) (AmsWebServer)Serving /application.js over http...
(00:00:32.373) <ESC>[0m(D) (AmsWebServer)Serving /github.svg over http...
(00:00:32.471) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(00:00:32.903) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:32.941) <ESC>[0m(D) (AmsWebServer)Serving /dayplot.json over http...
(00:00:32.995) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:34.829) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:37.365) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:38.277) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
[... 14 lines cut ...]
(00:00:56.395) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(00:00:56.433) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:57.331) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:01:00.895) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:01:02.157) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:01:02.368) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:01:04.865) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:01:07.358) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(00:01:07.358) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
[... cut ...]
(00:01:22.330) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:01:23.773) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:01:24.863) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(00:01:24.863) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:01:27.357) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:01:29.282) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:01:30.884) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
```

tronde-ams, 2022-01-22:
Prices re-appeared sometime between 00:30 and 02:15. 
Can not see anything in the log about this. data.json is also OK by now.

gskjold, 2022-01-22:
Thanks, it was gone between 00:00 and 01:00, I see the problem now.

tronde-ams, 2022-01-23:
Upgraded from v 220121.3 to 2.0.0 just before 19:00

1: Impossible to use the "upgrade" function in WEB interface. OK to download file and upgrade via system -> firmware.

2: Prices disappeared, and log says 

```

(18:55:41.597) <ESC>[0m(I) (MQTT_connect)(C1) Successfully connected to MQTT!
(18:55:41.878) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
(18:55:43.219) <ESC>[0m(D) (AmsWebServer)Serving /application.js over http...
(18:55:43.219) <ESC>[0m(D) (AmsWebServer)Serving /application.js over http...
(18:55:43.750) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:55:44.483) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(18:55:44.967) <ESC>[0m(D) (AmsWebServer)Serving /dayplot.json over http...
(18:55:44.998) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(18:55:45.029) <ESC>[0m(I) (EntsoeApi) Setting midnight millis 18264000
(18:55:46.776) <ESC>[0m(I) (EntsoeApi) Fetching prices for today
(18:55:46.792) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201222300&p<ESC>[0meriodEnd=202201232300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(18:55:46.823) <ESC>[0m(D) (EntsoeApi)Connection established
(18:55:46.823) <ESC>[0m(D) (EntsoeApi)Receiving data
(18:55:48.430) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(18:55:48.461) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(18:55:48.477) <ESC>[0m(D) (EntsoeApi)Connection established
(18:55:48.477) <ESC>[0m(E) (EntsoeApi)Communication error: 
(18:55:48.664) <ESC>[0m(E) (EntsoeApi)connection refused
(18:55:48.679) <ESC>[0m(D) (EntsoeApi)
(18:55:48.679) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:55:48.726) <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
(18:55:48.757) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201232300&p<ESC>[0meriodEnd=202201242300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(18:55:48.773) <ESC>[0m(D) (EntsoeApi)Connection established
(18:55:48.773) <ESC>[0m(E) (EntsoeApi)Communication error: 
(18:55:48.820) <ESC>[0m(E) (EntsoeApi)connection refused
(18:55:48.820) <ESC>[0m(D) (EntsoeApi)
(18:55:48.820) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(18:55:50.442) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(18:55:50.442) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(18:55:50.458) <ESC>[0m(D) (EntsoeApi)Connection established
(18:55:50.458) <ESC>[0m(E) (EntsoeApi)Communication error: 
(18:55:50.489) <ESC>[0m(E) (EntsoeApi)connection refused
(18:55:50.505) <ESC>[0m(D) (EntsoeApi)
(18:55:50.505) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:55:50.567) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:55:51.987) <ESC>[0m(D) (loop)(C1) Used 867 ms to update temperature
(18:55:52.876) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:55:54.483) <ESC>[0m(D) (AmsWebServer)Serving /entsoe.html over http...
(18:55:55.153) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:55:56.994) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
(18:55:58.429) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
```

No prices on the oter unit I do not log either.

gskjold, 2022-01-23:
There must be something with your network that causes bad connection. Slow DNS, slow internet or firewall issues. What can you tell me about the internet connection?

tronde-ams, 2022-01-23:
No other problems. 100 Mb fiber. My 2.0.4 shows prices correctly. Only seen this problem with 2.0.5. and 2.0.6.
220121.3 was OK except 00:00 - 01:00.

Seems to have enterd a loop. 
first for norges-bank

```
(19:20:28.437) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(19:20:29.326) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(19:20:29.326) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(19:20:29.341) <ESC>[0m(D) (EntsoeApi)Connection established
(19:20:29.341) <ESC>[0m(E) (EntsoeApi)Communication error: 
(19:20:29.373) <ESC>[0m(E) (EntsoeApi)connection refused
(19:20:29.373) <ESC>[0m(D) (EntsoeApi)
(19:20:29.373) <ESC>[0m(D) (AmsWebServer)Serving /dayplot.json over http...
(19:20:29.419) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(19:20:29.575) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:30.184) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:31.666) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:34.162) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(19:20:35.082) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(19:20:35.082) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(19:20:35.082) <ESC>[0m(D) (EntsoeApi)Connection established
(19:20:35.098) <ESC>[0m(E) (EntsoeApi)Communication error: 
(19:20:35.113) <ESC>[0m(E) (EntsoeApi)connection refused
(19:20:35.129) <ESC>[0m(D) (EntsoeApi)
(19:20:35.129) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:36.658) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(19:20:36.673) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:40.168) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(19:20:40.589) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(19:20:40.589) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(19:20:40.605) <ESC>[0m(D) (EntsoeApi)Connection established
(19:20:40.605) <ESC>[0m(E) (EntsoeApi)Communication error: 
(19:20:40.651) <ESC>[0m(E) (EntsoeApi)connection refused
(19:20:40.651) <ESC>[0m(D) (EntsoeApi)
(19:20:40.651) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:41.650) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:44.161) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(19:20:46.221) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(19:20:46.221) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(19:20:46.236) <ESC>[0m(D) (EntsoeApi)Connection established
(19:20:46.236) <ESC>[0m(E) (EntsoeApi)Communication error: 
(19:20:46.283) <ESC>[0m(E) (EntsoeApi)connection refused
(19:20:46.283) <ESC>[0m(D) (EntsoeApi)
(19:20:46.283) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:46.657) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:50.183) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:20:51.665) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(19:20:51.681) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(19:20:51.696) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(19:20:51.712) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(19:20:51.712) <ESC>[0m(D) (EntsoeApi)Connection established
(19:20:51.712) <ESC>[0m(E) (EntsoeApi)Communication error: 
(19:20:51.743) <ESC>[0m(E) (EntsoeApi)connection refused
(19:20:51.759) <ESC>[0m(D) (EntsoeApi)
```
until 

```
(19:25:45.585) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:25:46.662) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(19:25:46.677) <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
(19:25:48.846) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201232300&p<ESC>[0meriodEnd=202201242300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(19:25:48.861) <ESC>[0m(D) (EntsoeApi)Connection established
(19:25:48.877) <ESC>[0m(E) (EntsoeApi)Communication error: 
(19:25:48.970) <ESC>[0m(E) (EntsoeApi)connection refused
(19:25:48.970) <ESC>[0m(D) (EntsoeApi)
(19:25:48.970) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:25:50.172) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
[... 11 lines cut ...]
(19:25:56.739) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(19:25:56.755) <ESC>[0m(D) (EntsoeApi)Connection established
```
ending for a while

```
(19:27:35.160) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:27:36.673) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:27:40.183) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(19:27:40.433) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(19:27:40.433) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(19:27:40.448) <ESC>[0m(D) (EntsoeApi)Connection established
(19:27:40.448) <ESC>[0m(E) (EntsoeApi)Communication error: 
(19:27:40.479) <ESC>[0m(E) (EntsoeApi)connection refused
(19:27:40.479) <ESC>[0m(D) (EntsoeApi)
(19:27:40.479) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:27:41.665) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(19:27:41.681) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:27:44.161) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:27:46.673) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:27:50.183) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(19:27:51.665) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
```

Then repeat and repeat.

tronde-ams, 2022-01-24:
I re-flashed the 2.0.6 I can not log back to 220121.3 just after midnight. No prices as expected. 01:00 prices re-appeared correctly as before.

At the moment, I have one 220121.3 and one 2.0.4 showing correct prices, and one 2.0.6 not showing prices. I will disable prices on that unit now to avoid flooding the endpoints with refused requests.

gskjold, 2022-01-24:
Are you using static ip? If so, what have you used as dns?

gskjold, 2022-01-24:
Reason for asking, if you google "esp32 static ip connection refused" you will find a number of similar cases. Ex:
https://community.platformio.org/t/http-get-failed-error-connection-refused-when-using-static-ip/6968/7

If these cases are actual issues or if it's just malconfiguration is hard to tell, but it does suggest trying DHCP if you aren't already.

gskjold, 2022-01-24:
Very strange indeed. This problem remains a mystery to me. Works perfectly fine on my network. I've never seen a connection refused in my log.

In v2.0.6, timeout was increased from the default 1s to 5s. Attached is a firmware where I have set the timeout to 50s, just to see if that makes any change at all.

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7926374/esp32.zip)

tronde-ams, 2022-01-24:
I have been playing a little with the URL in the log

`https://transparency.entsoe.eu/api?securityToken="MY TOKEN&documentType=A44&periodStart=202201242300&p<ESC>[0meriodEnd=202201252300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2`

Does not work in a browser, but this does

`https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201242300&periodEnd=202201252300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2`

Is this extra text a bug in the log, or something that get injected by the code?
`<ESC>[0m`

This for norges-bank.no does work in a browser
`https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1`

but gives connection refused in the log.

?

gskjold, 2022-01-24:
It is the log framework that adds [0m

tronde-ams, 2022-01-24:
Now I got prices, but no HAN with new 220124.1

[image]

```
(16:08:54.033) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(16:08:54.033) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:08:59.234) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:04.325) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:09.471) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:14.605) <ESC>[0m(D) (loop)(C1) Used 866 ms to update temperature
(16:09:19.455) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:19.719) <ESC>[0m(D) (AmsWebServer)Serving /entsoe.html over http...
(16:09:20.272) <ESC>[0m(D) (AmsWebServer)Handling save method from http
(16:09:23.824) <ESC>[0m(D) (AmsWebServer)Received ENTSO-E config
(16:09:23.857) <ESC>[0m(I) (AmsWebServer)Saving configuration now...
(16:09:23.975) <ESC>[0m(I) (AmsWebServer)Successfully saved.
(16:09:24.042) <ESC>[0m(I) (EntsoeApi) Setting midnight millis 28272443
(16:09:24.076) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
(16:09:24.076) <ESC>[0m(I) (EntsoeApi) Fetching prices for today
(16:09:24.137) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201232300&p<ESC>[0meriodEnd=202201242300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(16:09:24.192) <ESC>[0m(D) (EntsoeApi)Connection established
(16:09:24.192) <ESC>[0m(D) (EntsoeApi)Receiving data
(16:09:25.717) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(16:09:25.752) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(16:09:25.752) <ESC>[0m(D) (EntsoeApi)Connection established
(16:09:25.781) <ESC>[0m(D) (EntsoeApi)Receiving data
(16:09:27.049) <ESC>[0m(D) (EntsoeApi)  got exchange rate 10.1366
(16:09:27.080) <ESC>[0m(D) (EntsoeApi) Resulting currency multiplier: 10.1366
(16:09:27.080) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:27.189) <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
(16:09:28.704) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201242300&p<ESC>[0meriodEnd=202201252300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(16:09:28.704) <ESC>[0m(D) (EntsoeApi)Connection established
(16:09:28.704) <ESC>[0m(D) (EntsoeApi)Receiving data
(16:09:30.562) <ESC>[0m(D) (AmsWebServer)Serving /dayplot.json over http...
(16:09:31.014) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(16:09:32.226) <ESC>[0m(D) (AmsWebServer)Serving /energyprice.json over http...
(16:09:32.278) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:32.314) <ESC>[0m(D) (loop)(C1) Used 866 ms to update temperature
(16:09:34.486) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:37.576) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:43.564) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(16:09:48.605) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:48.793) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:53.918) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:09:59.046) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(16:10:03.602) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:10:04.164) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(16:10:09.284) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
```

HAN came back when downgraded to 220121.3

[image]

gskjold, 2022-01-24:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7926689/esp32.zip)
Try this one

tronde-ams, 2022-01-24:
First impresdion is good :-)

I will kepp the log running.

[image]

hanseinar, 2022-01-25:
2.0.6 FAIL
220121.3 OK
ESP32
DHCP

tronde-ams, 2022-01-25:
220124.2 seems to be OK. Static IP as before.

No problem at midnight as far as I can see, and fecthed new prices at 13:58:56. You have made this random by now I understand?

```
(23:59:51.533) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(23:59:53.214) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:54.055) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(23:59:56.818) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(23:59:59.307) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:00.064) <ESC>[0m(I) (EntsoeApi) Rotating price objects at 1643065202
(00:00:01.271) <ESC>[0m(I) (EntsoeApi) Setting midnight millis 112388144
(00:00:01.353) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:01.689) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:04.043) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:05.104) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(00:00:05.104) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(00:00:05.104) <ESC>[0m(D) (EntsoeApi)Connection established
(00:00:05.104) <ESC>[0m(D) (EntsoeApi)Receiving data
(00:00:06.388) <ESC>[0m(D) (EntsoeApi)  got exchange rate 10.1366
(00:00:06.440) <ESC>[0m(D) (EntsoeApi) Resulting currency multiplier: 10.1366
(00:00:06.440) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:06.600) <ESC>[0m(D) (AmsDataStorage) Last day update: 1643061611
(00:00:06.600) <ESC>[0m(D) (AmsDataStorage) Last month update: 1642978811
(00:00:06.600) <ESC>[0m(D) (loop)(C1) Used 1 ms to update temperature
(00:00:06.600) <ESC>[0m(D) (AmsWebServer)Serving /energyprice.json over http...
(00:00:08.243) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:10.615) <ESC>[0m(D) (AmsDataStorage) Last day update: 1643061611
(00:00:10.957) <ESC>[0m(D) (AmsDataStorage) Last month update: 1642978811
(00:00:10.957) <ESC>[0m(I) (AmsDataStorage) Usage for hour 22: 2920
(00:00:10.957) <ESC>[0m(I) (AmsDataStorage) Usage for day 25: 58072
(00:00:10.957) <ESC>[0m(I) (readHanPort)(C1) Saving day plot
(00:00:10.957) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(00:00:11.568) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(00:00:12.230) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
```

```
(13:58:54.147) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
(13:58:55.598) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(13:58:56.315) <ESC>[0m(D) (AmsWebServer)Serving /dayplot.json over http...
(13:58:56.362) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(13:58:56.409) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(13:58:56.643) <ESC>[0m(D) (AmsWebServer)Serving /energyprice.json over http...
(13:58:56.705) <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
(13:58:58.562) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201252300&p<ESC>[0meriodEnd=202201262300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(13:58:58.593) <ESC>[0m(D) (EntsoeApi)Connection established
(13:58:58.593) <ESC>[0m(D) (EntsoeApi)Receiving data
(13:59:00.512) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(13:59:01.245) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(13:59:01.635) <ESC>[0m(D) (loop)(C1) Used 1 ms to update temperature
(13:59:01.666) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(13:59:01.697) <ESC>[0m(D) (AmsWebServer)Serving /index.html over http...
(13:59:03.819) <ESC>[0m(D) (AmsWebServer)Serving /application.js over http...
(13:59:04.022) <ESC>[0m(D) (AmsWebServer)Serving /github.svg over http...
(13:59:04.100) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(13:59:04.162) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(13:59:04.412) <ESC>[0m(D) (AmsWebServer)Serving /dayplot.json over http...
(13:59:04.474) <ESC>[0m(D) (AmsWebServer)Serving /monthplot.json over http...
(13:59:04.505) <ESC>[0m(D) (AmsWebServer)Serving /energyprice.json over http...
```

gskjold, 2022-01-25:
Great! Yes, the price fetch is random between 13:30 and 14:00

steggum, 2022-01-25:
We experienced the same issues with our ESP32. Earlier versions managed to fetch prices, but 2.0.5 and 2.0.6 has the same "connection refused" problem with both ENTSOE and norges-bank. Version 220124.2 seems to have fixed the issue and works fine so far.

tronde-ams, 2022-01-26:
220124.2 still OK.

Fetched new prices at 13:39:12 this time, så random seems to work as expected too.

gskjold, 2022-01-28:
Thank you for your feedback! Considering this issue fixed

tronde-ams, 2022-01-30:
Installed 2.0.7 and prices disappeared again. 
Connectoin refused for Entsoe and norges-bank.

Back to 220124.2 and everything fine.

gskjold, 2022-01-30:
There is no difference between those two versions

tronde-ams, 2022-01-30:
Ome works as expected, the other not. No idea about why.

2.0.7
```
(22:30:49.880) <ESC>[0m(I) (EntsoeApi) Setting midnight millis 5361009
(22:30:50.220) <ESC>[0m(I) (EntsoeApi) Fetching prices for today
(22:30:50.220) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201292300&p<ESC>[0meriodEnd=202201302300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(22:30:50.220) <ESC>[0m(D) (EntsoeApi)Connection established
(22:30:50.220) <ESC>[0m(D) (EntsoeApi)Receiving data
(22:30:51.820) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(22:30:51.850) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(22:30:51.850) <ESC>[0m(D) (EntsoeApi)Connection established
(22:30:51.880) <ESC>[0m(E) (EntsoeApi)Communication error: 
(22:30:51.930) <ESC>[0m(E) (EntsoeApi)connection refused
(22:30:51.960) <ESC>[0m(D) (EntsoeApi)
(22:30:51.960) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:30:51.990) <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
(22:30:51.990) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201302300&p<ESC>[0meriodEnd=202201312300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(22:30:52.020) <ESC>[0m(D) (EntsoeApi)Connection established
(22:30:52.020) <ESC>[0m(E) (EntsoeApi)Communication error: 
(22:30:52.070) <ESC>[0m(E) (EntsoeApi)connection refused
(22:30:52.070) <ESC>[0m(D) (EntsoeApi)
(22:30:52.070) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(22:30:52.860) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(22:30:52.910) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(22:30:52.910) <ESC>[0m(D) (EntsoeApi)Connection established
(22:30:52.910) <ESC>[0m(E) (EntsoeApi)Communication error: 
(22:30:52.910) <ESC>[0m(E) (EntsoeApi)connection refused
(22:30:52.940) <ESC>[0m(D) (EntsoeApi)
(22:30:52.940) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:30:53.870) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:30:56.360) <ESC>[0m(D) (loop)(C1) Used 866 ms to update temperature
(22:30:57.230) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(22:30:58.490) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(22:30:58.490) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(22:30:58.490) <ESC>[0m(D) (EntsoeApi)Connection established
(22:30:58.490) <ESC>[0m(E) (EntsoeApi)Communication error: 
(22:30:58.570) <ESC>[0m(E) (EntsoeApi)connection refused
```

Then back to 220124.2
```

(22:34:46.815) <ESC>[0m(I) (EntsoeApi) Setting midnight millis 5123000
(22:34:48.325) <ESC>[0m(I) (EntsoeApi) Fetching prices for today
(22:34:48.365) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201292300&p<ESC>[0meriodEnd=202201302300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(22:34:48.365) <ESC>[0m(D) (EntsoeApi)Connection established
(22:34:48.405) <ESC>[0m(D) (EntsoeApi)Receiving data
(22:35:07.824) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:08.194) <ESC>[0m(D) (MQTT_connect)(C1) Disconnecting MQTT before connecting
(22:35:08.224) <ESC>[0m(I) (MQTT_connect)(C1) Connecting to MQTT test.mosquitto.org:1883
(22:35:08.224) <ESC>[0m(I) (MQTT_connect)(C1) Successfully connected to MQTT!
(22:35:08.324) <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
(22:35:08.324) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202201302300&p<ESC>[0meriodEnd=202201312300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(22:35:08.364) <ESC>[0m(D) (EntsoeApi)Connection established
(22:35:08.364) <ESC>[0m(D) (EntsoeApi)Receiving data
(22:35:08.625) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(22:35:08.665) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(22:35:08.705) <ESC>[0m(D) (EntsoeApi)Connection established
(22:35:08.705) <ESC>[0m(D) (EntsoeApi)Receiving data
(22:35:10.278) <ESC>[0m(D) (EntsoeApi)  got exchange rate 10.1366
(22:35:10.328) <ESC>[0m(D) (EntsoeApi) Resulting currency multiplier: 10.1366
(22:35:10.328) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(22:35:10.708) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:10.958) <ESC>[0m(D) (loop)(C1) Used 866 ms to update temperature
(22:35:11.868) <ESC>[0m(D) (AmsWebServer)Serving /energyprice.json over http...
(22:35:11.868) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:11.948) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:14.488) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(22:35:16.618) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:16.828) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:20.328) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:21.851) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(22:35:22.188) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:24.317) <ESC>[0m(D) (loop)(C1) Used 1 ms to update temperature
(22:35:24.353) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:26.816) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(22:35:27.885) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:30.329) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(22:35:31.857) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(22:35:33.457) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
```

hanseinar, 2022-01-31:
Same experience
Installed 2.0.7 and prices disappeared.
Back to 220124.2 and everything fine.

gskjold, 2022-01-31:
Increasing timeout from 20s to 50s:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7968830/esp32.zip)

gvenaas, 2022-01-31:
That work for my device.

hanseinar, 2022-01-31:
That work for my device also.

tronde-ams, 2022-01-31:
220131.1 seems to be OK for me too.

closed 2022-02-04 by gskjold