UtilitechAS/amsreader-firmware #208: Daily total missing on ESP8266 (Pow-K)

henriklund, 2022-01-17:
**Describe the bug**
Since v2.0.3 the daily total is missing

**To Reproduce**
Steps to reproduce the behavior:
This one is hard to replicate. It started to occur after upgrade to v2.0.3 and seems to coincide with a spontaneous restart. At 07:36 this morning, the view was as follows (uptime at that point was 4 hours - there seems to have been a restart at 03:27 this morning):
[image]

Now (four hours later), the picture has changed:
[image]
(in the meantime the module restarted twice; 0915: and 10:57)

**Expected behavior**
Daily total to be retained / visibled

**Hardware information:**
 - Meter: Kamstrup (DK)
 - AMS reader: Pow-K
 - M-bus adapter (if applicable): N/A

**Relevant firmware information:**
 - 2.0.5
 - MQTT: yes
 - HAN GPIO: GPIO3
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: N/A

ArnieO, 2022-01-17:
Can you check your setting on System/Debugging. In addition to all "off", Debug level must be set to "Error". 
Like this:
[image]

henriklund, 2022-01-17:
It is already set to error :)

ArnieO, 2022-01-17:
> It is already set to error :)
OK, good. I had the same issue until I discovered that I had it on "Debug" (and thought that with Telnet disabled it did not matter; but it does). 

Interesting problem, as I had the clear impression that this issue was fixed now.
Just to rule anything else out: Could you show me what you have on System/GPIO?

henriklund, 2022-01-17:
[image]

ArnieO, 2022-01-17:
There is no temperature sensor on Pow-K, so please delete the "5" on "Temperature".
I believe trying to read a non-existing temp sensor "eats" processing time and could cause your issue. 

(Not related to your problem: You could also calibrate the ESP8266 ADC by adjusting "Multiplier" until the "ESP volt" value shows close to 3.3V.)

henriklund, 2022-01-17:
Ok. Changed the 5 to nothing and applied that. After changing the multiplier, the Pow-K restarted (again)
[image]

henriklund, 2022-01-18:
Changing those values did not do change anything significantly. Pow-K still restarts from time to time, and the daily stat still resets to 0.

ArnieO, 2022-01-18:
Thank you @henriklund .
Any idea here, @gskjold ?

gskjold, 2022-01-18:
Han this issue reported by email on a Norwegian meter as well. No idea yet, but must be a code issue. Thanks for catching that it was first set and then cleared, that narrows the search a lot!

gskjold, 2022-01-21:
Not sure what it is yet, but to narrow the search a bit more I have removed some code. Here is a firmware you can test:
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7915457/esp8266.zip)

henriklund, 2022-01-21:
Just FYI... last few days are a bit odd. At one point I noticed 2 bars of 23kWh, now there is one. However, there now also is 2x51kWh. Of the three last days, only the data from the 20th appear correct.

[image]

I have uploaded the new firmware and will report back on my observations :)

gskjold, 2022-01-22:
Looks like it missed midnight to 20th, and then used a total of 102kwh between midnight from 19th to 21st and spread that across the two missing days. Hopefully your graph will go back to normal now.

henriklund, 2022-01-22:
Just FYI, the actual consumption for 19th and 20th is 106. I also note that the Pow-K restarted 00:00:44 (just after midnight between 20th and 21st). Nevertheless, let's chalk this off as 'onetime'.

The Pow-K stil restarts at irregular intervals, however, now the data from past seem to persist:
[image]
 
The numbers are a wee bit of (compared to previously; from the 19th I would expect numbers around 55, 51 and 52 (I know that past numbers numbers do not change). I will keep an eye on that for the next few days.

henriklund, 2022-01-23:
Numbers are still 'off' for the past 5 days. Now the hourly accumulations are also disappearing. On top of that, the hourly measurement are not an actual representation of spend; 07 was 1,44 and 08 was 2,01 (and not 1,7 / 1,9). :/

[image]

henriklund, 2022-01-24:
Just an update....Numbers are still 'off' (by more than a factor 2 for yesterday).
[image]

gskjold, 2022-01-25:
I have not found out why this happens, but from experience the readings normalize after two lows and a super high. I've seen this on several devices now. I know how to replicate it, so I will experiment one weekend. Let me know how yours turn out the next days.

ArnieO, 2022-01-25:
Like this (those 3 initial points are from a previous code version). 
Running 2.0.4 now, and all good. 
[image]

henriklund, 2022-01-26:
Interesting behaviour.... however, after a two-low-one-superhigh data start to look normal.
[image]

gskjold, 2022-01-26:
Cool. One day I'm going to figure out what is going on there :)

henriklund, 2022-01-27:
Check my last picture (above). This was on par for the 24th / 25th with my Home Assistant (43.98 ; 45.,79).

Now the 25th has reset to 41 - actual usage ref. HA is 40.67.
[image]

(I am still running the FW uploaded to this issue last week). So something is still not right :/

I do not know if following observation can be useful for you (@gskjold)...
I have made a comparison between the monthplot.json and data received through MQTT for the past month. The MQTT data (tPI) allows a calculation of daily consumption; this differs from what can be found in monthplot (by up to +/- 1.5). I have also checked that tPI is on par with the meter (it is) and that each tPI reading I have is >= previous (it is).

henriklund, 2022-02-25:
Running v2.0.10 resolved the problem, however, since v2.0.11, the issue appears to be back:

- on 2022-02-23 00:03 daily total for 22nd updated to 48,32
- on 2022-02-23 22:03 daily total for 22nd changed to 0
- on 2022-02-24 01:03 daily total for 22nd changed to 41,31 (and 23rd was set to same) 
- on 2022-02-25 01:03 daily total for 24th was set to -2,6 (I do not produce electricity)

[image]

gskjold, 2022-03-12:
Should be fixed in 2.1.0

closed 2022-03-12 by gskjold