UtilitechAS/amsreader-firmware #209: Unable to flash firmware

stigvi, 2022-01-18:
**Describe your problem**
I tried to upgrade my 2.0.3 version, but it was not successful. The web interface is still reporting version 2.0.3 and I have refreshed my browser and even tried another computer. Could someone see anything wrong in the log below?

**Hardware information:**
 - AMS reader: ESP32

`PS D:\OneDrive - Heime\Data Files\Hjemmeautomasjon\AmsToMqttBridge> esptool.py --chip esp32 --port COM8 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 .\bootloader_dio_40m.bin 0x8000 .\ams2mqtt-esp32-partitions-2.0.5.bin 0x10000 .\ams2mqtt-esp32-2.0.5.bin
esptool.py v3.0
Serial port COM8
Connecting........___
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:11:ae:c8:e0:c0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 17104 bytes to 11184...
Wrote 17104 bytes (11184 compressed) at 0x00001000 in 0.3 seconds (effective 523.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1795.9 kbit/s)...
Hash of data verified.
Compressed 1232912 bytes to 742568...
Wrote 1232912 bytes (742568 compressed) at 0x00010000 in 17.6 seconds (effective 560.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
PS D:\OneDrive - Heime\Data Files\Hjemmeautomasjon\AmsToMqttBridge>`

ArnieO, 2022-01-18:
This is a known issue with ESP32 on the first 2.0.x versions (see other issues here). 
Fixed in v2.0.5.

stigvi, 2022-01-18:
> This is a known issue with ESP32 on the first 2.0.x versions (see other issues here). Fixed in v2.0.5.

I dont think you read my post. This is not about OTA updating. I am not able to flash my esp32 unit and was looking for some hints & tricks.

ArnieO, 2022-01-18:
You're right - I misunderstood. Sorry!

This is what I would try in your case:

- Erase flash completely (esptool.py erase_flash) --> [Documentation](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html?highlight=erase_flash#erase-flash-erase-flash-erase-region)
- Upload a basic blink sketch, verify that it blinks a LED.
- Reflash v2.0.5

If it still insists it is v2.0.3 ... Maybe an undetected error in the bin file (wrong version indication, @gskjold ?)

stigvi, 2022-01-18:
> You're right - I misunderstood. Sorry!
>
> This is what I would try in your case:
>
> * Erase flash completely (esptool.py erase_flash) --> [...]

Thanks. Erase flash did the trick. Strange.......

ArnieO, 2022-01-20:
I agree - this is strange. 
Your esptool.py command options seem to be exactly what is proposed in 
https://github.com/gskjold/AmsToMqttBridge/wiki/flashing, so maybe that is not the optimal set of options in all cases. 

Anyway, I think it is not uncommon to see cases with ESP8266 and ESP32 where the device appears "bricked" until a full flash erase is done. I've never really understood why.