UtilitechAS/amsreader-firmware #210: Undecoded messages to MQTT

toreamun, 2022-01-19:
It would be great if binary messages received from the HAN-port data stream could be published to a MQTT-topic. This would make a great fit with other solutions capable of parsing meter data from MQTT, like this Home Assistant integration: https://github.com/toreamun/amshan-homeassistant

Alternatives:
- Alt A: The format of the published message exactly as received from the meter (with or without flag-sequence at the start/end). Messages with invalid checksum and length can also be published (the receiver can validate the checksum).
- Alt B: Strip away the HDLC-framing and only publish the contained COSEM data in the APDU part. The data must be unescaped, and the checksum and expected frame length should be correct for the message to be published.

[image]

ArnieO, 2022-01-19:
Not sure if I understand the request, as MQTT is implemented, and data published to the topic you specify:
I'm currently not using MQTT, so this is just to illustrate:
[image]

toreamun, 2022-01-19:
Maybe I read the documentation wrong. Payload format described in https://github.com/gskjold/AmsToMqttBridge/wiki/Message-formats are JSON, not the message in same format as received from the meter. The [MQTT documentation](https://github.com/gskjold/AmsToMqttBridge/wiki/MQTT-configuration) lists 4 formats (JSON, Raw full, Raw minimal and Domoticz). None of them are described as the original message from the meter. I have also looked at the [MQTT source code](https://github.com/gskjold/AmsToMqttBridge/tree/master/src/mqtt) and I cant see that any of them is what i suggest.

ArnieO, 2022-01-20:
I read your initial message too rapidly, sorry!
What you request is a new option for MQTT formats, sending the raw data byte stream from the meter to MQTT (with or without HDLC framing).

So I will leave it to @gskjold to consider!

toreamun, 2022-01-20:
Yes, raw data byte stream. Both options (with or without HDLC framing) is not needed, but at least one of them would be great. Tibber Pulse sends full HDLC frame and works fine with [AMSHAN custom integrationHome Assistant](https://github.com/toreamun/amshan-homeassistant) for Home Assistant. I am the maintainer of that integration.

gskjold, 2022-01-20:
Agree this could be useful. Will implement support to send full frame as bytes to MQTT.

closed 2022-02-08 by gskjold