UtilitechAS/amsreader-firmware #211: Solar panels, strange readings.

Isaksson, 2022-01-20:
**Describe your problem**
Strange values after solar panel installation

**Hardware information:**
 - Meter: Aidon Sweden
 - AMS reader: ESP8266

**Relevant firmware information:**
 - Version: 2.0.4
 - MQTT: yes
 - HAN GPIO: UART2

Yesterday I got my solar panels installed and connected to the grid, and now the meter report very strange readings on my Phase current.
Have anyone seen this behavior before?
[image]

gskjold, 2022-01-20:
Thanks, I see now that current values pass through a unsigned int, will fix in next patch.

Isaksson, 2022-01-20:
Yes, I just verified on my meter and its because the current has a negative value now when exporting to the grid.
Thanks.

gskjold, 2022-01-21:
Try this version

[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7915361/esp8266.zip)

Isaksson, 2022-01-21:
Thanks for that, just installed the update, I used my mobile phone really nice to be able to update from mobile device 👍
Let's hope for some sun tomorrow so that I could verify this new version.

Isaksson, 2022-01-22:
It's working as expected, thanks.
[image]

gskjold, 2022-01-22:
Awesome! Thanks for confirming

closed 2022-01-22 by gskjold