UtilitechAS/amsreader-firmware #212: No connection to HAN or ESPHome

JJakaJonas, 2022-01-20:
**Describe your problem**
My AMS bridge do not take information out from my kamstrup powermeter.
I do know that the powermeter put data out do to my PC got the data when it was hooked up to the powermeter.
i also have the 2 encryptioncodes. 

also the AMS says that the powermeter manufacturer is Unknown in version 2.0.0 and up.

**Hardware information:**
 - Meter: Kamstrup Omnipower
 - AMS reader: ESP8266 NodeMCU V3
 - M-bus adapter (if applicable): not that  i know of

**Relevant firmware information:**
 - Version: 1.5.3, 1.5.8, 2.0.0, 2.0.3, 2.0.5
 - MQTT: yes
 - HAN GPIO: GPIO13
 - Temperature sensors: no

ArnieO, 2022-01-21:
How have you connected the NodeMCU to the meter? What is the power source for the NodeMCU? Have you run a Telnet debug (menu: System/Debugging) to see what is going on?

JJakaJonas, 2022-01-21:
I have connected pin 6 to GPIO g and pin 5 to pin GPIO13.

The power source for now is a USB cable.

I have tryed a serial debug were the only error is some price API

ArnieO, 2022-01-21:
Power source: OK. Just ensure the wall plug has enough "juice". 

Not sure I understand "pin 6 to GPIO g and pin 5 to pin GPIO13." How do you hook up to the meter? 
A photo is potentially helpful. 

Try to enable Telnet (System/Debugging).
Then run in a command window:  `C:\> telnet <IP adr of NodeMCU>`
And show us a screen dump of what you get.

SVH-Powel, 2022-01-21:
> **Describe your problem** My AMS bridge do not take information out from my kamstrup powermeter. I do know that the powermeter put data out do to my PC got the data when it was hooked up to the powermeter. i also have the 2 encryptioncodes.
> 
> also the AMS says that the powermeter manufacturer is Unknown in version 2.0.0 and up.
> 
> Also i cant get it to communicate whit ESPHome on my Home Assistant setup.
> 
> **Hardware information:**
> 
> * Meter: Kamstrup Omnipower
> * AMS reader: ESP8266 NodeMCU V3
> * M-bus adapter (if applicable): not that  i know of
> 
> **Relevant firmware information:**
> 
> * Version: 1.5.3, 1.5.8, 2.0.0, 2.0.3, 2.0.5
> * MQTT: yes
> * HAN GPIO: GPIO13
> * Temperature sensors: no

You cannot connect this to Home Assistant via esphome. AmstoMqttBridge require a mqtt broker and from there into HA.

JJakaJonas, 2022-01-21:
> Power source: OK. Just ensure the wall plug has enough "juice".
>
> Not sure I understand "pin 6 to GPIO g and pin 5 to pin GPIO13." How do you hook up to the meter? A photo is potentially helpful. [...]

The wall plug shuld have more than enough "Juice" to run the device.
It a power plug from a Chromecast gen 3.

Heres a photo of my setup as it is now.
[image]

my computer tells me that telnet is not a recognized as an internal or external command, operable program or batch file.

>

JJakaJonas, 2022-01-21:
> You cannot connect this to Home Assistant via esphome. AmstoMqttBridge require a mqtt broker and from there into HA.

I think you misunderstand my "ESPHome" thing...

I know the data goes from meter to AMS in there it goes to mqtt and there it gets to my mqtt server at HA...

My ESPHome problem is to get the progroming of the chip part im to ESPHome. 
Theres a link in my AMS webinterface there tells me theres no connection to ESPHome
[image]

ArnieO, 2022-01-21:
Your voltage is low, indicated at 2,95V. Is that real or due to something on the NodeMCU? 
You have to activate Telnet under Windows, it is not activated by default. Google will help you.

JJakaJonas, 2022-01-21:
i have read someware that the NodeMCU had som powerdivider thing going on.

But just to be sure will i change it out whit an ipad power plug (that shuld deff have the juice needed) and try a debug again.

JJakaJonas, 2022-01-21:
btw.... Google is your friend when you know what you are looking for... 

I'm clueless and have already tryed google for help for a few days and found nothing... therefor this post... to get pointed the right direction.......

JJakaJonas, 2022-01-21:
More or less the output when the option is set to debug.. 
Error shows even less than this.
[image]

And wiht the iPad charger the ESP voltage is the same as before.

JJakaJonas, 2022-01-21:
heres a serial debug at boot i made yesterday
[image]
.

ArnieO, 2022-01-21:
Ok, thanks. This shows there is nothing coming from the meter. 
You wrote earlier that you receved data from the meter to the computer? 
I also see from your photo that you have a wire on the data_in line, which you should not need to use. 
Where in Denmark are you located? Could I ask you to go to amsleser.no and consider what is written on Pow-K, there is an issue with Nord Energi Net.

JJakaJonas, 2022-01-21:
yeah.. I have a sparkel fun chip that i use to flash with. 
If i take that and connect it as a replacement for the ESP i get pushed data every 10. sec. 
and it was whit the same info as i use on the ESP (exept the keys do to it not was needed to test the connection)

the data in cable is only there for flashing the chip. 
I have a 6 pin connecter on the meter end. the cable is not connected to the ESP neither is the power cable.

i have been looking at that guide (the reason for the 1.5.3 version as i remember do to som of the guide was made on that version) but that did not help me at all.

JJakaJonas, 2022-01-21:
Totaly forgot to answer the last part.
I'm located in the north part of zealand (Nordsjælland) and i got my encryption keys from Radius (the owner company of my meter) and i have pasted them in to the meter setting at there respective place in the setup.

I also just upgraded to version 2.0.2 de to another part in the pow-k guide. but whit no luck.

JJakaJonas, 2022-01-21:
I have soved the problem.. 

how stupid it may sound was the cables between the meter and the ESP the problem..
And the worst part is i have changed them 5 or 6 times to rul them out..

JJakaJonas, 2022-01-21:
Now i just have a problem with the error "invalid HDLC, returned with -1" when i boot up my ESP.

ArnieO, 2022-01-22:
You’re not the first to experience that Dupont wires are not very reliable. Good to hear that you solved it.

gskjold, 2022-01-22:
It is not uncommon that the first couple of packages received are invalid. And there will always be a few that returns boundry flag error (-1) or checksum error (-2) every now and then, especially with development boards.

JJakaJonas, 2022-01-22:
okay . so everything here is good to go now as i understand it.

and as i see it is theres no way to get it to work whit my ESPHome instans on my home assistent?

ArnieO, 2022-01-24:
> and as i see it is theres no way to get it to work whit my ESPHome instans on my home assistent?

I don't have competence on HA/ ESPHome so cannot answer specifically on that. 
There may be other ways to do it, but the common way to transfer data to a home automation system is via MQTT.

JJakaJonas, 2022-01-24:
> > and as i see it is theres no way to get it to work whit my ESPHome instans on my home assistent?
> 
> I don't have competence on HA/ ESPHome so cannot answer specifically on that. There may be other ways to do it, but the common way to transfer data to a home automation system is via MQTT.

i think you misunderstand me..

I use the MQTT and i plan to keep on with that.

It's the programing of the ESP i was hoping to do from the ESP add-in, in HA. 
But i can't find a way to do it. also its not importent anyway. it was just a nice feat if it was possible.