UtilitechAS/amsreader-firmware #213: Wrong lables in graphs

Lommelund, 2022-01-20:
The graphs started to show wrong labels after  upgrade.  Meter is configured as TN 400V.

Have tried to switch between IT/TT and TN without any change. 

[image]

**Hardware information:**
 - Meter:  Aidon
 - AMS reader:  Pow-U

**Relevant firmware information:**
 - Version: 2.0.5
 - MQTT: yes
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]

The device reboots more often after upgrade from 2.0.3 to .4 and .5. Before I could see uptime several days. Now i see max a few hours. 
I power it from USB.

tronde-ams, 2022-01-20:
I guess this has something to do with a change made for 230V IT/TT meters.

For IT/TT this is correct, but for 400V TN it should read "Phase Voltage" and L1, L2, L3 as it was from the beginning. You do not show the labels for current, so I do not know if they changed as well. For 400V TN they should read "Phase current" and L1, L2 , L3 as from the beginning.

Lommelund, 2022-01-21:
Both voltage and current show the same.
L1-L2, L1-L3, L2-L3.

gskjold, 2022-01-21:
Fixed in next patch

closed 2022-01-21 by gskjold