UtilitechAS/amsreader-firmware #214: PowU does not pulish accumulated power

paaland, 2022-01-25:
**Describe the bug**
I've got a PowU connected to an Aiden meter in Norway. I can see in the UI that the accumulated power value reported by the meter change, but it's not reflected in the what's published in mqtt.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to AMS reader web page
2. Open MQTT explorer, examine the meter/import/active/accumulated
3. The value should correspond with that's in the web page
4. It does not

**Expected behavior**
When a new accumulated value is received, it should be published to mqtt as all the other values.

**Screenshots**
[image]

**Hardware information:**
 - Aidon
 - Pow-U

**Relevant firmware information:**
 - Version: 2.0.6
 - MQTT: yes
 - HAN GPIO12
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: none

**Additional context**
All the other values are published fine, it's just meter/import/active/accumulated that is not updated.

gskjold, 2022-01-25:
You will only receive a confirmed accumulated value at the top of the hour (approx ??:00:10). What you are seeing in the GUI is a real time calculated value. This value is mostly correct, but can deviate a bit, both higher and lower, so there is a good chance you will see a small jump in the value when the confirmed reading is received.

ArnieO, 2022-01-25:
So the value difference GUI vs MQTT is:
- GUI: Last hourly value from meter + estimated accumulated so far this hour 
- MQTT: Last hourly value from meter

gskjold, 2022-01-25:
Thanks for the summary, yes, that is correct.

paaland, 2022-01-25:
I "lost" a value too. I got a value at 11, then nothing at 12, but I got a value at 13. 
This cause the "previous" usage value to be reused. I think the whole thing might tie into this bug: https://github.com/gskjold/AmsToMqttBridge/issues/199

[image]

gskjold, 2022-01-25:
The graph is calculated from usage between successfully received meter readings (only received once every hour), if a reading is missed, the next calculation will be spread across all hours since last successful calculation. In your case, 9.8kwh have been used between 10-12 and was divided over the two hours.

tronde-ams, 2022-01-26:
Is this real time accumulated power a special for ESP8266 based units? I have only seen real time values when I power up an ESP32 until it reaches hh:00:10. From then it is only updated once an hour.

[image]
[image]

No change after 15 min. I reloaded the page before each capture.
[image]

Isaksson, 2022-01-28:
Could you please use the DevTools in chrome (F12) and verify on the Network tab that your browser is getting the updated data.
[image]

tronde-ams, 2022-01-28:
I don't have chrome on my desktop, and I have no inttention to install it either.

I have not seen real time update on any of my ESP32 based units  regardless of using Vivaldi or Firefox on two different desktops, or Vivaldi, Firefox or Chrome on android. I have no ESP8266 to test with at the moment. 

It does show real time when powered up until first hh:00:10, but not after. 
My meter is Aidon three phase IT.

I cut power on one unit for a re-start.

Counts steadily from 0

[image]

[image]

[image]
[image]
[image]
[image]
[image]

.

Fetches total from meter at 22:00:10
[image]

.
Then stops counting
[image]

Isaksson, 2022-01-28:
There is a developer function in Firefox, if you are on Windows you open it with F12.
https://developer.mozilla.org/en-US/docs/Tools

That browser also have the network monitor function.

tronde-ams, 2022-01-28:
Don't tell me much, I am afraid ;-(
[image]

Isaksson, 2022-01-28:
And you have to refresh the webpage to see any updates?
So between all the screenshots above you had to refresh the webpage?

tronde-ams, 2022-01-28:
No, I don't need to. 

I did for the first screenshots three days ago to make sure nothing was stuck, but I did not refresh today.

First screenshot here is same as 22:00
[image]

Next without refresh is updated with value from meter at 23:00:10
[image]

tronde-ams, 2022-01-28:
Previous screen shots was from Firefox

This is from Vivaldi. Same accumulated value, but seven minutes later, and different browser.
[image]

tronde-ams, 2022-01-28:
Firefox on android. Still same value
[image]

closed 2022-03-23 by gskjold