UtilitechAS/amsreader-firmware #215: no data shown. Meter outputs something every 10 sec.

zaico, 2022-01-27:
Hi

I can't seem to get data showing or verifying that the ESP8266 is reading the data.

I have an ESP8266 NodeMCU and have connected wires to GND and GPIO3 and to pin 5 and 6 on the meter.

I have received keys and input them in the AMS Web settings.

I have measured on pin 5 and 6 on the meter and it's giving 3.28v and every 10s dipping to 1.5v and then going back up to 3.28v so from what I can gather that means the meter is sending out data.

AMS reader is just not showing any data and HAN is read.

If I restart the esp HAN is green for 10 sec then turns yellow for 2-4 sec and then turns red.

debug shows nothing else then it took 1s to update temp. 

I hope some of you with more knowledge than me can help me.

Best regards
Alex

ArnieO, 2022-01-27:
Which country are you in? Pins 5 and 6 indicates to me you might be measuring on a meter with an RJ12 connector?

zaico, 2022-01-27:
Never mind, it's working now. Out of sudden data has started to come in.

closed 2022-01-27 by zaico

ArnieO, 2022-01-27:
Dupont pins are not very reliable, you are not the first with this kind of issue.
Denmark and Kamstrup, I assume.

zaico, 2022-01-27:
Yes Denmark and Kamstrup ;)

zaico, 2022-01-27:
The setting for main fuse, do you know what impact that has on the data?

gskjold, 2022-01-28:
https://github.com/gskjold/AmsToMqttBridge/wiki/Meter-configuration