UtilitechAS/amsreader-firmware #217: Norwegian electricity distribution tariff

gskjold, 2022-01-29:
Upcoming distribution tariff in Norway forces users to distribute consumption throughout the day by applying higher monthly fee if specified thresholds are exceeded.

Implement the following:
- [ ] Configuration of thresholds
- [ ] Real time tracking of imported electricity
- [ ] Include use for current hour in MQTT messages
- [ ] Include threshold for current month in MQTT messages
- [ ] Calculate cost if ENTSO-E API is enabled and available

ujhede, 2022-01-29:
Just out of interrest how does this work ?   If you spend more than x kwh during expensive hours you will pay a higer tarrif for the entire months usage or is it a fixed monthly amount ?

if you are going to include support for adding tarrifs, then you might also consider other models lige the one we use in Denmark where the distribution fee is higer for consumption between 5pm and 8pm between  October and March. 

See [https://radiuselnet.dk/elkunder/priser-og-vilkaar/tariffer-og-netabonnement/](url)

gskjold, 2022-01-29:
There are two parts to our distribution tariff; cost per kWh and a fixed cost per month. Currently these are fixed numbers throughout the year, but soon they will implement a new scheme where the fixed cost per month is higher for people who does not distribute their load. The way they do that is to increase the monthly fee if the consumption for any given hour within that month is higher than specified thresholds. Some more information about that from my local grid company here: https://www.l-nett.no/nynettleie/slik-blir-ny-nettleie-og-pris

Right now I am focusing on calculating and presenting the consumption within the current hour. The calculated cost is only for the electricity, not distribution cost. Implementing full support for energy tariff is not something I plan to do.

ujhede, 2022-01-29:
Wonder if they considered the thresholds bands carefully or that they don't know that most people with EV chargers will not be able to stay in the 5-10 kwh band without throtteling their 11kw chargers and turning off everything else when charging. :-)

bardahlm, 2022-01-29:
I'm currently using data from AMS to adjust how much I charge my car. By adjusting the current setting every minute I get close to my goal for the hour. For this to work one need to know instant consumption, average consumption in the current hour and how much the car is charging.

My formula is basicly this:

goal = 4900Wh # want to keep my consumption below 4900Wh/h
usage = min( current consumption, average consumption ) # or some smarter function that give a smoother curve
available = goal - usage + chargepower
charging_current = available / voltage

I charged my car this night from 0400 to 0800:

[image]

The charging current varied a lot to cope with other loads (note that the following graph has data every five minutes):

[image]

I ended up using 4,7 kWh, 4,8 kWh, 4,8 kWh and 4,8 kWh in those four hours.

[image]

tronde-ams, 2022-01-29:
Not only Norway. 
Here is something from Gothenburg in Sweden.
https://www.goteborgenergi.se/privat/elnat/elnatsavgift#prisvilla

I understand the Swedish approach to this is to allow for a zillion different versions, so you might be opening a can of worms by adding such functionalty to the code :-)

gskjold, 2022-01-29:
Very cool @bardahlm :)

ArnieO, 2022-02-01:
> Just read the below part again, do they mean that if you just hit a single hour in a month you will pay the higer "netleie" that month then you have to be crazy carefull not to use the owen, washing machine and tumble dryer while charging your car :-) That could potentially cost you NOK 250 just to do that once in a month.

@ujhede Exactly, and there are plenty of thinkable examples that do not sound reasonable.
Imagine grandmother living alone in her little house, saving as much as she can on electricity, in general low consumption. One day getting her son's family visiting for a few hours, and they borrow a plug to top up the car battery for the return trip. If one single hour crosses the power limit, grandma gets 250 NOK (or around 500 NOK if also next limit is passed) "penalty" added to her electricity bill next month.

@tronde-ams Indeed, this will be difficult. Also in Norway, the steps are not the same in all grid companies.
I hear rumors the grid companies are working on a common API to make this somewhat easier - if it becomes useable.

closed 2022-02-08 by gskjold

CaptainAnchor, 2022-05-20:
Is this something that might be able to be implemented: https://github.com/digin-energi/API-nettleie-for-styring