UtilitechAS/amsreader-firmware #218: Improve MQTT integration by adding status/debug topic and pr. messgae type subtopics

ujhede, 2022-01-29:
**Is your feature request related to a problem? Please describe.**

Just had an "incident" where for some reason data stopped flowing form the meter at midnight last night. I only noticed this because I am still having problems with ENTSO price download currency and was trying to find out what was happening.

**Describe the solution you'd like**

Would be very uselful if you could post a status/debug topic to MQTT so that you can setup alerts when the HAN indicator changes or the ENTSO download fails.  This could potentially also be used as an alternative to telnet debugging where logmessages could be sent to MQTT.  

Now that we are are at the subject of MQTT would you consider adding an option to post the different message types to their own sub-topic?   Especially consumption and price mesages would be very usefull and avoid parsing errors in HA, when a message does nor have the expected json object like data or prices. Not sure if this is relevant when we get HA autodiscovery.

And finally would you consider add the exchange rate used to convert prices to both the GUI and the prices mqtt message?

**Describe alternatives you've considered**

This is a perfect idea so I simply cannot think of any alternative :-)

**Additional context**
Add any other context or screenshots about the feature request here.

gskjold, 2022-01-30:
Thanks for the suggestion, I will think about how we could do this.

gskjold, 2026-01-29:
Status topic have been added in v2.3 via #722 
Topic per message was added in v2.3 via #749 
Currently planning for logging via MQTT in #1128

Closing this issue as it has been covered by others over the years.