UtilitechAS/amsreader-firmware #221: ESP32, ENTSOE API error

gvenaas, 2022-01-29:
**Describe the bug**
ENTSOE API error

Getting error on the ESP32, but tested OK in a Postman connection.

**To Reproduce**
Added my API key, selected NO2, 
No graph appeared on the WEB page

**Screenshots**
From debugging, Telnet Error level W
(E) (EntsoeApi)Communication error:
(E) (EntsoeApi)connection refused
(E) (EntsoeApi)Communication error:
(E) (EntsoeApi)connection refused

**Hardware information:**
 - Meter: Kamstrup
 - AMS reader: Wemos Lolin D32
 - M-bus adapter: TSS721 Module M-BUS To TTL with RX TX Indicator STM32

**Relevant firmware information:**
 - Version: [e.g. 2.0.6]
 - MQTT: no
 - HAN GPIO: GPIO16]
 - HAN baud and parity: [e.g. 2400 8N1]
 - Temperature sensors No

**Additional context**

gvenaas, 2022-01-29:
Set debug to I

(I) (EntsoeApi) Retrieving EUR to NOK conversion
(E) (EntsoeApi)Communication error:
(E) (EntsoeApi)connection refused
(I) (EntsoeApi) Retrieving EUR to NOK conversion
(E) (EntsoeApi)Communication error:
(E) (EntsoeApi)connection refused

gvenaas, 2022-01-29:
When changing to EUR in config, The prices appeared at once..

tronde-ams, 2022-01-29:
Try the last zip-file published in this topic

https://github.com/gskjold/AmsToMqttBridge/issues/205

It should give you a test version 220124.2 which solved trouble with prices for me.

gvenaas, 2022-01-29:
Seems to be the same experience that I have... Closing this issue, Thanks.

closed 2022-01-29 by gvenaas