UtilitechAS/amsreader-firmware #223: Support for Aidon 6550 - Direct meter 160a

netforceatg, 2022-02-01:
Hello,

Have a few of these running at friends and family, thanks for a great project. 

On one of the locations there's a Aidon 6550, 400V TNS 3-phase 160Amp fuse(Lyse). Voltage readout is fine. But current and Watt are wrong.  Is it possible to add support for this meter? 

I've attached debug output and picture. see below.

Telnet debug verbose:
https://pastebin.com/Cgcy5dKj
[image]

[image]

gskjold, 2022-02-01:
Had a look at your debug output, and the output is exactly the same as any other Norwegian Aidon meters. I see on your screenshot here that the uptime is only a few seconds, is it still the same after beeing up for a while? I needs to successfully get past beginning of an hour to show kwh reading.

Please note that having debugger enabled can cause sudden reboots and should not be used during normal operation.

ArnieO, 2022-02-01:
Potentially the same issue as https://github.com/gskjold/AmsToMqttBridge/issues/216 (alternative current transformers installed).

@netforceatg : Are you able to identify if there is a fixed factor between meter display and value read by AmsToMqttBridge?

gskjold, 2022-02-01:
The Aidon payload specifies scaling, so that should not be an issue if it is used correctly. But we all know how consistent NVE spec have been, so who knows...

@netforceatg when the GUI is all synced up with kwh reading and everything, send me a screenshot, a debug dump and explain which numbers differs from reality and by how much.

netforceatg, 2022-02-01:
Updated screenshot and pastbin, and hourly-consumtion values(from january from gridowner)

The daly consumption on this meter is averaging at 105-160kWh. From Lyse Lnett portal the consumtion from 1. January to 1.Febuary was 3809kWh. 

So I expect to see much higher numbers on current and kW side. 

https://pastebin.com/nmm7uDh0

[Timeverdier.xlsx](https://github.com/gskjold/AmsToMqttBridge/files/7977044/Timeverdier.xlsx)

[image]

ArnieO, 2022-02-01:
So it is clearly a scaling. Here is a comparison Excel / GUI:
[image]

gskjold, 2022-02-01:
The data is exactly the same as on my Aidon, so they have not done anything to the scaling. Also, the scaling provided in the data only supports power of 10.

Curious, is it OK for the grid company to provide incorrect values on the HAN port?

tronde-ams, 2022-02-01:
We can see from the meters name plate that this is intended for use with a current transformer. The maximum current is stated as 6A ( or is it 5?) which makes no sense in a normal installation. A meter for dirrect connection would have read more like 0,25-5(100)A.  All energy meters are specified like this, so this not specific for Aidon meters.

Aidon also says in the installation manual floating around on the net that 6550 is for use with a current transformer.

[image]

My understanding is that "rated current" will take the place of "basic current" if used with a current transformer.

The picture of the meters display shows a total energy use of only 1785 kWh from new, so there is clearly some scaling factor involved here.

I don't know if the grid company can tell the meter which scaling factor to use, or if they must rely on a scaling in their back end system. Meters for current transformers can use almost all kind of scaling factors, so I guess the meter itself will only show and publish what it actually measures after the current transformer.

Lyse should know, and tell how to handle this,

gskjold, 2022-03-18:
I have added support for multipliers in latest beta firmware posted in #247 . You will find these in Config -> Meter -> Multipliers