UtilitechAS/amsreader-firmware #225: Timestamp and List type,and  Cumulative active import as part of  Json Structure for MQTT

hjmNav, 2022-02-03:
When i try to parse/read the Json data from my MQTT topic, i find it hard to parse and use the data the way I expect
In the end i want to be able to show in realtime Cumulative active import, and then agregate them on an hourly and dayly basis.
There are several reasons for this
1. I want to calculate (based on the price of the power) how large the electical is so far this month and display that in grafana or something similar
2. I want to know (in real time) if my power usage is (looking to be) higher than last hour . Again I want to show a graph in grafana or similar my power consumtion (cumulative) and be able to use a lower granularity than hour.

Idially i would be able to read the actual value from the meter and do a calculation , but it seems like the meter value is not exposed in Json or in the data from AMS (according to speck)

It is posible to solve some of these requirements but it would be easier if the folowing features could be imlemented in the Json body

* Timestamp - It is posible that the field t is to be used , but for me t is allways 0. I was expecting a timestamp i can use to do my own calculations based on time.
* List type - If list type could be added to the json body it would simplify the code and do e.g routing or parsing based on one field, instead of looking for a field only existing in e.g List 3 etc. 
* if posible it would also be great if Cumulative hourly active import energy could be sent for each List1 (or give me insight to calculate this myself).

ArnieO, 2022-02-03:
The meter only reports cumulative energy consumption once per hour, you will then receive MQTT "List 3" according to https://github.com/gskjold/AmsToMqttBridge/wiki/Message-formats
In between you will either receive List 1 or List 2, depending on your meter.
The MQTT payloads only forwards the content of each payload arriving from the meter.

However, you will se that the GUI shows "real time" continuous accumulated energy by continuously adding power*"delta time" to the latest accumulated value from the meter, meaning that this reading is "live":
[image]

You can access a JSON version of the data in the GUI by `http://<ams_hostname>/data.json`
Estimated accumulated Import is "ic" as per https://github.com/gskjold/AmsToMqttBridge/wiki/Message-formats
This value is updating continuously as explained above.
[image]

You have two options to obtain what you need: Either read http://<ams_hostname>/data.json using a HTTP GET command or do the estimate yourself as indicated.

henriklund, 2022-02-03:
> In between you will either receive List 1 or List 2, depending on your meter

Then I will count my blessings as I receive the List 3 every 10 seconds (please do not change this!!). :) I rely heavily on the List 3 **tPI** to calculate daily / month / yearly cost based on actual pricing (from supplier and NordPool), power consumption etc.

ArnieO, 2022-02-04:
> I receive the List 3 every 10 seconds (please do not change this!!)

Lucky guy. This is not controlled by this firmware, it just forwards what the meter sends.

@everyoneInDenmark: Is this the general experience with payload from meters in Denmark?

gskjold, 2022-02-04:
Denmark has only one list and it contains everything. Lucky indeed.

hjmNav, 2022-02-04:
I only receive List3 one time in one hour, and thats why i have my issue. 
It would be dead simple to calculate if List 3 was sent more often.
@bardahlm i do not agree that "You're not really interested in how many Wh you have used in the current hour, you want to know what you end up at the end of the hour."
My primary goal is to analyze the current power usage (and the price impact) så i can (in theory) say : The shower you just took cost 10nok.  In short the List3 that gives me the meter (tPI value) is needed more often than once an hour. 
Seems like that is not posible in norway  :-(
it seems however that i can use "ic"value from data.json and use that field to have a lower granularity.
I have to call the Data.json endpoint e.g ones every minute but i guess that is just fine.
Note: I have not verified that the IC field can be used that way..It`s just a theory

bardahlm, 2022-02-04:
@hjmNav For that usecase you need to isolate the water heater from other consumers.

This is my water heater the last 48 hours:

[image]

Data comes from a separate meter on the water heater.

gskjold, 2022-02-04:
> Edit..After checking, it looks like the ic field is set to 0 each time i enter the UI, which makes the .data endpoint "not sutable" for getting "real time" usage within the hour.

When this happens, check the uptime value in the top left corner, did the device reboot? It certainly sounds like it.

Regardless, you are better of calculating the value yourself.

hjmNav, 2022-02-04:
@gskjold you are absolutly right. Seems like the device is rebooting quite a lot.  Uptime never goes above a fiew minutes.
Maybe it does not get enough power from the han port? 
[image]
As you can see the esp is read, and it changes from red to yellow to green a lot..
did not see any errors in the UI, so i just thought it struggled somehow.
any way I can see what`s causing the reboot?

hjmNav, 2022-02-04:
closing the issue, cause the original ticket is not an improvement.

closed 2022-02-04 by hjmNav

ArnieO, 2022-02-07:
> Seems like the device is rebooting quite a lot. Uptime never goes above a fiew minutes.
Maybe it does not get enough power from the han port?

Could be many reasons, from current draw to wiring. The power pin of the HAN port is specified to maximum 75 mA at 4,15V.