UtilitechAS/amsreader-firmware #228: 2.0.8 crash

tronde-ams, 2022-02-06:
Total crash at startup.
No MQTT on this.
Static IP

Updated first via WEB GUI  --> crash.
Flashed back to 220131.1 with USB -- OK
Flashed 2.0.8 with USB -- > crash.
Flashed back to 220131.1 again with USB -- OK

Data er gøy...

```
(20:06:21.032) (I) (loop)(C1) Successfully connected to WiFi!
(20:06:21.032) <ESC>[0m(I) (loop)(C1) IP:  192.168.1.40
(20:06:21.032) <ESC>[0m(I) (loop)(C1) GW:  192.168.1.1
(20:06:21.032) <ESC>[0m(I) (loop)(C1) DNS: 8.8.8.8
(20:06:21.032) <ESC>[0m(D) (loop)(C1) mDNS is enabled, using host: ams-7d80
(20:06:21.094) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(20:06:21.094) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(20:06:23.941) <ESC>[0m(D) (AmsDataStorage) Last day update: 1644170412
(20:06:23.941) <ESC>[0m(D) (AmsDataStorage) Last month update: 1644102010
(20:06:23.957) <ESC>[0m(I) (EntsoeApi) Setting midnight millis 14024000
(20:06:26.031) <ESC>[0m(I) (EntsoeApi) Fetching prices for today
(20:06:26.031) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202202052300&p<ESC>[0meriodEnd=202202062300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(20:06:26.063) <ESC>[0m(D) (EntsoeApi)Connection established
(20:06:26.063) <ESC>[0mGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
(20:06:27.406) 
(20:06:27.406) Core  1 register dump:
(20:06:27.406) PC      : 0x400f996a  PS      : 0x00060c30  A0      : 0x800fa996  A1      : 0x3ffb2190  
(20:06:27.406) A2      : 0x3ffb94c4  A3      : 0x3ffb21fc  A4      : 0x00000000  A5      : 0x62000e25  
(20:06:27.406) A6      : 0x00000000  A7      : 0x003fffff  A8      : 0x3ffb94d0  A9      : 0x3ffb2170  
(20:06:27.422) A10     : 0x3ffb21fc  A11     : 0x3f42f995  A12     : 0x00004a59  A13     : 0x00000000  
(20:06:27.437) A14     : 0x000054f8  A15     : 0x3ffc031c  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
(20:06:27.437) EXCVADDR: 0x00000000  LBEG    : 0x4008aced  LEND    : 0x4008acfd  LCOUNT  : 0xfffffff6  
(20:06:27.453) 
(20:06:27.453) 
(20:06:27.453) Backtrace:0x400f9967:0x3ffb21900x400fa993:0x3ffb21e0 0x400fab0b:0x3ffb2240 0x400dc01b:0x3ffb2260 0x400dc353:0x3ffb22a0 0x400d8256:0x3ffb2410 0x401016a1:0x3ffb2820 
(20:06:27.469) 
(20:06:27.469) 
(20:06:27.469) 
(20:06:27.469) 
(20:06:27.469) ELF file SHA256: 0000000000000000
(20:06:27.469) 
(20:06:27.469) Rebooting...
(20:06:27.469) ets Jun  8 2016 00:22:57
(20:06:27.469) 
(20:06:27.469) rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
(20:06:27.484) configsip: 0, SPIWP:0xee
(20:06:27.484) clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
(20:06:27.484) mode:DIO, clock div:2
(20:06:27.484) load:0x3fff0030,len:1284
(20:06:27.515) load:0x40078000,len:12808
(20:06:27.515) load:0x40080400,len:3032
(20:06:27.515) entry 0x400805e4
(20:06:27.983) [     5][E][esp32-hal-cpu.c:191] setCpuFrequencyMhz(): Can not switch to 240 MHz! Chip CPU frequency rated for 160MHz.
(20:06:28.093) Sensors: 3
(20:06:28.249) (I) (setup)(C1) AMS bridge started
(20:06:28.249) <ESC>[0m(I) (setup)(C1) Voltage: 0.00V
(20:06:28.264) <ESC>[0m(D) (setup)(C1) ESP32 LittleFS
(20:06:28.264) <ESC>[0m(D) (setup)(C1)  size: 1507328
(20:06:28.280) <ESC>[0m(I) -----------------------------------------------<CR>
```

gskjold, 2022-02-06:
Crap..

tronde-ams, 2022-02-06:
2.0.8 is OK if i disable ENTSO-E key.

gskjold, 2022-02-06:
Retracted version, but I cannot see any reason it should crash with ENTSO-E key, there is no difference for ENTSO-E between the two versions you have tried.

gskjold, 2022-02-06:
For the record, works perfectly fine on my ESP32

gskjold, 2022-02-06:
Question, is is still a problem if you disable all logging?

tronde-ams, 2022-02-06:
Same without logging. 
Delete API token, and 2.0.8 is OK

tronde-ams, 2022-02-06:
What about this?

`(20:06:27.983) [     5][E][esp32-hal-cpu.c:191] setCpuFrequencyMhz(): Can not switch to 240 MHz! Chip CPU frequency rated for 160MHz.`

gskjold, 2022-02-06:
Are you sure you have a real ESP32? As far as I know there is no ESP32 that is 160mhz? What kind of board do you have?

tronde-ams, 2022-02-06:
This board. No one knows whats inside anything from China. 

[image]

gskjold, 2022-02-06:
Indeed. WROOM32 should be 240mhz

tronde-ams, 2022-02-06:
I have another type as well. Same results.

This is of the "almost similar" type. See metal shield.

[image]

Boots like this, with no error for frequency.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
Sensors: 0
(I) (setup)(C1) AMS bridge started
<ESC>[0m(I) (setup)(C1) Voltage: 0.00V
<ESC>[0m(D) (setup)(C1) ESP32 LittleFS
<ESC>[0m(D) (setup)(C1)  size: 1507328
<ESC>[0m(I) -----------------------------------------------<CR>
<ESC>[0m(I) --WiFi configuration--<CR>
```
 I have been in China for business and trawled a HUGE electronics market place in Beijing for parts because "oh - but we can't find this in China". I have very little hair left on top...

You can buy a module today that is OK. Tomorrow you can get something completely different that looks exactly the same.

tronde-ams, 2022-02-07:
It seems like it does exist 160 MHz parts too, but that they are most likely not found in general use.
https://esp32.com/viewtopic.php?t=12307

This thread links to my first module which I also got from banggood.
https://www.reddit.com/r/esp32/comments/d65pw9/my_new_esp32_can_only_run_at_160mhz/

Then we know it does exist at least two different dev-boards with this problem.

It does not explain why my other module which boots without an error message for frequency also crashes, but it gives some ideas why other people also faced problems with Entso-e.

I understand it is possible to compile for 160 MHz, but I have no idea how to. It is almost impossible to know what we get when we buy these modules on the net, so maybe it is an idea to check out compiling for 160?

I checked the (believed to be) official Espressif store on aliexpress, but it is closed until end of march. It is probably the place to buy in small quantities and get what you expect.

https://www.aliexpress.com/store/912517182

gskjold, 2022-02-07:
Try this and check log
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8017287/esp32.zip)

gskjold, 2022-02-07:
And a 160mhz build for good measure: 
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8017426/esp32.zip)

tronde-ams, 2022-02-07:
I will answer this in two parts. 
This is for the first zip. (220207.1).

I flashed both with USB.
My first module (the one winch gives error for frequency) was running 220131.1 and flashed OK and also seems to be working OK.

Log says 3 sensors, but I have none now. I had thre for testing earlier, so I guess the addresses are saved?

Boot log for this unit:
```
(18:51:03.265) 
(18:51:03.265) rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
(18:51:03.265) configsip: 0, SPIWP:0xee
(18:51:03.281) clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
(18:51:03.281) mode:DIO, clock div:2
(18:51:03.281) load:0x3fff0030,len:1284
(18:51:03.281) load:0x40078000,len:12808
(18:51:03.281) load:0x40080400,len:3032
(18:51:03.297) entry 0x400805e4
(18:51:03.767) [     5][E][esp32-hal-cpu.c:191] setCpuFrequencyMhz(): Can not switch to 240 MHz! Chip CPU frequency rated for 160MHz.
(18:51:03.876) Sensors: 3
(18:51:04.032) (I) (setup)(C1) AMS bridge started
(18:51:04.032) <ESC>[0m(I) (setup)(C1) Voltage: 0.00V
(18:51:04.047) <ESC>[0m(D) (setup)(C1) ESP32 LittleFS
(18:51:04.047) <ESC>[0m(D) (setup)(C1)  size: 1507328
(18:51:04.063) <ESC>[0m(I) -----------------------------------------------<CR>
(18:51:04.079) <ESC>[0m(I) --WiFi configuration--<CR>
(18:51:04.079) <ESC>[0m(I) SSID:                 '"MY SSID"<CR>
(18:51:04.094) <ESC>[0m(I) Psk:                  '"MY PSK"'<CR>
(18:51:04.094) <ESC>[0m(I) IP:                   '192.168.1.40'<CR>
(18:51:04.110) <ESC>[0m(I) Gateway:              '192.168.1.1'<CR>
(18:51:04.110) <ESC>[0m(I) Subnet:               '255.255.255.0'<CR>
(18:51:04.110) <ESC>[0m(I) DNS1:                 '8.8.8.8'<CR>
(18:51:04.125) <ESC>[0m(I) DNS2:                 ''<CR>
(18:51:04.125) <ESC>[0m(I) Hostname:             'ams-7d80'<CR>
(18:51:04.125) <ESC>[0m(I) mDNS:                 'Yes'<CR>
(18:51:04.125) <ESC>[0m(I) <CR>
(18:51:04.125) <ESC>[0m(I) --MQTT configuration--<CR>
(18:51:04.141) <ESC>[0m(I) Enabled:              No<CR>
(18:51:04.157) <ESC>[0m(I) <CR>
(18:51:04.157) <ESC>[0m(I) --Web configuration--<CR>
(18:51:04.172) <ESC>[0m(I) Security:             0<CR>
(18:51:04.188) <ESC>[0m(I) <CR>
(18:51:04.188) <ESC>[0m(I) --Meter configuration--<CR>
(18:51:04.203) <ESC>[0m(I) Baud:                 2400<CR>
(18:51:04.219) <ESC>[0m(I) Parity:               11<CR>
(18:51:04.219) <ESC>[0m(I) Invert serial:        No<CR>
(18:51:04.219) <ESC>[0m(I) Distribution system:  1<CR>
(18:51:04.235) <ESC>[0m(I) Main fuse:            35<CR>
(18:51:04.235) <ESC>[0m(I) Production Capacity:  0<CR>
(18:51:04.235) <ESC>[0m(I) <CR>
(18:51:04.235) <ESC>[0m(I) --GPIO configuration--<CR>
(18:51:04.235) <ESC>[0m(I) HAN pin:              16<CR>
(18:51:04.250) <ESC>[0m(I) LED pin:              255<CR>
(18:51:04.250) <ESC>[0m(I) LED inverted:         No<CR>
(18:51:04.250) <ESC>[0m(I) LED red pin:          5<CR>
(18:51:04.250) <ESC>[0m(I) LED green pin:        255<CR>
(18:51:04.250) <ESC>[0m(I) LED blue pin:         255<CR>
(18:51:04.250) <ESC>[0m(I) LED inverted:         No<CR>
(18:51:04.266) <ESC>[0m(I) AP pin:               4<CR>
(18:51:04.266) <ESC>[0m(I) Temperature pin:      15<CR>
(18:51:04.266) <ESC>[0m(I) Temp analog pin:      255<CR>
(18:51:04.266) <ESC>[0m(I) Vcc pin:              255<CR>
(18:51:04.266) <ESC>[0m(I) Vcc multiplier:       1.000000<CR>
(18:51:04.281) <ESC>[0m(I) GND resistor:         0<CR>
(18:51:04.281) <ESC>[0m(I) Vcc resistor:         25650<CR>
(18:51:04.281) <ESC>[0m(I) <CR>
(18:51:04.281) <ESC>[0m(I) --Domoticz configuration--<CR>
(18:51:04.313) <ESC>[0m(I) Enabled:              No<CR>
(18:51:04.313) <ESC>[0m(I) <CR>
[... 159 lines cut ...]
(18:51:14.596) <ESC>[0m(I) (EntsoeApi) Setting midnight millis 18533000
(18:51:16.717) <ESC>[0m(I) (EntsoeApi) Fetching prices for today
(18:51:16.733) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken=";MY TOKEN"&documentType=A44&periodStart=202202062300&p<ESC>[0meriodEnd=202202072300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(18:51:16.749) <ESC>[0m(D) (EntsoeApi)Connection established
(18:51:16.749) <ESC>[0m(D) (EntsoeApi)Receiving data
(18:51:18.192) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(18:51:18.208) <ESC>[0m(I) (EntsoeApi) Retrieving EUR to NOK conversion
(18:51:18.223) <ESC>[0m(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(18:51:18.239) <ESC>[0m(D) (EntsoeApi)Connection established
(18:51:18.239) <ESC>[0m(D) (EntsoeApi)Receiving data
(18:51:19.128) <ESC>[0m(D) (EntsoeApi)  got exchange rate 10.0070
(18:51:19.144) <ESC>[0m(D) (EntsoeApi) Resulting currency multiplier: 10.0070
(18:51:19.144) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:51:19.159) <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
(18:51:19.264) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202202072300&p<ESC>[0meriodEnd=202202082300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(18:51:19.279) <ESC>[0m(D) (EntsoeApi)Connection established
(18:51:19.295) <ESC>[0m(D) (EntsoeApi)Receiving data
(18:51:20.424) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:51:20.611) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
(18:51:22.078) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(18:51:22.187) <ESC>[0m(D) (AmsWebServer)Serving /data.json over http...
(18:51:24.407) <ESC>[0m(D) (readHanPort)(C1) Valid data, start at byte 18
```

I will now test your last zip for 160 MHz and report back.

gskjold, 2022-02-07:
Alright, then we'll try to release v2.0.9 with the same changes and see how that works out

closed 2022-02-07 by gskjold

tronde-ams, 2022-02-07:
If you don't need the speed of 240 MHz, I think it can make sense to limit it to 160 MHz. 

I can manage to find a genuine espressif unit from an industrial distributor, but most people will just buy what they find.  As this code gains traction you must either support almost all kinds of boards (i.e. not push the limits), or make a very clear statment about where to buy one that is genuine. 

Most outlets for hobbyists do not know what they sell, even if it is visually equal to the real thing. Laser engravers are cheap and widespread in the backstreets of China as is silk screen printers...

***

Another thought about instability: I know there has been some discussions about unexpected reboots. Is this limited frequency range also present in some 8266-based boards?

ArnieO, 2022-02-07:
Interesting comment, @tronde-ams. 

So far i suspect "all" unexpected reboots on ESP8266 devices to have other reasons than processor not managing to keep up (dupont wires with bad connections, inadequate charge buffering, etc). But as problems seem to be more frequent on various more or less dubious Wemos D1 clones or other development boards ... you never know.

tronde-ams, 2022-02-07:
@ArnieO 
One thing I learnt while dealing with Chinese sub contractors is that "almost similar" was quite often followed by " oh - but something went wlooong". Not so funny then, but still a useful experience.

I don't know much about software and processor architectures, but pushing limits on these kind of products can be "interesting" I think. Other problem areas are lousy USB-connectors and cables as well as noisy USB "chargers". 

I have (still) not heard about fake Espressif silicon, but I am quite certain you will find fake modules (the metal clad modules) with Espressif marking on. Espressif makes genuine for sale, but they also sell silicon chips to other manufacturers to assemble their own modules. Fake such modules can give strange results, but also fake or badly designed development boards carrying these modules can make a lot of trouble, even if they use genuine Espressif parts . And we have absolutely no idea about what we get most of the time.

tronde-ams, 2022-02-08:
One problem with these hobby grade products we buy, be it on aliexpress or as a "branded" product from hobbyist outlets in our part of the world is that they are made in low volume compared to the large brands (Apple &Co). 

We can buy electronic parts from Digikey, Mouser, Farnell and many other distributors and be sure they are genuine parts. Such distributors are not common in China, at least not for low volume. They have to search their network or buy from huge marketplaces like the one I visited in Beijing. 

By doing so, they have no control about the real delivery chain, and can get both genuine parts as well as really bad fakes or what is known as "good quality" parts. These parts are not sold as genuine, but they are made in China as a replica of a brand name product with the intenion of being perfectly OK for most applications.

Another problem for Chinese manufacturers is that the goverment prefer the use of Chinese made parts. Getting western parts through customs can be a real pain, and this leads to a lot of "sharing" between companies when they buy parts from abroad. "You help me to get xxx thousand of this part, and I help you with that part" is rather normal. Again we loose control over the delivery chain.

Here is video about a market place for industrial parts
https://www.youtube.com/watch?v=_ziP_4pouN8

It's exactly the same for electronic parts. Completly weird for a westerner, but a part of their daily life.

Yes, many Chinese companies have adapted to our western idea about win-win, and that a recurring customer is good for business, but China is a huge country in number of people, and many stick to the more traditional idea of "null-sum" where the idea is that one win, and one lose, and that the Chinese is not the loser.

In addition Chinese people are world champions in making a living out of razor-thin margins. This allows for a lot of "creativity" and also cheating.

China is an extremely facinating experience, but we are completly lost most of the time when something goes wrong. We understand absolutely nothing about what they say or write, and only a few of them can make much sense of English. If you walk just outside the main street in a rather big town (100 000 ++) people, you can find they don't even use arabic numerals. They have a local version more like the roman numerals, but if course not similar. Quite fun...

It is often said as a joke that when a chinese say "yes", he can also mean "no" or "maybe". This is no joke, but a result of how they use their own langage. They can say yes to confirm they have understood what you asked about, not as an answer to your question. "Maybe" can be a way for them to buy some time, especially if they don't want to do what you asked for. "Face value" is most important for them, and if we don't manage to decipher what they say or do, we head for trouble.

Your product is techically quite simple, and will probaly not catch fire, so even if something should go wrong, you will not face a disaster. For anything meant for mains voltage, I would stay away from China if you don't have a local and trustwothy partner.

Good (and quite fun) reading is "Mr China" by Tim Clissold and "Det nye Kina" by Henning Kristoffersen.

A general understanding of Chinese history is also most useful. They have a very long and winding history, both good and bad, that has made them what they are today.

ArnieO, 2022-02-08:
> It is often said as a joke that when a chinese say "yes", he can also mean "no" or "maybe". This is no joke, but a result of how they use their own langage. They can say yes to confirm they have understood what you asked about, not as an answer to your question.

This sounds similar to Japanese. Many years ago I had a potentially difficult meeting with a Japanese (the European representative for the particular product we bought from one of the LARGE Japanese groups) where I had a list of complaints for a delivery. I was concerned that he might disagree, and was well prepared to the meeting. When I came to the list of complaints - where I expected protests - the Japanese guy nodded politely for each point I made and said "Yes". Point by point. I was more and more relieved, as I thought he agreed with me. Well... when I had finished my presentation of complaints, he started speaking - and disagreed thoroughly to most of my points. It was not until later I learned that he did not really say Yes - he said "Hai", which is the Japanese word for Yes, but is also used to signal "I understand what you say, please continue". So ... it was quite a learning experience for me!

> Your product is technically quite simple, and will probably not catch fire, so even if something should go wrong, you will not face a disaster. For anything meant for mains voltage, I would stay away from China if you don't have a local and trustworthy partner.

I totally agree. Chinese products used on mains voltage is not something I buy from Aliexpress or similar - but from a local shop that has to take responsibility for what they buy.

Again - thank you for sharing!