UtilitechAS/amsreader-firmware #229: Pow-K crashes after upgrade to v2.0.9

henriklund, 2022-02-07:
**Describe the bug**
Upgrading to v2.0.9 causes repeated crash of Pow-K

**To Reproduce**
Steps to reproduce the behavior:

1. Monitor MQTT flow
2. Upgrade (from v2.0.7) to v2.0.9
3. After reboot software will start to reboot after around 45secs.

**Expected behavior**
No unexpected reboots

Hardware information:
- Meter: Kamstrup (DK)
- AMS reader: Pow-K
- M-bus adapter (if applicable): N/A

Relevant firmware information:
- v2.0.7
- MQTT: yes
- HAN GPIO: GPIO3
- HAN baud and parity: 2400 8N1
- Temperature sensors: N/A

ArnieO, 2022-02-07:
Pow-K, GPIO12.
No reboot, but frequent unexpected "red" HAN.
[image]

henriklund, 2022-02-07:
Stable 30mins after revert. Data also looks OK
[image]

closed 2022-02-07 by gskjold

tronde-ams, 2022-02-07:
Same for ESP32

Behaves the same way as 2.0.8

This is the one previously giving error for frequency. I understand frequency for ESP32 is 160 MHz now.

```
(22:04:53.224) <ESC>[0m(D) (EntsoeApi)Connection established
(22:04:53.224) <ESC>[0mGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
(22:04:54.582) 
(22:04:54.582) Core  1 register dump:
(22:04:54.582) PC      : 0x400f9eb6  PS      : 0x00060c30  A0      : 0x800faeda  A1      : 0x3ffb2190  
(22:04:54.597) A2      : 0x3ffb94c4  A3      : 0x3ffb21fc  A4      : 0x00000000  A5      : 0x62017b68  
(22:04:54.597) A6      : 0x00000000  A7      : 0x003fffff  A8      : 0x3ffb94d0  A9      : 0x3ffb2170  
(22:04:54.597) A10     : 0x3ffb21fc  A11     : 0x3f42fc85  A12     : 0x00004a5a  A13     : 0x00000001  
(22:04:54.613) A14     : 0x000054f8  A15     : 0x3ffc031c  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
(22:04:54.628) EXCVADDR: 0x00000000  LBEG    : 0x4008aced  LEND    : 0x4008acfd  LCOUNT  : 0xfffffff6  
(22:04:54.628) 
(22:04:54.628) 
(22:04:54.628) Backtrace:0x400f9eb3:0x3ffb21900x400faed7:0x3ffb21e0 0x400fb04f:0x3ffb2240 0x400dc54b:0x3ffb2260 0x400dc883:0x3ffb22a0 0x400d82d6:0x3ffb2410 0x40101be5:0x3ffb2820 
(22:04:54.644) 
(22:04:54.644) 
(22:04:54.644) 
(22:04:54.644) 
(22:04:54.644) ELF file SHA256: 0000000000000000
(22:04:54.644) 
(22:04:54.644) Rebooting...
```

This is the other unit

```
(22:04:55.190) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken=2d23d8ba-a990-4311-bcff-ad03c981c56a&documentType=A44&periodStart=202202062300&p<ESC>[0meriodEnd=202202072300&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(22:04:55.206) <ESC>[0m(D) (EntsoeApi)Connection established
(22:04:55.206) <ESC>[0mGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
(22:04:56.578) 
(22:04:56.578) Core  1 register dump:
(22:04:56.578) PC      : 0x400f9eb6  PS      : 0x00060a30  A0      : 0x800faeda  A1      : 0x3ffb2190  
(22:04:56.578) A2      : 0x3ffb9438  A3      : 0x3ffb21fc  A4      : 0x00000000  A5      : 0x62017b6a  
(22:04:56.594) A6      : 0x00000000  A7      : 0x003fffff  A8      : 0x3ffb9444  A9      : 0x3ffb2170  
(22:04:56.594) A10     : 0x3ffb21fc  A11     : 0x3f42fc85  A12     : 0x00004a5a  A13     : 0x00000001  
(22:04:56.610) A14     : 0x000054f8  A15     : 0x3ffc031c  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c  
(22:04:56.610) EXCVADDR: 0x00000000  LBEG    : 0x4008aced  LEND    : 0x4008acfd  LCOUNT  : 0xfffffff6  
(22:04:56.625) 
(22:04:56.625) 
(22:04:56.625) Backtrace:0x400f9eb3:0x3ffb21900x400faed7:0x3ffb21e0 0x400fb04f:0x3ffb2240 0x400dc54b:0x3ffb2260 0x400dc883:0x3ffb22a0 0x400d82d6:0x3ffb2410 0x40101be5:0x3ffb2820 
(22:04:56.641) 
(22:04:56.641) 
(22:04:56.641) 
(22:04:56.641) 
(22:04:56.641) ELF file SHA256: 0000000000000000
(22:04:56.641) 
(22:04:56.641) Rebooting...

```

Other interesting thing I experienced.
It is impossible to flash older version into unit with 2.0.9 from USB. Flasher says hash is OK, but it does still start up as 2.0.9.

After some trial and error, I managed to delete the API-key from WEB GUI, but stil no USB flash. BUT - if I try to flash with firmware menu, it accepts older version and everything is OK again when I reactivate ENTSO-E key. 

I will stick to 220207.2 for a while.

ArnieO, 2022-02-07:
FYI: 
I reverted to 1.5.8: No problem.
Then 2.0.6: No problem. 
Then 2.0.7 again: Now no problem.after 3 min
Then 2.0.9: Checksum errors etc. immediately

tronde-ams, 2022-02-07:
@gskjold 
Next time you compile a version, whenever it be, consider adding the FW version in the log.