UtilitechAS/amsreader-firmware #231: Calling for testing

gskjold, 2022-02-08:
I couldn't leave the issues with 2.0.8 and 2.0.9 alone after all.. Finally able to reproduce the reboot issue, and I think I have identified the problem. ESP8266 have limited stack space, so I have made some adjustments to free some. Attached are binaries with the changes. May or may not make a difference for ESP32.

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8024584/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8024585/esp8266.zip)

ArnieO, 2022-02-08:
Pow-K with ESP8266 / UART0: 
Looks fine, no "red HAN" or other issue seen after 5 minutes uptime.
(I did not se reboot with 2.0.9, but a lot of HAN read errors.)

gskjold, 2022-02-08:
Thanks, get back to me tomorrow morning and let me know how stable it is.

gvenaas, 2022-02-08:
Lolin D32 Seems OK
No restart with debug on level Verbose (186KB Free)

ujhede, 2022-02-08:
When upgrading is it enought to just upload firmware.bin or will i need to do a full reflash with both partititions and firmware ?

ArnieO, 2022-02-08:
I have not seen "red HAN" but it seems to reboot from time to time:
[image]

gskjold, 2022-02-08:
> When upgrading is it enought to just upload firmware.bin or will i need to do a full reflash with both partititions and firmware ?

Upload firmware.bin via web is enough

> I have not seen "red HAN" but it seems to reboot from time to time. Free mem still around 20 kb (but shortly after reboot). I will try to keep an eye on it.

With or without mqtt? What payload?

hanseinar, 2022-02-08:
2.0.9 on my ESP32 was just a mess. Hardly boot and/or reboot all the time, and after a couple of minutes no response at all. 22.02.08.1 stable and functioning for over 2h. All green. See tomorrow, but it seems OK.

tronde-ams, 2022-02-08:
220208.1 ESP32 seems to be OK

I performed a factory reset before flashing because the statistics went haywire when I tried 2.0.9 and a failed flash then.
I am out of USB chargers so this must be fed from my computer. Sometimes it will disable the USB port. I will log until tomorrow.

ArnieO, 2022-02-08:
> With or without mqtt? What payload?

Without MQTT. Payload?

Lommelund, 2022-02-09:
8 hours and still up.
ESP8266 with MQTT. USB power. Debugging off.

bardahlm, 2022-02-09:
POW-K up for 14-15 hours now. MQTT (json). Free mem: 15.9kb. My uptime-record for the 2.x version is 4 days so have to wait some days to see if this is more stable :D

hanseinar, 2022-02-09:
ESP32 12h, all green, no reboot. MQTT. Raw minimal. Custom port. Debugging off.

gvenaas, 2022-02-09:
Device   : Lolin D32
Version  : 220208.1
Meter    : Kamstrup
MQTT     : No
Mem Free : 186Kb
Debug    : No
Up for   : 12h
Reeboots : No

Other    : Seen some red HAN signals, And total for the last day is 0

ArnieO, 2022-02-09:
Rebooted 3 hours ago

tronde-ams, 2022-02-09:
Mine has been up for 17 hours by now. Free mem: 187.7kb
ESP32 generic. No MQTT but logging active.

Started with factory reset, so no daily statistics yet. Hourly is the same as one 220207.2 and one 220124.2 which has been up 14 days. The last one is fed from USB-charger.

gvenaas, 2022-02-09:
Ref my message above.

Still up without restart. Going for 20+ hours now. 186Kb free mem.

rungrona, 2022-02-09:
Installed 220208.1 on Pow-U with Aidon meter with MQTT  - max uptime was approx. 7000 seconds - will continue to monitor uptime and voltage.

tronde-ams, 2022-02-10:
My 220208.1 ESP32 was running flawlessly for about 38 hours.

Updated to 2.0.10, and back to square one. 

This is the module that did not give error for frequency.

```
(13:29:13.036) 
(13:29:13.036) Rebooting...
(13:29:13.036) ets Jun  8 2016 00:22:57
(13:29:13.036) 
(13:29:13.036) rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
(13:29:13.036) configsip: 0, SPIWP:0xee
(13:29:13.036) clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
(13:29:13.046) mode:DIO, clock div:2
(13:29:13.046) load:0x3fff0030,len:1284
(13:29:13.046) load:0x40078000,len:12808
(13:29:13.056) load:0x40080400,len:3032
(13:29:13.056) entry 0x400805e4
(13:29:13.656) Sensors: 0
(13:29:13.816) (I) (setup)(C1) AMS bridge started
(13:29:13.816) <ESC>[0m(I) (setup)(C1) Voltage: 0.00V
(13:29:13.826) <ESC>[0m(D) (setup)(C1) ESP32 LittleFS
(13:29:13.846) <ESC>[0m(D) (setup)(C1)  size: 1507328
(13:29:13.846) <ESC>[0m(I) -----------------------------------------------<CR>
(13:29:13.846) <ESC>[0m(I) --WiFi configuration--<CR>
(13:29:13.856) <ESC>[0m(I) SSID:                 '"MY SSID"'<CR>
(13:29:13.866) <ESC>[0m(I) Psk:                  '"MY PSK'<CR>
(13:29:13.866) <ESC>[0m(I) IP:                   '192.168.1.42'<CR>
(13:29:13.866) <ESC>[0m(I) Gateway:              '192.168.1.1'<CR>
(13:29:13.866) <ESC>[0m(I) Subnet:               '255.255.255.0'<CR>
(13:29:13.876) <ESC>[0m(I) DNS1:                 '8.8.8.8'<CR>
(13:29:13.876) <ESC>[0m(I) DNS2:                 ''<CR>
(13:29:13.876) <ESC>[0m(I) Hostname:             'ams-e378'<CR>
(13:29:13.896) <ESC>[0m(I) mDNS:                 'Yes'<CR>
(13:29:13.896) <ESC>[0m(I) <CR>
(13:29:13.896) <ESC>[0m(I) --MQTT configuration--<CR>
(13:29:13.916) <ESC>[0m(I) Enabled:              No<CR>
(13:29:13.926) <ESC>[0m(I) <CR>
(13:29:13.926) <ESC>[0m(I) --Web configuration--<CR>
(13:29:13.946) <ESC>[0m(I) Security:             0<CR>
(13:29:13.956) <ESC>[0m(I) <CR>
(13:29:13.956) <ESC>[0m(I) --Meter configuration--<CR>
(13:29:13.966) <ESC>[0m(I) Baud:                 2400<CR>
(13:29:13.986) <ESC>[0m(I) Parity:               11<CR>
(13:29:13.986) <ESC>[0m(I) Invert serial:        No<CR>
(13:29:13.986) <ESC>[0m(I) Distribution system:  1<CR>
(13:29:13.986) <ESC>[0m(I) Main fuse:            35<CR>
(13:29:13.996) <ESC>[0m(I) Production Capacity:  0<CR>
(13:29:13.996) <ESC>[0m(I) <CR>
(13:29:13.996) <ESC>[0m(I) --GPIO configuration--<CR>
(13:29:14.006) <ESC>[0m(I) HAN pin:              16<CR>
(13:29:14.026) <ESC>[0m(I) LED pin:              2<CR>
(13:29:14.026) <ESC>[0m(I) LED inverted:         No<CR>
(13:29:14.026) <ESC>[0m(I) LED red pin:          255<CR>
(13:29:14.026) <ESC>[0m(I) LED green pin:        255<CR>
(13:29:14.036) <ESC>[0m(I) LED blue pin:         255<CR>
(13:29:14.036) <ESC>[0m(I) LED inverted:         Yes<CR>
(13:29:14.036) <ESC>[0m(I) AP pin:               255<CR>
(13:29:14.036) <ESC>[0m(I) Temperature pin:      255<CR>
(13:29:14.036) <ESC>[0m(I) Temp analog pin:      255<CR>
(13:29:14.046) <ESC>[0m(I) Vcc pin:              255<CR>
(13:29:14.046) <ESC>[0m(I) Vcc multiplier:       1.000000<CR>
(13:29:14.046) <ESC>[0m(I) GND resistor:         0<CR>
(13:29:14.056) <ESC>[0m(I) Vcc resistor:         0<CR>
(13:29:14.066) <ESC>[0m(I) <CR>
(13:29:14.066) <ESC>[0m(I) --Domoticz configuration--<CR>
(13:29:14.086) <ESC>[0m(I) Enabled:              No<CR>
(13:29:14.086) <ESC>[0m(I) <CR>
(13:29:14.086) <ESC>[0m(I) --NTP configuration--<CR>
[... 37 lines cut ...]
(13:29:24.906) EXCVADDR: 0x00000000  LBEG    : 0x4008aced  LEND    : 0x4008acfd  LCOUNT  : 0xfffffff6  
(13:29:24.916) 
(13:29:24.916) 
(13:29:24.916) Backtrace:0x400f94c3:0x3ffb22a00x400fa4e7:0x3ffb22f0 0x400fa65f:0x3ffb2350 0x400dbe2b:0x3ffb2370 0x400dc12b:0x3ffb23b0 0x400d817e:0x3ffb2410 0x401011f5:0x3ffb2820 
(13:29:24.926) 
(13:29:24.926) 
(13:29:24.926) 
(13:29:24.926) 
(13:29:24.926) ELF file SHA256: 0000000000000000
```

Whats going on here?
```
(13:29:23.606) <ESC>[0m(D) (EntsoeApi)Connection established
(13:29:23.606) <ESC>[0mGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
(13:29:24.866) 
```

Why does it go haywire because of ENTSO-E?

It seems to be something wery wrong with the exeption handling in this part of the code?

Is it possible to add more details in the log for just this part of the code?

When I delete the API token, everything is fine.

tronde-ams, 2022-02-10:
This verion of ams2mqtt-esp32-2.0.10.bin seems to be OK

gskjold, 2022-02-10:
weird, but then I know where to investigate

gvenaas, 2022-02-10:
Tested on my Lolin D32, and the version in the zip file workes ok .

henriklund, 2022-02-12:
An observation of uptime since JAN 26 (v2.0.6 - v2.0.10). Prior to v2.0.10 best case up time was 15 hour (one occassion), avg. was around 3-5 hours. With v2.0.10 there has been one(!); 15 hrs before first restart and the clock is currently at 33 hrs and ticking. Whatever you have done @gskjold, it seems to have added stability (at least in my case).

tronde-ams, 2022-02-12:
One ESP32 has ben running since you released the zip-file (56 h) and one for 51 h since it was updated. Both seems to be OK.

gskjold, 2022-02-12:
Excellent! So good to hear that. Chasing ghosts is exhausting..

gskjold, 2022-02-12:
Haven't looked into it yet. But I guess something is wrong with the configuration of the build environment on GitHub Actions.

ArnieO, 2022-02-14:
Mine seems to restart around once per day. 
Except for that, it runs smoothly!
[image]

henriklund, 2022-02-19:
@gskjold, uptime is still stable at an unprecedented level. So far 1x15 hours and then 2x4 days since updating to 2.0.10. Strangely enough each reboot has occurred around 07:00:03 (+/- 1 sec).

gskjold, 2022-02-19:
Cool! But yes, that is really strange. Wonder what that could be. I don't think I have anything scheduled for that time in the code..

ArnieO, 2022-02-19:
Mine seems to reboot approx once per day - and has just rebooted at 10:33.
Old Pow-K, ESP8266, UART0.

henriklund, 2022-02-19:
@gskjold, the odd part is that the first reboot (after 15hrs) started this 'trend'. It might be that the frequent reboots in the past obscured this behavior (I note reboots around the same time in the past, however, not with this kind of accuracy). Don't get me wrong,I am more than happy to stay with an almost regular reboot every fourth day. However, the regularity has made me wonder about whether this could be some kind of regular action (most plausible by the meter).

gvenaas, 2022-02-19:
Another "trend"
[image]

Device   : Lolin D32
Version  : 2.0.10
Meter    : Kamstrup
MQTT     : No
Mem Free : 186Kb
Debug    : No
Up for   : 8days
Reeboots : One, due to mains loss

tronde-ams, 2022-02-19:
One of my ESP32 is still OK after about 9 days (786700++ sec) and free mem: 189.4kb. My other unit rebooted two days ago, but I have a feeling it had to to with my computer disconnecting power to the USB port for a short period of time. It happens to do so sometimes. The long-runner is fed from a hub with external power while still connected to the same computer.