UtilitechAS/amsreader-firmware #234: Device dead after upgrade to v.2.0.10

paaland, 2022-02-10:
**Describe your problem**
I have a universal model from amsleser.no and tried to upgrade to 2.0.10 via ams2mqtt-esp8266-2.0.10.bin. Connected device to USB cable and power. Uploaded file via wifi and got the "reboot-wait" screen. 

The device never came online again. When pressing reset button I can see the blue power led on the esp-12 flash briefly, that's the only sign of life. Pressing down AP key for 3-5 seconds does nothing. Status led is constant blue and power led is constant green.

Upgraded from v2.0.7.

Do I need to purchase a FTDI device to flash again? Will that even help?

ArnieO, 2022-02-10:
Hi Pål, 

Something unusual happened during the upgrade, your device seems to be bricked. 

Reprogramming over FTDI will surely cure it - unless there is a major failure on the board.
What happened to you is unusual, I believe you are the second or third among currently around 200 Pow-U users.

**_Kind regards,
Egil Opsahl
amsleser.no_**

gskjold, 2022-02-10:
Bummer.. I use Pow-U myself, so it has already been tested on that model. A flash with FTDI should fix it, and if it doesn't, do a full wipe before flashing again. Consult wiki for details

closed 2022-02-20 by gskjold