UtilitechAS/amsreader-firmware #235: han connection issues

knutmyr, 2022-02-12:
Hi
Struggling to get this to work. I have tried an esp32 and 8266 without any luck. I'm using this https://www.aliexpress.com/item/33009939864.html?spm=a2g0o.9042311.0.0.27424c4dLlBe5Z adapter. I connected TX from adapter to GPIO16. Debug log don't give me much to work with. While booting HAN is green, then yellow but stop and stay at red... I've tried most baud rate and port setting combinations. Suspect the uart board is not working, gonna order the one in wiki pages and hope that solves it. If not anybody has any good ideas?

knutmyr, 2022-02-12:
(D) (loop)(C1) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop)(C1) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop)(C1) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /gpio.html over http...
(D) (AmsWebServer)Handling save method from http
(D) (AmsWebServer)Received GPIO config
(I) (AmsWebServer)Saving configuration now...
(I) (AmsWebServer)Successfully saved.
(I) (setupHanPort) Setting up HAN on pin 16 with baud 2400 and parity 11
(D) (setupHanPort)(C1) Hardware serial
(D) (AmsWebServer)Serving /index.html over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /dayplot.json over http...
(D) (AmsWebServer)Serving /monthplot.json over http...
(D) (loop)(C1) Used 0 ms to update temperature

tronde-ams, 2022-02-12:
Do you have the big one or the small one?

Both seems to have optocoupler outputs. How do you connect your module to the ESP? Optocouplers are normally floating with regard to power supply, so it can be you must connect more than TX.

If possible, show a detailed picture of your adapter and your ESP. I am quite certain this is easy to fix, but we need to be sure about your hardware.

knutmyr, 2022-02-12:
[image]
[image]
[image]
[image]
[image]
This is how it's connected right now, just read somewhere uart2 was reserved. I tested esp TX to an rpi3, and that worked just fine. I also tested this board on the rpi, didn't get any reading from it there either. Tested with od command and minicom

tronde-ams, 2022-02-12:
I have the same ESP32. 

RX2 is GPIO16. Try connecting TX from HAN to RX2 and leave RX from HAN unconnected.

If thhis fails, try to switch RX and TX from HAN. Sometimes the Chinese mix up these two. 
Vin to 3V3 as it seems like you have is correect, and GND to GND is also correct.

My board is configured as generic ESP32. I have not tried the other config options.

knutmyr, 2022-02-12:
Tried connecting both ways again, still red HAN. Leaves me to think there is 2 possible options: 1 bad mubs ttl adapter or 2 not opened HAN port. But cant tell since i dont know what to expect from debug and controller

knutmyr, 2022-02-13:
I have an Landis gyr with embriq interface. Will go to someone which has an open port working on tibber and see if there is a diff. Had dame issue with USB mbus adapter on my rpi too

tronde-ams, 2022-02-13:
You are not in Norway? Or is it a privately owned meter?

Please give more detailed information about the type of your meter and your grid company.

We know that some foreign meter must have some kind of hard-wiring on the HAN-port before they send data, even if the port is open. You should be able to see that data is available with a voltmeter if the port is active. 

We also know that privately owned meters usually don't work with this ESP-adapter because they use a different data format. MBUS is just a hardware standard.

knutmyr, 2022-02-13:
I'm in Norway,  just figured ut was better to do this in English if others have issues.
[image]
This is my meter. Going to a friend's soon to test there, he has a working tibber connected

bardahlm, 2022-02-13:
https://support.tibber.com/nb/articles/5688861-hvilke-strommalere-fungerer-pulse-pa has some info about that meter

knutmyr, 2022-02-13:
Final update... my friend has same meter. And I can confirm it works. I just need to get this box and put in between, which should been delivered when they opened my port. 
[image]

closed 2022-02-13 by knutmyr

gskjold, 2022-02-14:
If I read chapter 3.2 correctly, the meter itself puts out a payload via serial port with IEC62056-21 format which is interpreted by the PT800 module. My software should be able to read that format directly. That being said, I do not know of anyone who use it with that format, it have just been tested in my lab.

ArnieO, 2022-02-14:
@knutmyr  
Interesting to see a Landis Gyr in Norway!
Which grid company is this?

tronde-ams, 2022-02-14:
I understand that this meter is widespread in the Netherlands and that they read it by means of a series to USB converter.

Embriq seems to be a company that handles data collection and remote control for some grid companies. 

Since the Norwegian HAN port is different from most of Europe, I guess they have modified one of the optional modules for this meter, and labeled it with their logo. A permanent +27V was mentioned. I guess the embriq module is modified to supply 27V instead of (or in addition to) 5V, and that this is used by the OP027a adapter to generate the MBUS signal.

I found two manuals for the E350 meter.

[e350_series_2.pdf](https://github.com/gskjold/AmsToMqttBridge/files/8062879/e350_series_2.pdf)
[Landis+Gyr E350 Users Manual English.pdf](https://github.com/gskjold/AmsToMqttBridge/files/8062885/Landis%2BGyr.E350.Users.Manual.English.pdf)

knutmyr, 2022-02-14:
> @knutmyr Interesting to see a Landis Gyr in Norway! Which grid company is this?

This is Klive, former known as Nord-Østerdal kraftlag.

ArnieO, 2022-02-15:
Apparently, exploitation of the HAN port on those meters requires on-site intervention by an operator from the grid company - at no cost for the user. This sounds like an expensive solution for the grid company and a nuisance for the user. On the other meter types used in Norway (Kamstrup, Kaifa/Nuri, Aidon) the HAN port is opened remotely using the communication system the grid company uses to collect data from the meters.

From https://www.klive.no/han-porten/:
[image]

knutmyr, 2022-02-15:
All fixed today. It was the module they plugged into the meter. Changed it and it's up. This is the old meter they used, newer ones have this fixed so they only open port from office

ArnieO, 2022-02-15:
> All fixed today. It was the module they plugged into the meter. Changed it and it's up. This is the old meter they used, newer ones have this fixed so they only open port from office

Does this imply that the statement on Klive web page ("målermontør fra Klive må installere et kort i måleren din") is obsolete?

knutmyr, 2022-02-15:
Yes and no. Depends if you have old or new meter. They just changed all of the older ones, which didn't support hourly readout to the newest version with integrated HAN port

ArnieO, 2022-02-15:
> Yes and no. Depends if you have old or new meter. They just changed all of the older ones, which didn't support hourly readout to the newest version with integrated HAN port

So the Politerm800 module is replaced with a module with integrated HAN port? 
A photo would be greatly appreciated!

ArnieO, 2022-02-16:
@knutmyr Perfect - thank you!