UtilitechAS/amsreader-firmware #237: Option to keep select watt as unit instead of KW

Nesegreven, 2022-02-13:
**Is your feature request related to a problem? Please describe.**
If possible i think it would be nice to have the option to keep the granularity of watt as a unit.

**Describe the solution you'd like**
Keep watt as unit through option in settings and not switch to KW when usage is above 1000w

possible other
text size may need to be adjusted to not overflow onto the visual load indicator 

**Additional context**
I use it to find unnecessary standby device and other watt-thieves.

gskjold, 2022-02-20:
I will consider how to best present this option in a future release.

dknia, 2022-03-05:
I strongly support/endorse this request for exactly the same reason. 

I have recently purchased and installed the AMS, and this was more or less the first configuration item I was looking for.

closed 2022-03-12 by gskjold