UtilitechAS/amsreader-firmware #238: 3 digits after the decimal point?

SVH-Powel, 2022-02-15:
I see from the code, mqtt->publish(topic + "/meter/import/active/accumulated", String(data->getActiveImportCounter(), 3), true, 0); , that it is expected to deliver 3 digits after the decimal point. But getActiveImportCounter return a float which has a maximum precision of 7 digits. So when the counter is 12345.678 or 123456.789 , a float is not nearly enough to represent those numbers.

My meter is reporting somewhere around 60000kwh - 5 digits before the decimal point and 2 (or 3??) digits after. I have checked 2 months of data and the 3rd decimal after the decimal point is always 8, 9, 0, 1 or 2. It is never 3, 4, 5, 6 or 7. I dont think that is a coincidence. I shows that a float for storing this number creates errors.

SVH-Powel, 2022-02-18:
https://www.ibm.com/support/pages/single-precision-floating-point-accuracy

"A single-precision float only has about 7 decimal digits of precision (actually the log base 10 of 223, or about 6.92 digits of precision). The greater the integer part is, the less space is left for floating part precision."

gskjold, 2022-02-18:
Nice catch, will fix for 2.1.x series firmware

gskjold, 2022-02-19:
Test version with double instead of float:

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8101682/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8101683/esp8266.zip)

closed 2022-02-20 by gskjold