UtilitechAS/amsreader-firmware #242: Meter to TSS721 to NodeMCU not detecting anything

APayerl, 2022-02-27:
**Describe your problem**
My NodeMCU does not detect my meter or any measurements from the meter.

I have connected my TSS721 to my NodeMCU using:
NodeMCU::3v3 →TSS721::VDD
NodeMCU::GND → TSS721::GND
NodeMCU::D2 → TSS721::TXD

And I have connected TSS721 to my meter using RJ45 pin 1 to TSS721::MB_A and RJ45 pin 2 to TSS721::MB_B

Using a voltmeter I can read a constant voltage of ~24V between TSS721::MB_A and TSS721::MB_B and that coltage is dropped when a transmission is made (about every 15 seconds), but I am not able to detect to what level with my simple voltmeter... it says 18V, but I wouldn't count on it.
Between TSS721::GND and TSS721::TXD voltage is constant at 3.3V between transmissions and during a transmission (every 15 seconds) it drops to 1-1.2V (again, not to be trusted; only thing to be trusted is that it drops).

In my  configuration I have set 2400 8E1 (didn't check "Invert signal" box) and HAN: GPIO04

According to a email conversation I had with my power company I should use these settings to communicate with the meter if I'm using a MBus to USB adapter:
[image]

Can anyone give me a hint on why nothing happens in my meter?....
Since the voltage levels look okay and reacts every 15 seconds on both sides of the adapter, I guess it's not related to how I have connected the adapter, right?...

**Hardware information:**
 - Meter: Kaifa MA304H4PSE
 - AMS reader: NodeMCU (ESP8266) (https://i1.wp.com/www.teachmemicro.com/wp-content/uploads/2018/04/NodeMCUv3.0-pinout.jpg?ssl=1)
 - M-bus adapter: TSS721 (https://www.aliexpress.com/item/32751482255.html)

**Relevant firmware information:**
 - Version: 2.0.11
 - MQTT: yes
 - HAN GPIO: GPIO4

ArnieO, 2022-02-27:
TSS721 requires correct setting of a number of support components. It also has two outputs: one inverting andond non-inverting. Which one are you feeding to the GPIO, and does it correctly follow the input signal?

APayerl, 2022-02-27:
Here is a image showing my meter to TSS connection... 
[image]

And here is one showing my TSS to NodeMCU connection... (blue for ground and orange for serial and brown for 3.3V)

APayerl, 2022-02-27:
> TSS721 requires correct setting of a number of support components. It also has two outputs: one inverting and one non-inverting. Which one are you feeding to the GPIO, and does it correctly follow the input signal?

Not sure if I understood your question... my TSS only has one TXD output as far as I know...

ArnieO, 2022-02-27:
Ok, I meant the TSS721 component. You have a module that holds all components. What do you measure on the signal that goes to the NodeMCU?

tronde-ams, 2022-02-27:
`In my configuration I have set 2400 8E1 (didn't check "Invert signal" box) and HAN: GPIO04`

Have you tried to check the "invert signal" box?

It seems you have connected 3,3V from the ESP to "VIN"and GND from ESP to "GND"? If not, do so. 

The values you mesure appears to be OK.

ArnieO, 2022-02-27:
And what is the output from Telnet debug?

APayerl, 2022-02-27:
> Ok, I meant the TSS721 component. You have a module that holds all components. What do you measure on the signal that goes to the NodeMCU?

As specified in first post 3.3V when no transmission. When transferring then it drops (guess it bounces more then my voltmeter can measure, or me see)

APayerl, 2022-02-27:
> It seems you have connected 3,3V from the ESP to "VIN"and GND from ESP to "GND"? If not, do so. 
> 
Yea

tronde-ams, 2022-02-27:
I find your "dropped" values to be OK. This is because your voltmeter will measure more or less the average value of the square wave signal. I tested on one of my test-units just now, and I see the same as you.

Here is a quick scope test of the same signal. Average value 1.33V

[image]

ArnieO, 2022-02-27:
> > Ok, I meant the TSS721 component. You have a module that holds all components. What do you measure on the signal that goes to the NodeMCU?
> 
> As specified in first post 3.3V when no transmission. When transferring then it drops (guess it bounces more then my voltmeter can measure, or me see)

Ok, this looks correct. Have you tried Telnet debug to see what the NodeMCU receives?

APayerl, 2022-02-27:
I have not tested telnet debugging and one message got repeated over and over again:
(readHanPort) Unknown data format

Does this mean my meter is not yet supported in the project? Is there an easy way to implement a meter specific format?...

tronde-ams, 2022-02-27:
You wrote:
`In my configuration I have set 2400 8E1 **(didn't check "Invert signal" box)** and HAN: GPIO04`

Have you tried to check the "invert signal" box?

tronde-ams, 2022-02-27:
Do you have some kind of documetation for this module? Aliexpress does not say much.

What about the jumper marked with red?
I assume it has something to do about isolated / non-isolated output and sharing of power supply. 

[image]

tronde-ams, 2022-02-27:
It seems like your interface is OK from what I read here, but unclear about the jumper.
https://www.hjemmeautomasjon.no/forums/topic/1982-lesing-av-ams-data-amshan-iot/?do=findComment&comment=32025

APayerl, 2022-02-28:
> Do you have some kind of documetation for this module? Aliexpress does not say much.
> 
> What about the jumper marked with red?
> I assume it has something to do about isolated / non-isolated output and sharing of power supply. 
> 
> 
> [image]
> 
> 

Unfortunately I have no other module information! :/

APayerl, 2022-02-28:
It actually seems my config is correct...
I left it on during the night and these has arrived some packets but the packets does not get delivered every 15 seconds as they should... (not even every minute)
I think the format of my meter is not supported at the moment...
AmsToMqttBridge lists the meter type as MA304H4 but that is not the full meter specification.
My meter company har several meters in use by Kaifa:
MA304H4PSE(my meter)
MA304H4D
MA105H2E(one phase)
MA304T4(CT)

According to my power company all these have slightly different format and mine in particularly is extra special...

Would it help the project to get a sample output from these meters?

gskjold, 2022-02-28:
Yes, that would be very helpful, put the debugger in verbose mode to see data received.

APayerl, 2022-02-28:
At the moment my NodeMCU does not show anything else then the header so I'm not able to change debugging level....

But I'll add the document sent to me by my power company!

MA304H4PSE 
```
7E <- Encapsulator 
A1 1D 01 00 01 10 B0 AE   <- Header 
E6                        <- Header 
E7                        <- Header 
00                        <- Header 
0F                        <- Header 
40 00 00 00 00            <- Header 
02 24 <- Struct, 36 element 
    09  <- octet-string 
        06 <- 6 bytes 
            01 00 00 02 81 FF   <- 1.0.0.2.129.255 
    09  
        07 <- 7 bytes 
            4B 46 4D 5F 30 30 31 <- KFM_001 
    09  
        06 <- 6 bytes 
            00 00 60 01 00 FF  <- 0.0.96.1.0.255 Meter -ID (GIAI GS1 -16 digit ) 
    09  
        10 <- 16 bytes 
            37 33 34 30 31 35 37 30 33 30 35 34 32 34 37 36 <- 7340157030542476 
    09  
        06 <- 6 bytes 
            00 00 60 01 07 FF <- 0.0.96.1.7.255 Meter type 
    09  
        07 <- 7 bytes 
            4D 41 33 30 34 48 34 <- MA304H4 
    09 06  
        01 00 01 07 00 FF   <- 1.0.1.7.0.255 Active power+ (Q1+Q4) 
    06                      <- Unsigned 
        00 00 00 03         <- 3W 
    09 06  
        01 00 02 07 00 FF   <- 1.0.2.7.0.255 Active power -  (Q2+Q3) 
    06  
        00 00 00 00         <- 0W 
    09 06  
        01 00 03 07 00 FF   <- 1.0.3.7.0.255 Reactive power + ( Q1+Q2) 
    06  
        00 00 00 00         <- 0VAr 
    09 06  
        01 00 04 07 00 FF   <- 1.0.4.7.0.255 Reactive power - ( Q3+Q4) 
    06  
        00 00 00 00         <- 0VAr 
    09 06  
        01 00 1F 07 00 FF   <- 1.0.31.7.0.255 IL1 Current phase L1 
    06  
        00 00 00 20         <- 0,032A 
    09 06  
        01 00 33 07 00 FF   <- 1.0.51.7.0.255 IL2 Current phase L2 
    06  
        00 00 00 00         <- 0,000A 
    09 06  
        01 00 47 07 00 FF   <- 1.0.71.7.0.255 IL3 Current phase L3 
    06  
        00 00 00 00         <- 0,000A 
    09 06  
        01 00 20 07 00 FF   <- 1.0.32.7.0.255 ULN1 Phase voltage 4W meter , Line voltage 3W meter 
    06  
        00 00 09 12         <- 232,2 V 
    09 06  
        01 00 34 07 00 FF   <- 1.0.52.7.0.255 ULN2 Phase voltage 4W meter , Line voltage 3W meter 
    06  
        00 00 00 00         <- 0,0 V 
    09 06  
        01 00 48 07 00 FF   <- 1.0.72.7.0.255 ULN3 Phase voltage 4W meter , Line voltage 3W meter 
[... 219 lines cut ...]
        00 00 00 00 <- 0,000 A L1 
        06  
        00 00 00 00 <- 0,000 A L2 
        06  
        00 00 00 00 <- 0,000 A L3 
        06  
        00 00 09 83 <- 243,5 V L1 
        06  
        00 00 00 00 <- 0,0 V L2 
        06  
        00 00 00 00 <- 0,0 V L3 
        09  
            0C  
                07 E5 05 07 05 0E 2F 2D FF 80 00 00 <- Date and time again 
        15 <- Unsigned Long Long? 
        00 00 00 00 00 00 00 00 <- 0 Wh Active Energy Import 
        15  
        00 00 00 00 00 00 00 00 <- 0 Wh Active Energy Export 
        15  
        00 00 00 00 00 00 00 00 <- 0 VArh Reactive Energy Import 
        15  
        00 00 00 00 00 00 00 00 <- 0 VArh Rective Energy Export 
46 AD <- CRC 
7E <- Encapsulator 
```

APayerl, 2022-02-28:
Now I got telnet access and changed my debugger to verbose.

This was the output....

```
(D t:37077596ms) (loop) Used 0 ms to update temperature
(D t:37092606ms) (loop) Used 0 ms to update temperature
(D t:37097437ms) (readHanPort) Frame dump (1b):
(D t:37097439ms) E7
(W t:37097440ms) (readHanPort) Unknown data format E7
(D t:37107617ms) (loop) Used 0 ms to update temperature
(D t:37122627ms) (loop) Used 0 ms to update temperature
(D t:37127416ms) (readHanPort) Frame dump (1b):
(D t:37127418ms) E7
(W t:37127418ms) (readHanPort) Unknown data format E7
(D t:37137629ms) (loop) Used 0 ms to update temperature
(D t:37152631ms) (loop) Used 0 ms to update temperature
(D t:37157435ms) (readHanPort) Frame dump (1b):
(D t:37157437ms) E7
(W t:37157441ms) (readHanPort) Unknown data format E7
(D t:37167641ms) (loop) Used 0 ms to update temperature
(D t:37182642ms) (loop) Used 0 ms to update temperature
(D t:37187434ms) (readHanPort) Frame dump (1b):
(D t:37187436ms) E7
(W t:37187437ms) (readHanPort) Unknown data format E7
```

And then it continues like this...

APayerl, 2022-02-28:
> The payload you have outlined above is the full payload sent every hour. It could be that the payload sent every 2.5s and every 10s is different. Did you get any information about those?

In the document I got from my power company it states that: "The data stream is pushed from the meter every 15 second. One way communication is enforced - it's not possible to send anything to the meter, only read." 
I've translated it here but it's very clear that it's only every 15 seconds anything is getting pushed.
Also i've stood with my volt meter at the meter for several minutes and it only sends pulses every 15 seconds, all other times the voltmeter is indicating a stable 24V... Can't say if the data changes with every hour push but it sure does not look like anything is sent as you say every 2.5 and 10 seconds.... Only every 15 seconds....

gskjold, 2022-02-28:
Aha, its a Swedish meter, got it. I was assuming Norway, sorry about that :) Then it should work with the payload specified. It sounds like you may have some data corruption before it reaches your esp.

ArnieO, 2022-02-28:
Could it be that this meter runs the IEC 62056-7-5 standard at 115 200 baud?
@APayerl Is the connector on your meter RJ45 (an ethernet cable fits) or RJ12 (similar but more narrow than Ethernet)?

ArnieO, 2022-02-28:
15 second interval is unusual. 
As you have a meter with RJ45 connector and the signals are clearly M-bus, we would expect it to follow the Norwegian "NVE standard". Since Kaifa meters in Norway push data each 2,5 second, we would expect meters in Sweden to do the same. So it is interesting to see your meter has 15 second interval. 

How do you power your interface device; via the NodeMCU? Could it be that the power supply from the NodeMCU is too weak?

APayerl, 2022-02-28:
> @APayerl 
> Please tell us the name of your grid company.

It's called "Mälarenergi", a regional provider
> 
> I know MA304 exist in two versions in Sweden. One using the Norwegian 8-pin port, and one using the Swedish 6-pin port.
> 
> From your measured 24V, it seems like you have the 8-pin port, but please confirm.

Yes my version is RJ45(8 pin)

APayerl, 2022-02-28:
Just got a frame...

```
D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 1 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 1 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (287b):
(D) 7E A1 1D 01  00 01 10 B0  AE E6 E7 00  0F 40 00 00
(D) 00 00 02 24  09 06 01 00  00 02 81 FF  09 07 4B 46
(D) 4D 5F 30 30  31 09 06 00  00 60 01 00  FF 09 10 37
(D) 33 34 30 31  35 37 30 33  30 33 31 38  35 38 33 09
(D) 06 00 00 60  01 07 FF 09  07 4D 41 33  30 34 48 34
(D) 09 06 01 00  01 07 00 FF  06 00 00 0C  27 09 06 01
(D) 00 02 07 00  FF 06 00 00  00 00 09 06  01 00 03 07
(D) 00 FF 06 00  00 07 BE 09  06 01 00 04  07 00 FF 06
(D) 00 00 00 00  09 06 01 00  1F 07 00 FF  06 00 00 19
(D) 55 09 06 01  00 33 07 00  FF 00 00 FF  76 09 06 01
(D) 00 04 08 00  FF 7E A1 1D  01 00 01 10  B0 AE E6 E7
(D) 00 0F 40 00  00 00 00 02  24 09 06 01  00 00 02 81
(D) FF 09 07 4B  46 4D 5F 30  30 31 09 06  00 00 60 01
(D) 00 FF 09 10  37 33 34 30  31 35 37 30  33 30 33 31
(D) 38 35 38 33  09 06 00 00  60 01 07 FF  09 07 4D 41
(D) 33 30 34 48  34 09 06 01  00 01 07 00  FF 06 00 00
(D) 0C 22 09 06  01 00 02 07  00 FF 06 00  00 00 00 09
(D) 06 01 00 03  07 00 FF 06  00 00 07 A6  09 06 01
(W) (readHanPort) Boundry flag missing
(D) (readHanPort) Frame dump (1b):
(D) 09
(W) (readHanPort) Unknown data format 09
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
```

tronde-ams, 2022-02-28:
Then we know
`Alla nya elmätare från Mälarenergi följer den norska standarden för HAN-port (AMS+HAN) som sänder ut data enligt protokoll IEC 62056-7-5, annex D ”MBUS”.`

https://www.malarenergi.se/kunskapsbanken/el/han-porten/

tronde-ams, 2022-02-28:
Parametrar som du kan ta ut via HAN-porten är:

    Mätarens aktuella datum och tid
    Mätarnummer
    Mätartyp
    Momentan effekt i alla fyra kvadranter (import, export, aktiv, reaktiv)
    Total energi (mätarställning) i alla fyra kvadranter (import, export, aktiv, reaktiv)
    Spänning per fas
    Ström per fas

Dataströmmen pushas ut från mätaren var 15:e sekund.

ArnieO, 2022-02-28:
> Just got a frame...
> 
> ```

Then it looks like your issue is HW related in some way. Difficult to debug further without an oscilloscope, I think...

APayerl, 2022-02-28:
> > Just got a frame...
> > 
> > ```
> 
> Then it looks like your issue is HW related in some way. Difficult to debug further without an oscilloscope, I think...

crap... 😕
Maybe I should just buy the universal hardware from the project... should that work?

tronde-ams, 2022-02-28:
Make sure the wiring and all connectors between your meter and MBUS interface is OK.  It seems like you use standard network cable (cat 5?) which is OK. I have tested with 15 m without problem from my Aidon meter.

Try another power supply for the ESP.

What does the WEB-interface of the ESP look like? It shows uptime. Any indication your ESP is re-booting often?

ArnieO, 2022-02-28:
There is not any hardware being part of this firmware project, the firmware is aimed at being functional on any ESP8266 or ESP32 based hardware (with some minor functionality difference due to limitations in the 8266).

I do my best to share my experience here, to the best of my ability, to help those struggling with their own hardware. However, I have no hands-on experience with such adapters as the one you use. There is at least one (long) thread on hjemmeautomasjon.no where quite a bit of experience with various such setups are shared. You might want to look into that and see if you find relevant advise. You seem to be close, as you are receiving data (with seemingly some read errors).

That said,I sell readers from my webshop amsleser.no (currently open for sales to Norway, Sweden and Denmark). The Pow-U should work on your meter. There is around 100 of those currently operational in Norway, none yet in Sweden if I recall correctly. Feel free to email me on post@amsleser.no.

APayerl, 2022-02-28:
> Make sure the wiring and all connectors between your meter and MBUS interface is OK.  It seems like you use standard network cable (cat 5?) which is OK. I have tested with 15 m without problem from [...]

If I remember correctly it's even cat6 and the length is about 6m.

I have ordered a new power supply (raspberry 3 power supply) instead of the old mobile adapter I currently use...

[image]

ArnieO, 2022-02-28:
> Make sure the wiring and all connectors between your meter and MBUS interface is OK. It seems like you use standard network cable (cat 5?) which is OK. I have tested with 15 m without problem from [...]

Good advise here. One farmer in Norway uses a 50 meter buried cable. The big advantage with M-bus is that the cable can be long.

ArnieO, 2022-02-28:
Ouch... RSSI that low, less than -90dBm is very low. Is probably not the direct cause of read error, but I believe the ESP8266 in such situation will crank up the emission power, straining the power supply, potentially giving secondary issues. The ESPs are known for needing good decoupling, I am in general incertain if the various development boards like NodeMCU are sufficiently decoupled for such a demanding application.

tronde-ams, 2022-02-28:
Yes, the cable can be long if twisted pair. Else usually not. 

@APayerl 
Do you have access to some resistors? 

@gskjold 
What makes the ESP-indicator turn red?

ArnieO, 2022-02-28:
Red ESP is due to low voltage in the ESP. Good catch, I did not notice. But there is some quirk with  Vcc reading on the NodeMCU.

APayerl, 2022-02-28:
> Ouch... RSSI that low, less than -90dBm is very low. Is probably not the direct cause of read error, but I believe the ESP8266 in such situation will crank up the emission power, straining the power [...]

Might be since its quite far from the AP but I'm working on getting it closer...

APayerl, 2022-02-28:
> Yes, the cable can be long if twisted pair. Else usually not. 
> 
> @APayerl 
> Do you have access to some resistors? 
> 
> @gskjold 
> What makes the ESP-indicator turn red? 

Yea but only 0.5w resistors...

APayerl, 2022-02-28:
> Red ESP is due to low voltage in the ESP. Good catch, I did not notice. But there is some quirk with  Vcc reading on the NodeMCU.

Could a better charger solve it?...

tronde-ams, 2022-02-28:
Try to connect a restistor of about 47 kohm (value not important) between D2 and GND on the ESP. This is to make sure the input is not floating.

APayerl, 2022-03-01:
> Try to connect a restistor of about 47 kohm (value not important) between D2 and GND on the ESP. This is to make sure the input is not floating.

Wow you are my hero!
I connected a 47kOhm  between TSS721::GND and TSS721::TXD and that actually solved my problem! :D
[image]

After flashing the firmware that [gskjold](https://github.com/gskjold) posted I still had problems though with decimals I think....
[image]

gskjold, 2022-03-01:
In meter config, have the manufacturer been detected?

APayerl, 2022-03-01:
No....
[image]

tronde-ams, 2022-03-01:
This is how I believe your output from the interface is

[image]

This should not cause any problems, because the output would be either close to 3,3V or GND.  But I have experienced a lot of trouble with this configuration of the optocoupler in this particular application. I believe it has to to with poor decoupling of the 3,3V in the ESP, and that the input will receive radio interference because of that. We have a radio transmitter only a few mm away from the input pin on the ESP. The extra resistor will aid to tie the input more close to GND. 

A capacitor instead of the extra resistor will kill the noise, but it will also damage the sharp edges of the square wave signal that is the data, and cause trouble that way. 

You can try to add a film capacitor between 3V3 and GND on the ESP and see if it helps,

I took a different approach to the MBUS adapter, and made my own. 

(https://github.com/gskjold/AmsToMqttBridge/files/8162635/very_simple_han_to_ttl.norsk.pdf)

In this design, you can see the opto coupler is between 3,3V and out, and the resistor between out and GND.
I have been using this design for two years now, and some of my friends have got some kits. None have told about any problems. This design is probably best for people that can tweak it if they face some problems, so don't see it as a fool prof design.  I don't think the type of optocoupler is critical. I have many of this type, so I used it because of that.

ArnieO, 2022-03-01:
Great advise! But the data he receives seems to be... odd. Were you exporting energy at 14:30 today, @APayerl ?

APayerl, 2022-03-01:
> Great advise! But the data he receives seems to be... odd. Were you exporting energy at 14:30 today, @APayerl ?

Yea I very well could have (Solar panels)...
But if you look at the phase voltage I am 99.999% sure I did not have more than 2kV per phase in a house that should have ~230V... Same as the current, seriously doubt it was more then 2kA on phase one for example... 
So did I export 282W or 28.2W?.... Who knows....
But I think the import gauge is correct... At the moment it says 2.8kW import, and I am inclined to believe it (3 computers running + washer)

ArnieO, 2022-03-01:
@gskjold will have to look into this, it is surprising that the data format is so different compared to what we are used to from other meters.

APayerl, 2022-03-02:
> We don't know if the resistor solved the whole problem, but we can be quite certain his setup suffers from some kind of instability.
> 
> I found a short user manual for this meter. @APayerl , try to compare the values from the meter with the values from the adapter [matarmanual-utokad.pdf](https://github.com/gskjold/AmsToMqttBridge/files/8163970/matarmanual-utokad.pdf)
> 
> I do also have a strong feeling the Swedish grid companies have more (probably too much) freedom when they deciede about what to push to the HAN-port compared to Norwegian companies.

Today I had it running all day and got beautiful graphs in Home Assistant so I think the resistor worked! Unfortunately I went to exchange the charger with a "better" one but when I tried to put the micro-USB cable into the NodeMCU the USB contact(of the NodeMCU) fell off... Guess I'm literally back to square one with having to get a new NodeMCU.... :S But when I do I'll check the values like you proposed!

tronde-ams, 2022-03-02:
You can feed an external positive voltage to the pin "Vin" on your ESP and 0V to "GND" if you can manage to do so in a way. 
Most likely it will be OK with +5V, but something between +7V and +12V is what Vin is what it is intended for. Some places you can read up to +20, but I know this can be too much for the voltage regulator on the module.

gskjold, 2022-03-03:
I think I see why it wasn't detected, try this one:

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8175980/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8175981/esp8266.zip)

gskjold, 2022-03-03:
Found a typ-o, so yet another version: 
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8176008/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8176009/esp8266.zip)

APayerl, 2022-03-04:
Does this indicate that no frames are getting received?
(When my board broke I had to resolder most of it)

ArnieO, 2022-03-05:
> Does this indicate that no frames are getting received?

If that is "verbose" mode debugging, you are not receiving any data.

APayerl, 2022-03-05:
@gskjold So I just installed your newest code, but unfortunately it's still wrongly formatted....
[image]
[image]
(My solar panels are currently exporting)

gskjold, 2022-03-11:
Lets try one more thing:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8234255/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8234257/esp8266.zip)

If it still doesn't work, please provide data dump from verbose mode.

APayerl, 2022-03-22:
> Lets try one more thing:
> [esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8234255/esp32.zip)
> [esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8234257/esp8266.zip)
> 
> 
> If it still doesn't work, please provide data dump from verbose mode.

Well done! I'd say you have succeeded! 😘
[image]
[image]

gskjold, 2022-03-23:
Excellent! Will be public either 2.0.12 (if released) or 2.1.0

closed 2022-03-23 by gskjold