UtilitechAS/amsreader-firmware #243: HAN interface RED after upgrading to 2.0.11 (Pow-K)

Claustn, 2022-02-28:
**After upgrading to 2.0.11 I no longer get data, and HAN is red**
I have been running 1.5.5 for several months now without any issues but decided I wanted to try the latest version (2.0.11)

I went into the web interface and uploaded the ams2mqtt-esp8266-2.0.11.bin file, device restarted but did not come back up after 5 minutes.. I took the Pow-K out and resat it... Then I could connect to the wifi hotspot and confirm it was running 2.0.11 and I configured Wifi. When the devices came back up I configured the Meter (Kamstrup | 2400 | 8N1 and entered my auth/encryption keys)  But it never seems to get data (HAN icon is red)

I measured the port with a multimeter and it seems that every 10 sec I get a voltage drop, also I get a blue blink about every 10 seconds as well when the board is connected. (So I assume data is sent)

I have tried to downgrade to 1.5.5 and 1.5.8 without any luck. I have tried to enable debugging but it does not even seem to log anything from serial communication
See below example

```
* Please type the command and press enter to execute.(? or h for this help)
***
(D) (AmsWebServer)Serving /data.json over http...
l
* Debug: Command received: l
* Show debug level: Off
d
* Debug: Command received: d
* Debug level set to Debug
(loop) Used 0 ms to update temperature
l
* Debug: Command received: l
* Show debug level: On
(D) (AmsWebServer)Serving /data.json over http...
t
* Debug: Command received: t
* Show time: On
p
* Debug: Command received: p
* Show profiler: On
(D t:1764044ms p:^14377ms) (AmsWebServer)Serving /data.json over http...
(D t:1770048ms p:^6003ms) (AmsWebServer)Serving /data.json over http...
(D t:1770272ms p:^0223ms) (loop) Used 0 ms to update temperature
(D t:1776054ms p:^5781ms) (AmsWebServer)Serving /data.json over http...
(D t:1782066ms p:^6011ms) (AmsWebServer)Serving /data.json over http...
(D t:1785273ms p:^3206ms) (loop) Used 0 ms to update temperature
(D t:1788041ms p:^2767ms) (AmsWebServer)Serving /data.json over http...
(D t:1794043ms p:^6001ms) (AmsWebServer)Serving /data.json over http...
(D t:1800039ms p:^5995ms) (AmsWebServer)Serving /data.json over http...
(D t:1800275ms p:^0235ms) (loop) Used 0 ms to update temperature
(D t:1806080ms p:^5805ms) (AmsWebServer)Serving /data.json over http...
(D t:1812046ms p:^5965ms) (AmsWebServer)Serving /data.json over http...
(D t:1815276ms p:^3229ms) (loop) Used 0 ms to update temperature
(D t:1818040ms p:^2763ms) (AmsWebServer)Serving /data.json over http...
(D t:1824071ms p:^6030ms) (AmsWebServer)Serving /data.json over http...
(D t:1830109ms p:^6037ms) (AmsWebServer)Serving /data.json over http...
(D t:1830279ms p:^0169ms) (loop) Used 0 ms to update temperature
(D t:1836048ms p:^5768ms) (AmsWebServer)Serving /data.json over http...
```

**Hardware information:**
 - Meter: Kamstrup (With encryption in Denmark)
 - AMS reader:Pow-K
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 1.5.0]
 - MQTT:  yes
 - HAN GPIO: [ GPI12]
 - Temperature sensors [e.g. 3xDS18B20]

ArnieO, 2022-02-28:
Can you please show screen clips of your config/meter and system/GPIO settings?
With v2.0.11 please.

Claustn, 2022-02-28:
I am not sure what has happened, but it seems to work now after upgrading from 1.5.5 to 2.0.11 for the x'th time

I set the GPIO to UART0 (I Had tried that a few times before, actually went through all of the GPIO pins in the list)

Thank you for your quick response, and sorry for the noise

[image]
[image]

ArnieO, 2022-02-28:
No worries, just glad you got it up and running! 
Yes that is an old one, using UART0. On newest boards I have written the HAN GPIO as part of the text. Maybe a good idea if you make a note, it is easy to forget.
Thank you!

closed 2022-03-03 by gskjold