UtilitechAS/amsreader-firmware #244: Random failed HAN-port reads

rugaard, 2022-03-02:
_**Disclaimer**: Before I start describing my issue, I just want to note, that I'm not 100% sure this is a bug in AmsToMqttBridge, since it happens on the HAN-port, so it might just be my Kamstrup device. But just to make sure, I thought I would create this issue, to rule everything out._

---

**Describe the bug**
While having a green and working HAN indicator and a nice set of graphs, randomly the HAN indicators turns red. I therefore turned on the telnet debugger and tried trailing the log.

Without any specific pattern (at least, that I could figure out), the HAN port reader fails with different errors. 
So far I've encountered:

- Frame checksum error
- Unknown format
- Boundry flag missing

_Note: I live in Denmark and therefore I'm receiving the data from the HAN port with the help of the GPK60/GPK61 encryption keys._

**To Reproduce**
I am not able to reproduce specifically, it just happens randomly.

**Expected behavior**
No red HAN-indicator and no `readHanPort` errors.

**Hardware information:**
 - Meter: Kamstrup (Denmark)
 - AMS reader: Pow-K

**Relevant firmware information:**
 - Version: 2.0.11
 - MQTT: yes
 - HAN baud and parity: 2400 8N1

**Additional context**
I've grabbed a good amount of logging from the Telnet output, where there's multiple failed HAN-port reads, but also multiple successful HAN-port reads. Hopefully some of you can make some use of it :)

Link: [Telnet output log](https://gist.github.com/rugaard/102d3c9b7b15751591c31cd697949f8b)

Jelboen, 2022-03-02:
Above issue is exactly the same I'm experiencing. I can read data and everything, And I can listen to the MQTT feed on the receiving side. Data goes through, but I guess for these failed readings, data is lost.

ArnieO, 2022-03-02:
Very interesting - and again "something strange" with Kamstrup meters. 

I am in Norway, also on a Kamstrup meter. Sometimes when I open the GUI I will see yellow HAN indication in the GUI.

Here is one invalid frame between two good ones:
[image]
(I am in Norway, our 10 sec payloads are shorter than the one you get in Denmark.)

Her is another invalid:
[image]

Note that both invalid frames are identical until the 32nd byte (last byte on 2nd row). 

Bitwise comparison of the first bytes of a good and a bad payload:
[image]

It seems unlikely that this is due to bit errors - which could indicate either noisy data/read error, HW error or something happening in the ESP during the serial read-in of data.

I honestly have no idea what could be the cause.
Could the meters from time be pushing payloads that are undocumented - hence unreadable?

rugaard, 2022-03-02:
I'm absolutely no expert in anything power related. So I might be wrong here, but could it be because there's not enough stored power on the ESP8266 board to finish/send the received data? 

Regarding danish Kamstrup meters pushing anything undocumented, it would not surprise if that is the case at this point.

But really, what you're saying, is that there is not really anything we can do about it? I'm a novice when it comes anything electrical and PCB board related, so I would have no idea where to even start begin debugging or testing 😏

ArnieO, 2022-03-02:
> ...could it be because there's not enough stored power on the ESP8266 board to finish/send the received data?

The Pow-K has a huge super capacitor that ensures stable voltage, which is why I am confident this issue is something else.

kabu-kabu, 2022-03-04:
I experience those errors only while accessing the web interface. And after the frame checksum error my esp8266 board reboots. Overnight everything was fine. No web access, no errors, no reboots.

Can someone confirm this behaviour?

I'm in Austria on a Kaifa meter.

ArnieO, 2022-03-04:
> And after the frame checksum error my esp8266 board reboots.

Interesting. No, I have not seen any link between checksum errors / reboot / using the GUI. My board reboots now and then (1-3 times per day), potentially a memory management issue that @gskjold is looking into.
Which version firmware are you running?
Can you please keep an eye on the evolution of "Free mem" on the status bar and report back? It should be at around 15-20 kb.

rugaard, 2022-03-04:
> Can you please keep an eye on the evolution of "Free mem" on the status bar and report back? It should be at around 15-20 kb.

I'm constantly below 10 kb of free memory. It jumps between 6.0 kb and 9.6 kb pretty much the entire day.

closed 2022-03-17 by gskjold

flmmmm, 2022-03-27:
I have had the same problem with esp8266, han port yellow most of the time, but esp32 runs perfect. han port green and mqtt values every 10 sec. Danish Kamstrup Omnipower without encryption.
//Flemming

MikOsle, 2022-05-02:
Hi. I'm still have this issue with random failure of the HAN-port. I'm running 2.1.2 on a D1 and have added a super capacitor to the 3.3Vcc. Meter: Kamstrup (Denmark). My experience is that whenever it is failing I just have to navigate in the web interface and then it is working again