UtilitechAS/amsreader-firmware #247: Testing v2.1.0

gskjold, 2022-03-12:
Attached is a new major version with changes outlined below. If anyone could test this on their setup, it would be highly appreciated.

!Warning!
This is a firmware which is still under development. Testing this firmware is on your own risk. Please only test if you have possibility to flash via TTL or USB if anything goes wrong.

Changes:
- Improved algorithm to update hour and day graph values
- Removed support for loading config from v1.4.x series firmware
- Visualize export as separate negative bar in graph

Added:
- Month threshold tracking according to Norwegian distribution tariff
- Ability to backup and restore from human readable config file
- Support for Home Assistant auto discovery
- Send uncoded payload to MQTT (HEX encoded bytes)
- Possibility to view wattage display in watt always instead of kw (Click the value to change)
- Possibility to factory reset during boot
- Possibility to adjust WiFi power

**Norway**
- [ ] Aidon 1p
- [ ] Aidon 3p IT
- [ ] Aidon 3p TN
- [ ] Kaifa 1p
- [ ] Kaifa 3p IT
- [ ] Kaifa 3p TN
- [ ] Kamstrup 1p
- [ ] Kamstrup 3p IT
- [ ] Kamstrup 3p TN

**Denmark**
- [ ] Kamstrup 1p
- [ ] Kamstrup 3p

**Sweden**
- [ ] Aidon 3p TN

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8238253/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8238254/esp8266.zip)

Isaksson, 2022-03-13:
Phase Current graph works great now when negative values on all three phases.
I just installed this version so I have not used it for long but everything seems fine for now.

rugaard, 2022-03-13:
@Jelboen Wouldn't you be up for testing the Kamstrup 3p in Denmark? 🤪  Then we could also test to see, if it fixes our issue (#244) as well.

If not, I could probably do it. Only thing is my meter is located outside behind annoying trap doors, so it's a bit complicated if I have to flash it multiple times 😆  But if no one else signs up, I wouldn't mind helping out @gskjold 👍

tronde-ams, 2022-03-13:
ESP32:
First impression on Aidon 3p IT without export is good. It does update all values in  real time, also the total imported kWh.

I can see you push real time daily and hourly values on /data.json now. Have you made any changes for MQTT as well?

kabu-kabu, 2022-03-14:
Just testet on ESP8266. Partly working with Kaifa 3p TN:
- Board doesn't restart anymore while accessing gui.
- Data is received from HAN and pushed to MQTT.
- But data is not displayed in gui.

ArnieO, 2022-03-14:
@BKFlister Did you use usb power when you tried to upgrade? 
It is necessary to use external power when upgrading Pow-U, or else it will go down for lack of energy, the upgrading sequence is power-hungry.

svhaaland, 2022-03-14:
Testet on ESP8266 with Aidon 3p IT (Norway)

- Web GUI - OK
- Config Download - OK
- Config Upload - Failed 
- MQTT HA values - OK
- MQTT HA AutoDiscover - Failed - Missing values:
> "uniq_id":"�i�?_rssi","obj_id":"�i�?_rssi","val_tpl":"{% if value_json.rssi is defined %} {{ value_json.rssi }} {% else %} {{(states.sensor.,~�?_rssi|float)}} {% endif %}"

tronde-ams, 2022-03-15:
ESP32
I have tested the config file now. It does save and re-load, but something seems to be wrong with re-creation of the dayplot and monthplot.

I have two ESP32 running c38034c. 

One of them is OK, the other went haywire in the plots when I upgraded from 2.0.10. 
I can load the config from the OK unit, and everything except the plots are transfered OK. The plots does not change.

I have not tried a factory reset yet, in case there is something I should try first,

gskjold, 2022-03-15:
> I can see you push real time daily and hourly values on /data.json now. Have you made any changes for MQTT as well?

Raw mode sends '/realtime/import/hour' and '/realtime/import/threshold'.
JSON mode has realtime: { h: 0.00, t: 0 } for hourly and threshold.

Have not modified HA or Domoticz.

gskjold, 2022-03-15:
Can confirm that config download/upload has a problem. New update in first post.

gskjold, 2022-03-15:
@kabu-kabu if you use Google Chrome, tell me if there is any output in the developer console.

First open dev tools:
[image]

Then go to the console tab:
[image]

Reload page

gskjold, 2022-03-15:
@BKFlister is it totally dead, or is there LED activity? 

If there is LED activity, it should light the green LED for 1s, then flash other colors. During the 1s the green LED is on, press and hold AP button (do not start pressing it before green), it should then change to yellow for 2s and then red. When it turns red, release button. It should then factory reset and should be reachable via AP.

If that doesn't work, you can also try to repeat, but keep holding through the red and flashing led, then release.

kabu-kabu, 2022-03-15:
No output in the Chrome dev console.

Import and export graphs are 0 and within the next two boxes I get red message boxes "Data table is not defined." using Firefox on Android.

[image]

gskjold, 2022-03-15:
Aha, I see. It must have trouble decoding the full frames. Could you enable the internal debugger in System -> Debugging and set it to verbose. Give me some output from the telnet console.

bjornsivertsen, 2022-03-15:
> @BKFlister Did you use usb power when you tried to upgrade? It is necessary to use external power when upgrading Pow-U, or else it will go down for lack of energy, the upgrading sequence is power-hungry.

I have updated around 7-8 times now without an external power supply with zero issues. But I suppose it must have been more about luck?

Maybe I should get one, just to be safe when this is publicly released.

kabu-kabu, 2022-03-15:
Here are some HAN messages.

```
(I) (readHanPort) Intermediate segment
(D) (readHanPort) Frame dump (256b):
(D) 68 FA FA 68  53 FF 00 01  67 DB 08 4B  46 4D 65 50
(D) 9A 0F 03 81  F8 20 00 06  C9 F1 D0 C8  1C 4A E3 F2
(D) 3E 09 3B E9  62 13 67 F2  15 13 E7 0E  D2 4B 84 B2
(D) D2 45 0B DD  92 3D 27 16  D2 28 23 B1  A3 B6 03 CD
(D) 56 4F D3 5D  C4 E3 6D 2A  25 54 7A 60  EE 35 A9 3A
(D) 42 3A FD BE  A1 B1 18 15  7A 18 5B 6D  8D 87 DB 72
(D) 37 6A 19 E4  EE 05 93 5A  35 22 46 EE  14 27 79 4D
(D) 31 E4 50 D1  7A CD A0 E2  33 62 E1 A6  9F 26 74 1B
(D) 30 39 DE F4  AB 2C 9E 47  87 00 84 AE  00 12 FE 9E
(D) F9 94 5D 7B  DA 5E 4D 41  A8 D1 22 B9  BC 5F E4 DF
(D) 69 53 93 4B  58 FB BA E8  86 0D CE 03  69 3F AF E7
(D) 84 1E E3 C7  BF AB 12 58  E1 BE 50 CF  34 9A EB 25
(D) 92 D1 48 DE  D4 D5 61 BA  B2 8A 19 AA  FF 08 B4 7B
(D) 29 25 C3 DC  C2 73 43 DB  45 24 A3 22  60 E3 20 20
(D) 86 D7 10 4C  08 99 1C 3E  31 C6 6E CC  55 6B D7 96
(D) 51 6A 21 CD  0F B0 1C A7  3C E1 E5 8B  0C 57 51 16
(D)
(I) (readHanPort) Final segment
(D) (readHanPort) Frame dump (26b):
(D) 68 14 14 68  53 FF 11 01  67 C1 F8 CF  A9 1A E3 04
(D) 0A 3B AE 85  60 FB 7B 0C  57 16
(D) (readHanPort) Frame dump (271b):
(D) 68 00 00 68  53 FF 10 01  67 DB 08 4B  46 4D 65 50
(D) 9A 0F 03 81  F8 20 00 06  C9 F1 0F 80  4C 7F B7 0C
(D) 07 E6 03 0F  02 0A 06 28  00 FF C4 00  02 23 09 0C
(D) 07 E6 03 0F  02 0A 06 28  00 FF C4 00  09 06 01 00
(D) 01 08 00 FF  06 00 27 86  74 02 02 0F  00 16 1E 09
(D) 06 01 00 02  08 00 FF 06  00 82 48 8E  02 02 0F 00
(D) 16 1E 09 06  01 00 01 07  00 FF 06 00  00 00 00 02
(D) 02 0F 00 16  1B 09 06 01  00 02 07 00  FF 06 00 00
(D) 12 FF 02 02  0F 00 16 1B  09 06 01 00  20 07 00 FF
(D) 12 09 56 02  02 0F FF 16  23 09 06 01  00 34 07 00
(D) FF 12 09 47  02 02 0F FF  16 23 09 06  01 00 48 07
(D) 00 FF 12 09  56 02 02 0F  FF 16 23 09  06 01 00 1F
(D) 07 00 FF 12  02 B3 02 02  0F FE 16 21  09 06 01 00
(D) 33 07 00 FF  12 02 A6 02  02 0F FE 16  21 09 06 01
(D) 00 47 07 00  FF 12 02 A6  02 02 0F FE  16 21 09 06
(D) 01 00 0D 07  00 FF 12 03  E7 02 02 0F  FD 16 FF 09
(D) 0C 31 38 31  32 31 30 30  39 36 33 38  37 ED 16
(D) (readHanPort) System title:
(D) 4B 46 4D 65  50 9A 0F 03
(D) (readHanPort) Initialization vector:
(D) 4B 46 4D 65  50 9A 0F 03  00 06 C9 F1
(D) (readHanPort) Additional authenticated data:
(D) 20 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00
(D) 00
(D) (readHanPort) Authentication tag:
(D) 00 00 00 00  00 00 00 00  00 00 00 00
(D) (readHanPort) Valid data, start at byte 44
(V) (EnergyAccounting)  calculating threshold, currently at 10
(V) (EnergyAccounting)  new threshold 10
(D) (AmsWebServer)Serving /data.json over http...
(I) (readHanPort) Intermediate segment
(D) (readHanPort) Frame dump (256b):
(D) 68 FA FA 68  53 FF 00 01  67 DB 08 4B  46 4D 65 50
(D) 9A 0F 03 81  F8 20 00 06  C9 F2 60 25  42 F3 1A 74
(D) 10 B9 C6 04  D2 62 D5 EB  48 3F FA 9D  C4 E3 3C 3A
(D) 98 B8 31 0C  42 AE 68 D7  7F 18 A5 64  75 9B CD AF
(D) 20 8A AD BD  22 B2 71 46  69 B1 B7 76  2B 9C 2C 4F
(D) 32 67 2C EE  9B 6B EB 29  45 CC 93 35  D5 72 6B 46
(D) 47 01 38 FE  FB 2F E7 3C  FA C8 E1 AF  AC 62 9F D1
(D) AD CE 71 B4  3D E6 B5 91  14 DE B6 9D  72 D1 55 D7
(D) 53 56 0B E7  CD B5 52 05  79 45 AA A1  77 11 B8 70
(D) DF 06 B2 29  87 A7 6E 7C  94 94 18 2E  31 19 25 20
(D) 1A A6 46 94  40 B1 59 BA  C0 F1 34 68  65 A2 E8 A9
[... 342 lines cut ...]
(D) 06 01 00 02  08 00 FF 06  00 82 48 BE  02 02 0F 00
(D) 16 1E 09 06  01 00 01 07  00 FF 06 00  00 00 00 02
(D) 02 0F 00 16  1B 09 06 01  00 02 07 00  FF 06 00 00
(D) 13 8F 02 02  0F 00 16 1B  09 06 01 00  20 07 00 FF
(D) 12 09 52 02  02 0F FF 16  23 09 06 01  00 34 07 00
(D) FF 12 09 56  02 02 0F FF  16 23 09 06  01 00 48 07
(D) 00 FF 12 09  55 02 02 0F  FF 16 23 09  06 01 00 1F
(D) 07 00 FF 12  02 C5 02 02  0F FE 16 21  09 06 01 00
(D) 33 07 00 FF  12 02 BD 02  02 0F FE 16  21 09 06 01
(D) 00 47 07 00  FF 12 02 B7  02 02 0F FE  16 21 09 06
(D) 01 00 0D 07  00 FF 12 03  E7 02 02 0F  FD 16 FF 09
(D) 0C 31 38 31  32 31 30 30  39 36 33 38  37 1C 16
(D) (readHanPort) System title:
(D) 4B 46 4D 65  50 9A 0F 03
(D) (readHanPort) Initialization vector:
(D) 4B 46 4D 65  50 9A 0F 03  00 06 C9 F8
(D) (readHanPort) Additional authenticated data:
(D) 20 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00
(D) 00
(D) (readHanPort) Authentication tag:
(D) 00 00 00 00  00 00 00 00  00 00 00 00
(D) (readHanPort) Valid data, start at byte 44
(V) (EnergyAccounting)  calculating threshold, currently at 10
(V) (EnergyAccounting)  new threshold 10
```

gskjold, 2022-03-15:
@kabu-kabu Thanks! What country are you in? Have this meter worked with previous versions?

kabu-kabu, 2022-03-15:
I live in Austria, my meter is Kaifa MA309.

Jumped into the project by using version 2.0.11 - my meter was recognized as "manufacturer: Sagemcom, model: unknown"; I got plausible data which was pushed via MQTT; only problem was a regular restart while accessing the gui.

With version 2.1.0 my meter is not recognized ("manufacturer: unknown") but I do get plausible data via MQTT. gui is not working.

ArnieO, 2022-03-15:
> I live in Austria, my meter is Kaifa MA309.
>
> Jumped into the project by using version 2.0.11 - my meter was recognized as "manufacturer: Sagemcom, model: unknown"; I got plausible data which was [...]

Interesting!
What hardware are you using? Regular restart while accessing GUI is probably linked to issues outside of the firmware. 
I suspect some restart issues are linked to Wifi. What is your RSSI (indicated on GUI)?

gskjold, 2022-03-15:
Thanks! I figured it was Austria. The format is same as Sagecom, but I find it strange it was recognized as that as well, because the identifier differs. I will use your data dump as source for debugging locally when I have time.

tronde-ams, 2022-03-15:
ESP32 / 42a627e
All graphs updated correctly, but real time price values for day and month did not change.
[image]
[image]

gskjold, 2022-03-15:
Day cost is calculated on the fly based on values from graph, cost for those hours received from entsoe, and entsoe multiplier. Did any of those change?

tronde-ams, 2022-03-15:
> Day cost is calculated on the fly based on values from graph, cost for those hours received from entsoe, and entsoe multiplier. Did any of those change?

The costs are OK now. Updated when new hour? 
All graphs on both units was equal when the prices was wrong.

I have also tested the raw MQTT /realtime/import/hour and /realtime/import/threshold to be OK. 

As of now, I find no other errors from what I can test. I do only use the GUI and raw MQTT to an android app.

One wish: Is it possible to push /realtime/import/day as MQTT as well?

tronde-ams, 2022-03-15:
> 
> [image] [image]

I see one thing I did not notice. The 150 kWh treshold did not change. I had one one hour value of 131 kW in the unit that went haywire, and it seems like it has triggered the 150 kWh that is still there even when the costs have changed back to correct values.

bjornsivertsen, 2022-03-15:
> > I am using Kamstrup in Denmark, and there appears to be issues. HAN status flashes between yellow and green.
> 
> Can you please show "System/GIPO"? You might need to set "HAN" to GPIO12 again.

[image]

Here you go

bjornsivertsen, 2022-03-15:
tel

> > Where is the output located?
> 
> @bjornsivertsen You will not be able to use serial debugging from a Pow-K. Use Telnet, then open a command window and type "telnet . You might need to activate Telnet on your PC, it is inactive by default (Google will help!)

Thank you, it looks like this:

`(D) (readHanPort) Frame dump (1b):
(D) 6F
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0D E2 8C
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
(D) 4B 20 2E A4  64 32 F0 C0  31 47 7F C9
(W) (readHanPort) Unknown data format 6F
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (320b):
(D) 7B 22 69 6D  22 3A 39 32  30 30 2C 22  6F 6D 22 3A
(D) 30 2C 22 6D  66 22 3A 34  30 2C 22 69  22 3A 30 2C
(D) 22 65 22 3A  30 2C 22 72  69 22 3A 30  2C 22 72 65
(D) 22 3A 30 2C  22 69 63 22  3A 30 2E 30  30 30 2C 22
(D) 65 63 22 3A  30 2E 30 30  30 2C 22 72  69 63 22 3A
(D) 30 2E 30 30  30 2C 22 72  65 63 22 3A  30 2E 30 30
(D) 30 2C 22 75  31 22 3A 30  2E 30 30 2C  22 75 32 22
(D) 3A 30 2E 30  30 2C 22 75  33 22 3A 30  2E 30 30 2C
(D) 22 69 31 22  3A 30 2E 30  30 2C 22 69  32 22 3A 30
(D) 2E 30 30 2C  22 69 33 22  3A 30 2E 30  30 2C 22 66
(D) 22 3A 30 2E  30 30 2C 22  66 31 22 3A  30 2E 30 30
(D) 2C 22 66 32  22 3A 30 2E  30 30 2C 22  66 33 22 3A
(D) 30 2E 30 30  2C 22 76 22  3A 33 2E 34  32 32 2C 22
(D) 72 22 3A 2D  35 36 2C 22  74 22 3A 2D  31 32 37 2E
(D) 30 30 2C 22  75 22 3A 35  37 33 36 2C  22 6D 22 3A
(D) 31 33 38 30  38 2C 22 65  6D 22 3A 31  2C 22 68 6D
(D) 22 3A 33 2C  22 77 6D 22  3A 31 2C 22  6D 6D 22 3A
(D) 30 2C 22 6D  65 22 3A 30  2C 22 70 22  3A 6E 75 6C
(D) 6C 2C 22 6D  74 22 3A 32  35 35 2C 22  64 73 22 3A
(D) 31 2C 22 65  61 22 3A 7B  22 78 22 3A  31 2E 35 C3
(D)
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0D E2 8C
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
(D) 4B 20 2E A4  64 32 F0 C0  31 47 7F C9
(W) (readHanPort) Unknown data format 7B
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...`

dioden55, 2022-03-15:
hi
Tested pow-u and Aidon meter.
it works as earlier, 
how to test changing wfi power, what are the power span to test ?

BKFlister, 2022-03-15:
@gskjold  Thanks. 
There was LED activity, but the factory reset did not work.
USB Power was plugged in a power bank, but it seems that the power bank "turned off" during the upgrade and the upgrade failed. Earlier I had the HAN cable plugged in, that probably saved me on previous upgrades.

I got hold of a FTDI-adapter and have managed to flash it with the latest 2.1.0
It is now up and running again.

Now I will try to get data in using "Home Assistant" as Payload.

tronde-ams, 2022-03-15:
ESP32 / 42a627e

Real time calculation for hour did not reset when new hour. Same on both units and MQTT

I used 2 kWh between 20:00 and 21:00. Same consumption on all three units (one is 220124.2) I have running at the moment. It did reset at 20:00.

[image]

tronde-ams, 2022-03-15:
The calculation did reset correctly on both units at 22:00. Bar graphs also OK.

What vent wrong at 21?

kabu-kabu, 2022-03-16:
@ArnieO I'm using a Joy-It Node MCU 8266 and a MIKROE-4137 connected to GPIO04. RSSI is on my screenshot: -41 and better; I even turned down TX power. AP is pretty close and wireless is not busy - I do like wired connections.

Restarts are gone with 2.1.0 and I do get plausible data via MQTT. In my situation it's just GUI which is not working as expected.

ArnieO, 2022-03-16:
> @ArnieO I'm using a Joy-It Node MCU 8266 and a MIKROE-4137 connected to GPIO04. RSSI is on my screenshot: -41 and better; I even turned down TX power. AP is pretty close and wireless is not busy - I [...]

OK, so RSSI is good, your Wifi not busy... sounds good. 
This is an unusual problem that I don't believe we've seen before. I'll let @gskjold have some time to think about it. 
In the mean time: Did you try to completely erase flash memory before reflashing? In some cases that has resolved strange issues so might be worth a try.

ArnieO, 2022-03-16:
> Thank you, it looks like this:

@bjornsivertsen Thank you! So the firmware often reads payloads as invalid. Valid payloads start with "7E". 

@gskjold 
Maybe you could add to the first post a request to inform whether hardware is based on ESP8266 or ESP32? 
Would help tracking down if particular issues could be related to ESP8266 (did we reach the end of its capability somewhere?). Any idea what could be causing what @bjornsivertsen is experiencing?

I have for the last days tested an ESP32-based Pow-K board in my Kamstrup. I'll revert to an ESP8266 board with v2.1.0 now firmware and see what happens.

ArnieO, 2022-03-16:
So...

- Norway, Kamstrup 3P TN, non-encrypted.
- Old Pow-K board with HAN on UART0/GPIO3 (meaning hardware serial).
- MQTT active
[image]
GUI is fine, and Telnet seems to confirm nice and stable reading:
[image]

From time to time I see these short payloads that are unreadable. It is not the first time I see it, and suspect this is actually something the meter sends from time to time:
[image]

@gskjold 
Maybe a far fetched theory but... could there be an issue with hardware serial vs software serial when it comes to capturing data? 

I will power up a board that is identical to the one @bjornsivertsen runs (HAN to GPIO12, so software serial) - and see what that gives.

gskjold, 2022-03-16:
> What vent wrong at 21?

I will check the logic here, it probably missed the hourly data.

gskjold, 2022-03-16:
> how to test changing wfi power, what are the power span to test ?

Not sure it is an easy thing to test, but the range is 0-19.5 for esp32 and 0-20.5 for esp8266

gskjold, 2022-03-16:
> This is an unusual problem that I don't believe we've seen before. I'll let @gskjold have some time to think about it.

Unusual problem indeed, I need some time to think about this.

> Maybe you could add to the first post a request to inform whether hardware is based on ESP8266 or ESP32?

Done

> Any idea what could be causing what @bjornsivertsen is experiencing?

Nothing comes to mind.. But it could be either a buffer issue or some sort of timing issue

bjornsivertsen, 2022-03-16:
Is there anything more I can do to collect data or troubleshoot?

gskjold, 2022-03-16:
> Is there anything more I can do to collect data or troubleshoot?

Could you try to open /data.json from your browser and post the content?

bjornsivertsen, 2022-03-16:
> > Is there anything more I can do to collect data or troubleshoot?
> 
> Could you try to open /data.json from your browser and post the content?

{"im":9200,"om":0,"mf":40,"i":0,"e":0,"ri":0,"re":0,"ic":0.000,"ec":0.000,"ric":0.000,"rec":0.000,"u1":0.00,"u2":0.00,"u3":0.00,"i1":0.00,"i2":0.00,"i3":0.00,"f":0.00,"f1":0.00,"f2":0.00,"f3":0.00,"v":3.423,"r":-53,"t":-127.00,"u":63803,"m":16080,"em":1,"hm":1,"wm":1,"mm":0,"me":0,"p":null,"mt":255,"ds":1,"ea":{"x":1.5,"t":5,"h":{"u":0.23,"c":0.00},"d":{"u":4.67,"c":0.00},"m":{"u":4.67,"c":0.00}},"c":1647430665}

gskjold, 2022-03-16:
I think I got it now, try this:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8261776/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8261777/esp8266.zip)

bjornsivertsen, 2022-03-16:
> I think I got it now, try this: [esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8261776/esp32.zip) [esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8261777/esp8266.zip)

You definitely did something right, as all visualisations are showing up now.

[image]

HAN status is green perhaps 90% of the time, when it briefly goes to red and yellow. Here are the HAN logs through Telnet

(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (2b):
(D) 7B 7E
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0D FC 31
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
(D) F9 D0 D0 A3  F6 04 4C A2  0F 66 67 A2
(W) (readHanPort) Unknown data format 7B
(D) (loop) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (491b):
(D) 7E A1 E9 41  03 13 C6 37  E6 E7 00 DB  08 4B 41 4D
(D) 45 01 C2 37  8C 82 01 D0  30 00 0D FC  36 0F 00 00
(D) 00 00 0C 07  E6 03 10 03  0D 06 1E FF  80 00 00 02
(D) 41 0A 0E 4B  61 6D 73 74  72 75 70 5F  56 30 30 30
(D) 31 09 06 01  01 01 08 00  FF 06 00 0B  FE 43 09 06
(D) 01 01 02 08  00 FF 06 00  00 00 00 09  06 01 01 03
(D) 08 00 FF 06  00 00 62 F0  09 06 01 01  04 08 00 FF
(D) 06 00 0B E0  1C 09 06 01  01 00 00 01  FF 06 01 4C
(D) 4F A9 09 06  01 01 01 07  00 FF 06 00  00 01 F8 09
(D) 06 01 01 02  07 00 FF 06  00 00 00 00  09 06 01 01
(D) 03 07 00 FF  06 00 00 00  00 09 06 01  01 04 07 00
(D) FF 06 00 00  01 90 09 06  00 01 01 00  00 FF 09 0C
(D) 07 E6 03 10  03 0D 06 1E  FF 80 00 00  09 06 01 01
(D) 20 07 00 FF  12 00 E4 09  06 01 01 34  07 00 FF 12
(D) 00 E3 09 06  01 01 48 07  00 FF 12 00  00 09 06 01
(D) 01 1F 07 00  FF 06 00 00  00 E3 09 06  01 01 33 07
(D) 00 FF 06 00  00 00 5A 09  06 01 01 47  07 00 FF 06
(D) 00 00 00 00  09 06 01 01  15 07 00 FF  06 00 00 01
(D) 79 09 06 01  01 29 07 00  FF 06 00 00  00 7F 09 06
(D) 01 01 3D 07  00 FF 06 00  00 00 00 09  06 01 01 21
(D) 07 00 FF 12  00 4C 09 06  01 01 35 07  00 FF 12 00
(D) 52 09 06 01  01 49 07 00  FF 12 00 64  09 06 01 01
(D) 0D 07 00 FF  12 00 4E 09  06 01 01 16  07 00 FF 06
(D) 00 00 00 00  09 06 01 01  2A 07 00 FF  06 00 00 00
(D) 00 09 06 01  01 3E 07 00  FF 06 00 00  00 00 09 06
(D) 01 01 16 08  00 FF 06 00  00 00 00 09  06 01 01 2A
(D) 08 00 FF 06  00 00 00 00  09 06 01 01  3E 08 00 FF
(D) 06 00 00 00  00 09 06 01  01 15 08 00  FF 06 00 09
(D) BD 65 09 06  01 01 29 08  00 FF 06 00  02 40 DD 09
(D) 06 01 01 3D  08 00 FF 06  00 00 00 00  D2 D9 F5 78
(D) BC 56 81 1C  AA A5 3F 53  30 96 7E
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0D FC 36
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
[... 94 lines cut ...]
(D) 72 65 22 3A  34 30 31 2C  22 69 63 22  3A 37 38 35
(D) 39 2E 38 37  30 2C 22 65  63 22 3A 30  2E 30 30 30
(D) 2C 22 72 69  63 22 3A 32  35 33 2E 32  38 30 2C 22
(D) 72 65 63 22  3A 37 37 38  32 2E 36 38  30 2C 22 75
(D) 31 22 3A 32  32 38 2E 30  30 2C 22 75  32 22 3A 32
(D) 32 36 2E 30  30 2C 22 75  33 22 3A 30  2E 30 30 2C
(D) 22 69 31 22  3A 32 2E 33  32 2C 22 69  32 22 3A 30
(D) 2E 39 31 2C  22 69 33 22  3A 30 2E 30  30 2C 22 66
(D) 22 3A 30 2E  37 38 2C 22  66 31 22 3A  30 2E 37 37
(D) 2C 22 66 32  22 3A 30 2E  38 32 2C 22  66 33 22 3A
(D) 31 2E 30 30  2C 22 76 22  3A 33 2E 34  32 34 2C 22
(D) 72 22 3A 2D  35 34 2C 22  74 22 3A 2D  31 32 37 2E
(D) 30 30 2C 22  75 22 3A 39  32 34 2C 22  6D 22 3A 31
(D) 33 33 39 32  2C 22 65 6D  22 3A 31 2C  22 68 6D 22
(D) 3A 47
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0D FC 39
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
(D) 3F 71 04 10  5B 71 DC E8  25 4B 9C C6
(W) (readHanPort) Unknown data format 7B

ArnieO, 2022-03-16:
Last version working. I still get relatively frequent (invalid) payloads of varying length starting with "7B 22". 
Always the same start bytes.

gskjold, 2022-03-16:
Are we sure this isnt a serial parity problem? 8N1 vs 8E1 ?

ArnieO, 2022-03-16:
> Are we sure this isnt a serial parity problem? 8N1 vs 8E1 ?

With 90% of payloads ok? If so it is in the code, not in the config setting.

ArnieO, 2022-03-16:
Anyhow; my parity is correct set to 8N1. If I change it to 8E1, this is what I get:
[image]

gskjold, 2022-03-16:
My correct parity is 8E1, if I change to 8N1 it still works. Can't actually see any errors.

gskjold, 2022-03-16:
I've introduced a shared buffer in the code to reduce memory footprint. I should not have used the shared buffer for serial as well.

Hint: decode payload to ascii

tronde-ams, 2022-03-16:
ESP32 / 42a627e / Aidon 3p IT / Norway

Found another discrepancy.

I lost one hourly packet and the graphics showed 0 as expected. When the next hourly packet was read, the graphics updated as expected. It seems like you have improved this because it showed correct value for both the missing hour and the last hour.  

Then to the problem:

The total value of the two missing hours was 5.3 kWh (2.7 + 2.6). Because of this the max value changed from 4.1 / 5 kWh to 4.1 / 10 kWh.
[image]

tronde-ams, 2022-03-16:
> > What vent wrong at 21?
> 
> I will check the logic here, it probably missed the hourly data.

I think it is best to rely only on the internal real time clock for everything related to the real time values. Real time values are very useful for control of equipment, but only if the values are to be trusted. If we miss a few readings from the meter it will not give a significant error, but if we miss the hourly reading the real time values will be rendered more or less useless for control.

kabu-kabu, 2022-03-16:
@ArnieO: I just uploaded new version 53296cc and everything is working as expected.
Kaifa M309 meter is recognized as Sagemcom again, GUI is updated, no restarts and plausible data via MQTT.
For me so far: All good!

BTW: On main page in first graph it says Sagecom and on meter config it says Sage_m_com.

stbitdk, 2022-03-17:
Esp8266 / 53296cc / Kamstrup / Denmark

I just tested ver 53296cc. It worked, but I've got a few errors:

1. Han port flipping between green and red.
2. Log file shows a few "(readHanPort) Unknown data format 7B" 

Logfile attached.

[ams200_53296cc_17mar22.txt](https://github.com/gskjold/AmsToMqttBridge/files/8288604/ams200_53296cc_17mar22.txt)

gskjold, 2022-03-17:
Thank you for testing! "Unknown data format 7B" is a known problem. I just updated first post with new version beafcd3 that should fix this problem :)

bjornsivertsen, 2022-03-17:
We are not out of the woods yet. Though it rarely leaves green status now, it still jumps to red and yellow occasionally.

Unknown data format A1?

```
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (491b):
(D) 7E A1 E9 41  03 13 C6 37  E6 E7 00 DB  08 4B 41 4D
(D) 45 01 C2 37  8C 82 01 D0  30 00 0E 26  F7 02 69 E4
(D) AB C0 35 43  A2 1F B3 03  D0 30 74 80  05 E6 FA 7C
(D) DF 37 3A B1  E8 3C 86 82  A5 19 AB 93  98 C3 2D 4A
(D) 37 DF D1 07  55 89 8B 03  6D 13 07 35  0B 32 31 0F
(D) F7 FE 5A 06  4F 54 13 44  3E 8C 22 B0  D1 2F CF C8
(D) 4C FE 0B 58  67 A0 8B 7D  E1 6A CB 64  E8 9E 8D 56
(D) DC E1 31 55  87 70 97 E9  09 1A D5 C1  E1 F8 45 7E
(D) 33 07 6F 60  8C FB 87 CF  3C 4E 41 52  4E CF C6 8B
(D) B3 68 B0 7E  6E 15 D5 16  2D 15 1D 56  A0 98 7B 2A
(D) AF AD FF 7B  00 83 C6 6C  C3 5A AB E7  49 4E AB C1
(D) 08 F0 6A 10  C2 A6 FE 28  73 69 4C 6B  4F 17 2A A5
(D) 99 AF D6 F6  46 BF 85 2F  B7 6E 26 74  08 92 E0 BD
(D) 2D 05 DC 90  21 AF E8 BA  B4 3F 54 E0  45 FE 4D FF
(D) 04 DF D3 4A  44 BE 29 54  A0 DD 13 F7  3A 12 D1 81
(D) 24 4E D2 DE  7F FF 6D 6D  BE 18 90 77  8B 8E 9D 1E
(D) 91 2E 32 17  58 E2 B5 B5  64 6A A3 55  B2 0D 2D F3
(D) 5F 11 EA EB  2F 17 2B 71  3B 99 69 62  EB 94 8A D3
(D) 28 05 B2 3A  45 39 1C D2  4E A5 4F 59  3E 6A BD 90
(D) C7 B6 F0 EA  AC D9 A0 B8  A4 B3 BB 51  28 90 0C 9A
(D) 41 93 DB F9  01 C1 E2 EE  F3 F3 4D F5  50 19 AA 8F
(D) F8 79 ED B5  E9 2C F2 23  4F F3 B2 29  19 E9 53 11
(D) F0 97 61 15  C8 BE F3 0D  DF F3 83 86  A1 90 B9 35
(D) A6 22 BE AD  20 BB 04 A4  B0 1A 54 B3  7D 78 9C B6
(D) 31 40 B8 AF  36 AD 91 F0  3C 0A 3E C7  ED C6 30 AB
(D) 1F 65 1C ED  CB 59 3C F3  94 3D B7 F8  7A 85 7E A2
(D) C1 8A 6B AE  A7 76 BE 23  3D E6 73 22  E0 F7 96 86
(D) 6A 5E 0B 8B  AB 7B 8D 74  FB 8E 3F C7  0C A6 56 7A
(D) D4 16 BE 79  D7 72 10 C2  CA B3 99 27  65 7F D0 92
(D) BD 9F 44 E5  1B D4 34 7C  3A 8F 0A 59  C9 F2 DD EC
(D) 53 B1 33 B4  2B 37 A0 13  FD 7E 7E
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0E 26 F6
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
(D) D1 C7 EC 7D  E5 83 CC 43  29 4E C9 BE
(W) (readHanPort) Frame checksum error
(D) (readHanPort) Frame dump (1b):
(D) A1
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0E 26 F6
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
(D) D1 C7 EC 7D  E5 83 CC 43  29 4E C9 BE
(W) (readHanPort) Unknown data format A1
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (491b):
(D) 7E A1 E9 41  03 13 C6 37  E6 E7 00 DB  08 4B 41 4D
(D) 45 01 C2 37  8C 82 01 D0  30 00 0E 26  FA 0F 00 00
(D) 00 00 0C 07  E6 03 11 04  13 1F 0A FF  80 00 00 02
(D) 41 0A 0E 4B  61 6D 73 74  72 75 70 5F  56 30 30 30
(D) 31 09 06 01  01 01 08 00  FF 06 00 0C  02 89 09 06
(D) 01 01 02 08  00 FF 06 00  00 00 00 09  06 01 01 03
(D) 08 00 FF 06  00 00 62 F0  09 06 01 01  04 08 00 FF
(D) 06 00 0B E4  82 09 06 01  01 00 00 01  FF 06 01 4C
(D) 4F A9 09 06  01 01 01 07  00 FF 06 00  00 01 9B 09
(D) 06 01 01 02  07 00 FF 06  00 00 00 00  09 06 01 01
(D) 03 07 00 FF  06 00 00 00  00 09 06 01  01 04 07 00
(D) FF 06 00 00  01 AA 09 06  00 01 01 00  00 FF 09 0C
(D) 07 E6 03 11  04 13 1F 0A  FF 80 00 00  09 06 01 01
(D) 20 07 00 FF  12 00 E2 09  06 01 01 34  07 00 FF 12
(D) 00 E0 09 06  01 01 48 07  00 FF 12 00  00 09 06 01
(D) 01 1F 07 00  FF 06 00 00  00 F4 09 06  01 01 33 07
(D) 00 FF 06 00  00 00 20 09  06 01 01 47  07 00 FF 06
(D) 00 00 00 00  09 06 01 01  15 07 00 FF  06 00 00 01
(D) 7B 09 06 01  01 29 07 00  FF 06 00 00  00 20 09 06
(D) 01 01 3D 07  00 FF 06 00  00 00 00 09  06 01 01 21
(D) 07 00 FF 12  00 47 09 06  01 01 35 07  00 FF 12 00
(D) 30 09 06 01  01 49 07 00  FF 12 00 64  09 06 01 01
(D) 0D 07 00 FF  12 00 45 09  06 01 01 16  07 00 FF 06
(D) 00 00 00 00  09 06 01 01  2A 07 00 FF  06 00 00 00
(D) 00 09 06 01  01 3E 07 00  FF 06 00 00  00 00 09 06
(D) 01 01 16 08  00 FF 06 00  00 00 00 09  06 01 01 2A
(D) 08 00 FF 06  00 00 00 00  09 06 01 01  3E 08 00 FF
(D) 06 00 00 00  00 09 06 01  01 15 08 00  FF 06 00 09
(D) C0 E0 09 06  01 01 29 08  00 FF 06 00  02 41 A9 09
(D) 06 01 01 3D  08 00 FF 06  00 00 00 00  2A 01 C3 12
(D) CD 34 65 0F  91 2C 75 45  29 C6 7E
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0E 26 FA
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
(D) 2A 01 C3 12  CD 34 65 0F  91 2C 75 45
(D) (readHanPort) Valid data, start at byte 47
(V) (EnergyAccounting) Adding 0.0046 kWh
(V) (EnergyAccounting)  calculating threshold, currently at 0
(V) (EnergyAccounting)  new threshold 0
(D) (AmsWebServer)Serving /data.json over http...
```

bjornsivertsen, 2022-03-17:
It did solve the A1 error. Thank you!

Still getting a bit of yellow status, about the same time Telnet says there is a Checksum error.

I mean it works, but the information might be valuable to you. :)

```
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (491b):
(D) 7E A1 E9 41  03 13 C6 37  E6 E7 00 DB  08 4B 41 4D
(D) 45 01 C2 37  8C 82 01 D0  30 00 0E 27  5E 38 B6 82
(D) 3A 3B 1B 9C  48 83 27 8E  CD EE 45 9C  09 C1 4A BC
(D) EB 03 1A 67  D4 44 46 D3  CF 2A 57 BF  1D 4F 9D 1E
(D) B7 5C A8 F1  60 9D 68 A7  3F 67 48 CD  29 42 BE 3A
(D) 33 92 CA 6E  01 0D 7E 98  28 E2 4B C3  78 6C 45 BA
(D) 27 96 39 74  BF D8 72 95  B3 07 61 63  F4 5A 01 25
(D) E6 73 D9 12  FB 16 29 02  D7 F5 80 75  7C EB 0F C6
(D) 43 A6 09 9D  51 34 66 84  23 FF 4B DC  D5 CC B0 78
(D) 4E BC 5C 9C  2F 84 02 DD  7E F1 F8 0D  E8 21 68 46
(D) 71 3F 85 D7  1D D0 3C 11  18 60 76 D8  9C 28 1C 66
(D) FE 7D 7A 38  C5 9F 09 76  DE C3 C5 72  03 CF DC 9E
(D) B9 73 52 71  F8 49 31 9E  06 8E 4A AB  7D A0 AF 9D
(D) 39 4E 9F 7A  98 B7 86 A4  A9 14 3A 0A  D8 5F 9B E0
(D) A6 CE DD 56  CA F6 64 6B  C2 79 8A 16  67 71 7C 88
(D) 03 A2 64 62  DC 9E D8 96  AF 60 B9 8A  4F C1 90 51
(D) FD B0 3E 40  55 D1 CE 8D  38 C3 46 E7  17 B3 32 14
(D) DC 43 32 50  30 98 18 87  44 8A DD 35  DE 94 87 C4
(D) A0 4C 0B 27  35 40 2E F7  70 90 04 90  D0 B2 89 C0
(D) 41 9C 41 9F  7A 84 0C 62  5E 59 87 89  6C A6 2B 75
(D) C9 94 3F 34  AC 51 13 5E  55 D3 D1 0C  78 5C 37 12
(D) 59 AA 26 C3  78 08 41 39  7B 98 CB C4  AA BF 83 2B
(D) B7 E1 FB 0D  04 F2 1F B9  E1 B8 21 5A  8E 07 04 3C
(D) B7 DC C8 BD  91 15 CB DA  25 35 13 C3  15 2C 1F 8F
(D) D8 E0 04 4F  0B 4B C7 9F  45 42 8A 69  1E 94 CF 9D
(D) D8 CD 9E 2A  B1 CC CF C2  6F CA DC BC  7B 19 86 36
(D) 57 35 BB 4E  7C FF 70 88  78 21 97 8C  47 CB D6 3E
(D) 4F A9 DC D8  1C 0D 4A C9  71 48 74 72  0B A4 CF 0D
(D) 54 7A 40 C6  70 B8 55 39  7A D6 CC DA  5E F3 FC 3C
(D) 91 F3 B4 20  23 2C 80 B4  79 41 D5 04  E6 7A 40 93
(D) A8 D6 03 3E  1F D8 96 3E  F0 19 7E
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 C2 37 8C
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 C2 37 8C  00 0E 27 5D
(D) (readHanPort) Additional authenticated data:
(D) 30 09 DF 55  4F A7 4E C8  92 27 B3 7D  32 11 77 1B
(D) 97
(D) (readHanPort) Authentication tag:
(D) 26 55 FC 0B  A5 ED 6B B7  9E 79 13 E1
(W) (readHanPort) Frame checksum error
(D) (AmsWebServer)Serving /data.json over http...
```

gskjold, 2022-03-17:
Cool! Although it should be possible to receive all data flawlessly (in a perfect world), random checksum errors are not uncommon. There are a number of things that could cause this, but I think we should be happy if it mostly works :)

bjornsivertsen, 2022-03-17:
I am very pleased!

Now just to get the auto discovery working for HA, which I am also struggling with. But that is a discussion for another issue. ;)

gskjold, 2022-03-18:
Some adjustments made for HA discovery. Also added multiplier config for those who have CT meters or have a format which is interpreted with incorrect scaling. First post updated.

enordli, 2022-03-18:
I tested the latest version, Aidon 6540 with current transformer, powerfactor 40
I have hardcoded the powerfaktor before, but not anymore. It still works with 50 meter old telephone cable between the POW-U and energy-meter
The HA discovery looks ok now
Backup of configuration is nice, backup of history data would be nice too, i lost mine when i tried to go back to the previous firmware.

[image]

[image]

gskjold, 2022-03-18:
Beautiful!

Graph values should be included in config file download, and should also restore if you upload. File is human readable, so you should find "dayplot" and "monthplot" in the bottom of the file.

enordli, 2022-03-18:
I made a backup before i lost the history.
so i made a new backup with of the latest version and replaced the 3 last lines from the first backup.
I did a restore from the config, but it loaded and mabe restarted, and i restarted it again, but noting changed, still missing history.  
[configfileltest-cfg.log](https://github.com/gskjold/AmsToMqttBridge/files/8307409/configfileltest-cfg.log)

[image]

i tried to backup only Thresholdsdata  and replased the 3 lines with the backup data. did a upload but no luck

[26 more replies]