UtilitechAS/amsreader-firmware #248: ESP32 decryption failed (mbed lib)

tyler123durden, 2022-03-17:
**Describe the bug**
Decryption of payload fails on ESP32 devices with error HDLC_ENCRYPTION_DECRYPT_FAILED. Same release with ESP8266 works but of course ESP doesn't use mbed enryption library. 
Decrypted payload looks perfectly fine one both platforms. 

Unfortunately I'm not able to set up the development environment to fix this issue, I get errors compiling code but maybe my platformIO setup is screwed up because it complains about not finding xtensa compiler. 

```
            success = mbedtls_gcm_auth_decrypt(&m_ctx, sizeof(cipher_text), config->initialization_vector, sizeof(config->initialization_vector),
                config->additional_authenticated_data, aadlen, config->authentication_tag, authkeylen,
                cipher_text, (unsigned char*)(ptr));
            if (authkeylen > 0 && success == MBEDTLS_ERR_GCM_AUTH_FAILED) {
                return HDLC_ENCRYPTION_AUTH_FAILED;
            } else if(success == MBEDTLS_ERR_GCM_BAD_INPUT) {
                return HDLC_ENCRYPTION_DECRYPT_FAILED;        // <<<<<<--------- error here
            }
```
Erroneous ESP32 log: 
`(I) (readHanPort)(C1) Intermediate segment
(D) (readHanPort)(C1) Frame dump (263b):
(D) 68 01 01 68  53 FF 00 01  67 DB 08 53  41 47 59 05 

(D) EA 7A 8E 81  F8 20 00 00  3B FB 4E 9D  51 05 9B 50 

(D) 51 88 1C 69  DE 0B AE A9  B3 A1 18 91  90 05 2D 43 

(D) 34 51 7E 45  33 B2 04 C7  B2 B8 0F 5C  5B 28 C4 89 

(D) 43 86 8E C4  89 13 1F A0  11 9D FB BB  9A DC 05 43 

(D) FE 69 3A 0E  76 8B 96 0F  0F F3 E9 E4  30 76 FA 46 

(D) 43 A1 55 8B  7C 04 7F CD  D6 70 E4 2C  E4 87 E7 81 

(D) 74 D1 85 8A  D2 10 28 EE  BE DF 59 15  3E 62 DC 26 

(D) 9C 3B FD D3  23 B3 1D CB  16 A6 7A CE  50 02 98 45 

(D) 95 1F A4 A8  05 1A D1 D7  20 C2 CC 62  48 DD EC C0 

(D) C2 02 C8 DA  97 01 84 22  ED 1B D3 89  26 E3 32 E4 

(D) A5 94 1E A0  2F BE 9A C2  72 D9 22 BD  C2 D3 BC D9 

(D) 5B 70 7D AE  17 41 5D 34  60 25 AF 9F  48 F3 6E 66 

(D) 69 7D C8 E4  E1 79 55 6E  78 16 E0 ED  46 6D 5D 33 

(D) B3 64 1F 75  02 1F 7A CE  21 B4 93 4D  7D 88 24 5C 

(D) 4E D9 99 04  63 98 FC 86  0C 0D 03 6C  99 02 CD 33 

(D) D6 DD D6 BE  C1 17 16 

(I) (readHanPort)(C1) Final segment
(D) (readHanPort)(C1) Frame dump (19b):
(D) 68 0D 0D 68  53 FF 11 01  67 E0 63 67  8E A2 6E 7B 

(D) 5B E9 16 

(D) (readHanPort)(C1) Frame dump (271b):
(D) 68 00 00 68  53 FF 10 01  67 DB 08 53  41 47 59 05 

(D) EA 7A 8E 81  F8 20 00 00  3B FB 0F 80  0B E2 E3 0C 

(D) 07 E6 03 11  04 0D 35 23  00 FF C4 02  02 23 09 0C 

(D) 07 E6 03 11  04 0D 35 23  00 FF C4 02  09 06 01 00 

(D) 01 08 00 FF  06 00 05 3B  56 02 02 0F  00 16 1E 09 

[... 214 lines cut ...]

(D) (readHanPort) Initialization vector:
(D) 53 41 47 59  05 EA 7A 8E  00 00 49 2F  

(D) (readHanPort) Additional authenticated data:
(D) 20 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 

(D) 00 

(D) (readHanPort) Authentication tag:
(D) 00 00 00 00  00 00 00 00  00 00 00 00  

(D) (readHanPort) Valid data, start at byte 44
(V) (AmsDataStorage) Time is: 1647538514`

**Hardware information:**
 - Meter: [e.g. Aidon]
 - AMS reader: ESP32 DevkitC
 - M-bus adapter (if applicable): Mikro Elektronike MBus Slave

**Relevant firmware information:**
 - Version: 2.0.11

**Additional context**
Add any other context about the problem here.

ArnieO, 2022-03-18:
Did you try the pre compiled code available here? 
https://github.com/gskjold/AmsToMqttBridge/releases

tyler123durden, 2022-03-18:
Yes, I'm using compiled release version 2.0.11 and also 2.0.5 I think. Then I tried to setup the dev environment but got to strange xtense compile not found error.

tyler123durden, 2022-03-20:
I've got a development setup now up an running and as far as I understand mbetls_gcm_auth_decrypt will return an MBEDTLS_ERR_GCM_BAD_INPUT error when no authentication data is provided. 
Intern in mbedtls_gcm_finish this will trigger: 
```
    if( tag_len > 16 || tag_len < 4 )
        return( MBEDTLS_ERR_GCM_BAD_INPUT );
```
because tag_len is 0, which is authkeylen from our input. 

So we could either ignore authentication related errors or use dedicated decryption calls without authentication stuff. The below version works for me on ESP32
```
        #elif defined(ESP32)
            uint8_t cipher_text[len - authkeylen - 5];
            memcpy(cipher_text, ptr, len - authkeylen - 5);

            mbedtls_gcm_context m_ctx;
            mbedtls_gcm_init(&m_ctx);
            int success = mbedtls_gcm_setkey(&m_ctx, MBEDTLS_CIPHER_ID_AES, config->encryption_key, 128);
            if (0 != success) {
                return HDLC_ENCRYPTION_KEY_FAILED;
            }
            if (0<authkeylen) {
                success = mbedtls_gcm_auth_decrypt(&m_ctx, sizeof(cipher_text), config->initialization_vector, sizeof(config->initialization_vector),
                    config->additional_authenticated_data, aadlen, config->authentication_tag, authkeylen,
                    cipher_text, (unsigned char*)(ptr));
                if (authkeylen > 0 && success == MBEDTLS_ERR_GCM_AUTH_FAILED) {
                    return HDLC_ENCRYPTION_AUTH_FAILED;
                } else if(success == MBEDTLS_ERR_GCM_BAD_INPUT) {
                    return HDLC_ENCRYPTION_DECRYPT_FAILED;
                }
            }
            else {
                mbedtls_gcm_starts(&m_ctx, MBEDTLS_GCM_DECRYPT, config->initialization_vector, sizeof(config->initialization_vector),NULL, 0);
                if (0 != success) {
                    return HDLC_SOME_ERROR_TBD1;
                }
                success = mbedtls_gcm_update(&m_ctx, sizeof(cipher_text), cipher_text, (unsigned char*)(ptr));
                if (0 != success) {
                    return HDLC_SOME_ERROR_TBD2;
                }
            }
            mbedtls_gcm_free(&m_ctx);
        #endif
```

ArnieO, 2022-03-21:
@gskjold : Help me remember; I had the impression decryption has been verified some time ago on ESP32-based hardware?

gskjold, 2022-03-21:
@ArnieO Yes, but only for encryption with both keys.

@tyler123durden Thank you for providing input to a solution, I will try to include this in the upcoming 2.1 release if possible.

tyler123durden, 2022-03-21:
great. Also great work and project!

I've seen that I forgot to check error from mbedtls_gcm_starts. 
Another minor drawback in the above code ist that in error cases mbedtls_gcm_free will not get called, which may lead to a memory leak.

closed 2022-03-26 by gskjold