UtilitechAS/amsreader-firmware #249: Add list of values to MQTT config

kabu-kabu, 2022-03-18:
Would it be possible to add a list of values to MQTT config page? So you can easily control which values are transferred?

As I do not need most of the values my receiving unit has to discard most of the transferred messages. It would be more reasonable not to transfer them at all.

ArnieO, 2022-03-18:
Thank you for you suggestion, but...

The saved transmission would be extremely small considering the amount of data that traverses a wifi network. 
And you will anyway have to parse the JSON string on the receiving side, so the gain there is also negligible - in my opinion. 

As we are already approaching the limit of what can be squeezed into an ESP8266, we need to be restrictive to which features are added, and make sure they are really needed. Thank you for your understanding.

gskjold, 2022-03-18:
I have though about this in the past, but like @ArnieO says we are a bit squeezed for resources. Besides, it doesn't cost much to send them, and discarding is easy. But thank you for the suggestion.

closed 2022-03-18 by gskjold