UtilitechAS/amsreader-firmware #250: hardcode the wifi ssid and password

ttqf, 2022-03-21:
**Describe your problem**
I am running a generic esp8266, more specifically esp-01s. I am able to install, but when I setup the wifi, I get an error message about not able to save configuration. is it possible to hardcode the wifi ssid and password?

is it possible to hardcode wifi ssid and password? if so how and where?
**Hardware information:**
 - Meter: Aidon
 - AMS reader: [e.g. Pow-U, ESP32 etc]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version:n/a
 - MQTT: n/a
 - HAN GPIO: hardware pins
 - Temperature sensors: none

ArnieO, 2022-03-21:
I don't think that would help you. 

I believe there are limitations in ESP-01S that will prevent your setup from working. As far as I remember, the ESP-01S only has 1 MB flash memory, while ESP-12E/F have 4 MB.

ttqf, 2022-03-21:
i am aware of the limitations of the esp-01. this is just a testing idea since I have one laying around :)