UtilitechAS/amsreader-firmware #252: custom hourly price-points

jonaslevring, 2022-03-22:
**Is your feature request related to a problem? Please describe.**
I would like to expose the final electricity price for endusers. Final price that is based on various elements like. Baseprice, Distribution cost, PSO fee, Taxes, VAT.
Most public endpoints like eg. Nordpool only provice the baseprice, where the additional elements are more local to the specific installation.

**Describe the solution you'd like**
I would like the possibility to poll a endpoint every hour and get the next XX hours of prices seperated into the resolution available by the source. The source would most likely be provided by a webserver or similar with custom integration and pricepoints. So overall a bit more advanced solution to setup.
In the proposed format I have made it HH:MM, to be able to support any higher resolution in future.

**Describe alternatives you've considered**
Cant really see many alternatives than educating my family in adding distibutioncost, fees, taxes and finally add VAT :-) i think the actual calculation would be more relevant.

**Additional context**
I have attached an example of how the json format could look like.
[Example.json.txt](https://github.com/gskjold/AmsToMqttBridge/files/8323576/Example.json.txt)

gskjold, 2022-03-23:
Thank you for your input. I agree that this would be a nice feature, but it has some challenges. Seeing that this firmware is used throughout Europe, there are many many different compositions of electricity tariffs. For example there are countries that have different tariff during night time, which adds another level of complexity. I am open for input on this matter, preferably from more countries. I assume your are from Denmark?

If we avoid including fixed monthly costs, Something like this might cut it:
[image]

jonaslevring, 2022-03-23:
Yes.. from Denmark..
Your proposed solution could be a very good starting point.. I have a few additional inputs:

- It should be possible to define the formula for calculation.. Especially due to taxes and fees which may not be related to all price tariffs.. It should enough to provide it as a single formula. Eg. of expressions: ((1*4)+(2+3)) would result in: (0,54*1,25)+(0,75+0) (During daytime) and (0,54*1,25)+(0+0,45) (During nighttime)
- Each of the price elements and total cost incl and excl VAT should be posted to MQTT as well (So calculations does'nt need to be made multiple places)
- In Denmark we see different fees during winter and summer times (based on dates). I dont know if its the same for other countries ?
- And in addition I dont know if there are different tariffes during a week.. Eg. weekends are generally cheaper

ArnieO, 2022-03-23:
Thank you @jonaslevring for many good ideas. 👍 
Such a formula interpreter could be a nice thing to have!
Probably better to use letters as identifiers for each parameter; but that is an easy detail compared to coding the formula interpreter...

nvanggaard, 2023-01-05:
@gskjold i think your idea using with lines where you can define for different timeslots. I agree that it only solves the problem "per day", so if the price changes during summer/winther time, or weekday / weekends, that isn't solved using this method. 

I think it could be solved by using crontab expressions instead of Hour start and Hour End. How ever, that increases the complexity of the user interface

gskjold, 2023-01-05:
Thanks for the input. This is something that needs a bit of planning. Unless a reasonable and almost finished implementation is presented, it is unlikely I will get around to this any time soon unfortunately...

svendhoest, 2023-01-10:
I would like to add that, in DK the time based tariffs are different for summer and winter.

gskjold, 2023-12-25:
Its taken a while, but here is my proposed solution, which I believe will work for most cases:
[image]

You can find it via the the "Configure price" link in config:
[image]

If you want to test this, here is a build of v2.3 branch. **WARNING**, this is a development build, please make sure you are able to re-flash stable firmware via USB if necessary. Also, back up your config before flashing.

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13766891/esp32.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13766892/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13766893/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13766894/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13766895/esp8266.zip)

jonaslevring, 2023-12-28:
Looks promising @gskjold

Can you describe how calculation is done? And can the equation be customized?

MrAdam, 2023-12-28:
That looks really good @gskjold!
Is it possible to reorder the lines, or would one have to delete and re-add in the right order?

MrAdam, 2023-12-28:
I just installed the firmware and tried setting up the custom pricing matching my provider, and have a few comments.
1.  I'd like to have more than two decimals of precision for the prices, as e.g. the government tariff for me is 0.697. After adding a several tariffs, and taxes, the number can vary quite a bit depending on how careful one does the rounding.
2. It doesn't seem like the "Future energy prices" on the dashboard reflects the custom price - looks like it shows the price without the configuration

gskjold, 2023-12-28:
I forgot to say that the value is given in 1/100th of the currency, thats why I kept the two decimal precision.

Good point regarding tariff changing through the year, I will have to think about that one...

MrAdam, 2023-12-29:
> I forgot to say that the value is given in 1/100th of the currency, thats why I kept the two decimal precision.
> 
> Good point regarding tariff changing through the year, I will have to think about that one...

Right, I'll try fixing my numbers then - maybe it should be more clear, or maybe the units should match what is used elsewhere to avoid potential mistakes?

Feel free to write if you have a new version you need tested, I'm more than happy to test it out.

ClausMelander, 2024-01-05:
Just tried the esp32s2 firmware 9ff6cf3 and trying to set up more correct prices. 
The firmware seams stable and functioning as planned :)
I'm on a variable price based on the DK1 with some fixed extras and some % are added to the DK1 price, but I cannot find out how to use the DK1 price in the calculations, or if it's even possible.

ClausMelander, 2024-01-05:
> Just tried the esp32s2 firmware [9ff6cf3](https://github.com/UtilitechAS/amsreader-firmware/commit/9ff6cf355fc59e440039778f50bd36f1701229a0) and trying to set up more correct prices. The firmware seams stable and functioning as planned :) Just 2 points... Is it possible to subtract from export price? it's not possible to insert negative numbers... Maximum value seams to be 65.53 units, but between 17-20 the fixed add is 98.00 units (Denmark)

also just noted that the real-time graph seams to only function for about 5 min.
[image]

gskjold, 2024-01-06:
> Can you describe how calculation is done? And can the equation be customized?

I have not added support for equations, only additions. I understand that this would add flexibility to the solution, but it would also significantly increase code complexity, and I am just not going there.

> maybe it should be more clear, or maybe the units should match what is used elsewhere to avoid potential mistakes?

Fair point, I have changed this to 1/1 currency instead, makes more sense.

> Is it possible to subtract from export price?

Can you please explain the use case?

> Maximum value seams to be 65.53 units, but between 17-20 the fixed add is 98.00 units (Denmark)

I have changed this to max 99.9999 (DKK, not øre), which should be more than enough.

Attached is a new build, where start and end day/month is possible, as well as changes outlined above. Please note that I have not taken into account existing data, so your price config will get messed up when installing this version, forcing you to setup everything from scratch.

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13850793/esp32.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13850794/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13850795/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13850796/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13850798/esp8266.zip)

[image]

ClausMelander, 2024-01-07:
> > Is it possible to subtract from export price?
> 
> Can you please explain the use case?
Yes... the company that buys my export uses the below subtractions to the market price:
Vindstød (example):
Børsens spotpris (*) 125,0000 øre/kWh
Energinets indfødningstarif -0,3750 øre/kWh
Energinets balancetarif -0,1600 øre/kWh
Netselskabets indfødningstarif -5,20 øre/kWh
Vindstøds balancetarif -1,0000 øre/kWh
Din kWh pris	118,265 øre/kWh
So I would like to subtract 6,735 øre/kWh to make the export price correct.

GaupeLnx, 2024-01-12:
Got pointed to this thread. when i mentioned the government support that is in Norway and already there for the 3rd year. So as for its temporary, that is just how one looks at it.  I predict this will continue for more years to come. I still would like to plea for a possibility to instead of putting in a multiplier (like for vat) to make it possible for a formula. (even if if would require for the user to give the input in code.  )

haldi4803, 2024-01-22:
> Attached is a new build, where start and end day/month is possible, as well as changes outlined above. Please note that I have not taken into account existing data, so your price config will get messed up when installing this version, forcing you to setup everything from scratch.

Uhm...
Seeing as this has bootloader.bin boot_app0.bin firmware.bin flash.sh and partitions.bin included.
Is the Firmware.bin OTA compatible with v2.2.28 ?
Can i just press the "select Firmware file for Upgrade" Button in the web interface and flash this test version?

P.S Here is a bill from Switzerland, CKW.

haldi4803, 2024-01-22:
[image]

haldi4803, 2024-01-22:
[image]

ArnieO, 2024-01-22:
> Is the Firmware.bin OTA compatible with v2.2.28 ?

Yes, that is the one. The others are included by default in case someone needs to flash a "virgin" ESP32.

> Can i just press the "select Firmware file for Upgrade" Button in the web interface and flash this test version?

Yes

haldi4803, 2024-01-22:
@ArnieO 

Price Config is not Saved.
Not sure if i'm doing something wrong or what.
Activated Verbose logging, here is the Telnet Log:
i click on Configure price, get to the new Homepage. Press the Blue add Button. Name it, press on Save and get thrown back into the default configuration page. Price not saved.

<details>
```
*** Remote debug - over telnet - for ESP32 - version 3.0.5
* Host name: ams-9686 IP:192.168.1.118 Mac address:D4:F9:8D:6E:86:96
* Free Heap RAM: 118956
* ESP SDK version: 4.4.5.230722
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter

* Please type the command and press enter to execute.(? or h for this help)
***
(D) Serving /data.json over http...
(V) HDLC frame:
(V) 7E A1 CB 2B  21 13 57 E7  E6 E7 00 0F  00 00 00 00
(V) 0C 07 E8 01  16 01 0F 0E  00 FF 80 00  00 02 41 0A
(V) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(V) 06 01 01 01  08 00 FF 06  00 01 3D 0F  09 06 01 01
(V) 02 08 00 FF  06 00 00 00  00 09 06 01  01 03 08 00
(V) FF 06 00 00  00 04 09 06  01 01 04 08  00 FF 06 00
(V) 00 E3 47 09  06 01 01 00  00 01 FF 06  02 03 C3 6C
(V) 09 06 01 01  01 07 00 FF  06 00 00 01  84 09 06 01
(V) 01 02 07 00  FF 06 00 00  00 00 09 06  01 01 03 07
(V) 00 FF 06 00  00 00 00 09  06 01 01 04  07 00 FF 06
(V) 00 00 01 10  09 06 00 01  01 00 00 FF  09 0C 07 E8
(V) 01 16 01 0F  0E 00 FF 80  00 00 09 06  01 01 20 07
(V) 00 FF 12 00  F0 09 06 01  01 34 07 00  FF 12 00 EF
(V) 09 06 01 01  48 07 00 FF  12 00 F0 09  06 01 01 1F
(V) 07 00 FF 06  00 00 00 B9  09 06 01 01  33 07 00 FF
(V) 06 00 00 00  05 09 06 01  01 47 07 00  FF 06 00 00
(V) 00 18 09 06  01 01 15 07  00 FF 06 00  00 01 77 09
(V) 06 01 01 29  07 00 FF 06  00 00 00 07  09 06 01 01
(V) 3D 07 00 FF  06 00 00 00  06 09 06 01  01 21 07 00
(V) FF 12 00 57  09 06 01 01  35 07 00 FF  12 00 3F 09
(V) 06 01 01 49  07 00 FF 12  00 0B 09 06  01 01 0D 07
(V) 00 FF 12 00  51 09 06 01  01 16 07 00  FF 06 00 00
(V) 00 00 09 06  01 01 2A 07  00 FF 06 00  00 00 00 09
(V) 06 01 01 3E  07 00 FF 06  00 00 00 00  09 06 01 01
(V) 16 08 00 FF  06 00 00 00  00 09 06 01  01 2A 08 00
(V) FF 06 00 00  00 00 09 06  01 01 3E 08  00 FF 06 00
(V) 00 00 00 09  06 01 01 15  08 00 FF 06  00 01 18 E9
(V) 09 06 01 01  29 08 00 FF  06 00 00 0A  BC 09 06 01
(V) 01 3D 08 00  FF 06 00 00  19 69 21 0F  7E
(V) LLC frame:
(V) E6 E7 00 0F  00 00 00 00  0C 07 E8 01  16 01 0F 0E
[... 449 lines cut ...]
(V) 00 FF 06 00  00 01 0F 09  06 00 01 01  00 00 FF 09
(V) 0C 07 E8 01  16 01 0F 0E  1E FF 80 00  00 09 06 01
(V) 01 20 07 00  FF 12 00 EF  09 06 01 01  34 07 00 FF
(V) 12 00 EF 09  06 01 01 48  07 00 FF 12  00 F0 09 06
(V) 01 01 1F 07  00 FF 06 00  00 00 B3 09  06 01 01 33
(V) 07 00 FF 06  00 00 00 06  09 06 01 01  47 07 00 FF
(V) 06 00 00 00  18 09 06 01  01 15 07 00  FF 06 00 00
(V) 01 66 09 06  01 01 29 07  00 FF 06 00  00 00 0B 09
(V) 06 01 01 3D  07 00 FF 06  00 00 00 06  09 06 01 01
(V) 21 07 00 FF  12 00 56 09  06 01 01 35  07 00 FF 12
(V) 00 4E 09 06  01 01 49 07  00 FF 12 00  0B 09 06 01
(V) 01 0D 07 00  FF 12 00 51  09 06 01 01  16 07 00 FF
(V) 06 00 00 00  00 09 06 01  01 2A 07 00  FF 06 00 00
(V) 00 00 09 06  01 01 3E 07  00 FF 06 00  00 00 00 09
(V) 06 01 01 16  08 00 FF 06  00 00 00 00  09 06 01 01
(V) 2A 08 00 FF  06 00 00 00  00 09 06 01  01 3E 08 00
(V) FF 06 00 00  00 00 09 06  01 01 15 08  00 FF 06 00
(V) 01 18 EC 09  06 01 01 29  08 00 FF 06  00 00 0A BD
(V) 09 06 01 01  3D 08 00 FF  06 00 00 19  69
(V) DLMS
(V) (EnergyAccounting) Adding 0.0010 kWh import
(V) (EnergyAccounting)  calculating threshold, currently at 0
(V) (EnergyAccounting)  new threshold 0
```
</details>

ArnieO, 2024-01-22:
> Price Config is not Saved.

OK, this is a test version - there might be a bug. 
@gskjold , can you look into this?

(And I think @haldi4803 forgot to include the telnet debug log?)

gskjold, 2024-01-25:
I am unable to reproduce problem with prices not being saved. @haldi4803 please provide what data you enter into the fields before saving.

New firmware, subtraction now available:
[image]

[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051033/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051034/esp32.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051035/esp8266.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051036/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051038/esp32s2.zip)

ClausMelander, 2024-01-28:
> I am unable to reproduce problem with prices not being saved. @haldi4803 please provide what data you enter into the fields before saving.
> 
> New firmware, subtraction now available: ![Screenshot from 2024-01-25 12-32-35](https://private-user-images.githubusercontent.com/4446828/299643334-7eb927a5-3089-41b7-be2b-e74c2e3817a1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY0NTg1MDMsIm5iZiI6MTcwNjQ1ODIwMywicGF0aCI6Ii80NDQ2ODI4LzI5OTY0MzMzNC03ZWI5MjdhNS0zMDg5LTQxYjctYmUyYi1lNzRjMmUzODE3YTEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMjhUMTYxMDAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTIxYzY0ZDAzNjUyYzgwMGQwZGVlNTcxNGNjZDVjMTkyMzYxZjE2NGE1OGMyNGZmNDllMmI2YTIxZmU3YjhiYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Bkt2V1vEMnqWGnyTPMNtwYEVLx5CkjHTOR7Q0nYSR0U)
> 
> [esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051033/esp32solo.zip) [esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051034/esp32.zip) [esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051035/esp8266.zip) [esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051036/esp32c3.zip) [esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14051038/esp32s2.zip)

Trying the new firmware cfa4502 (esp32s2) on top of 9ff6cf3 and get the following issues:
1. Price-config page stock on Loading
2. Month display on start page says "28. Undefined 17:03"
3. bottom of Amp and Phase displays a lot of overlying text
[image]

ClausMelander, 2024-01-28:
> Trying the new firmware cfa4502 (esp32s2) on top of [9ff6cf3](https://github.com/UtilitechAS/amsreader-firmware/commit/9ff6cf355fc59e440039778f50bd36f1701229a0) and get the following issues:
>
> 1. [...]

Ok... unit was "reset" when reverting to firmware 2.2.28 (because of the issues) so having to setup AP again. After this I updated with old config backup and everything was back to "normal"
Then trying again the new cfa4502 (esp32s2) and now it looks much better :)
Still a few issues though:
1. bottom of Amp and Phase still displays a lot of overlying text.
2. export price are not shown on export meter (only import price on import meter).
3. I had also the issue with prices not saved, but it's only the first time for some reason, thereafter it's saved :)
4. After each setup save, it's like the unit reloads the home page 2 times... minor bug though.
5. Not really a bug I think, but had to figure out a new value for Meter Buffer size (512kB) as the 128kB didn't work.
6. Could use one more digit on the export price calculation 0,00375 but 0,0038 is so close I almost forgot to mention ;)

[image]
[image]

Anyway, great work with the calculations! Now the Import price is exactly following the price shown in the power company app :)
Not sure about the export price yet, but the price settings page looks good :)

haldi4803, 2024-01-30:
i guess switching from d497aa9 to cfa4502 does cause some issues. because i'm stuck at a "loading" window in the configuration.
So Factory Reset and Restoring Data helps.

But still it wont save Prices.

Here is a quick video.
https://github.com/UtilitechAS/amsreader-firmware/assets/17104473/0a69e0e5-0c6b-4f3c-8b8c-bf592b4bdbeb

P.S Months are undefined?

gskjold, 2024-02-15:
I found the problem, fixed it in this version:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292957/esp32s2.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292958/esp8266.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292959/esp32solo.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292960/esp32c3.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292961/esp32.zip)

haldi4803, 2024-02-15:
I can confirm adding Prices works.

haldi4803, 2024-02-15:
Price is Calucalted ON Usage and not retrospectively?
[image]

Or what is that graph supposed to Display? Price (HF/100) How much you pay /get per kWh in this certain Hour if you have Solar on the roof?

[image]
P.S top right, 15. Undefined 20:34 ^^ still same issues with the months right?

haldi4803, 2024-02-15:
oh btw @gskjold your Power Values are okay?
Because if you take a look at my Screenshot you can see  the 1.3kW Actual usage (that's correct). If that goes on for 1hour that would be 1.3kW used right? But in the 24h graph there is a 9.3kW (that seems to be wrong).
Multipliers are not activated in settings... so that should be default?
[image]

If you check the 31 days graph you can see that this issue has persisted since 22 january when i first flashed the Test Build.

[image]

these are the official values from my provider.
[image]

gskjold, 2024-02-16:
You most likely need multiplier on kwh set to 0.1 (not uncommon on Kamstrup)

Price graph shows price you pay per imported kwh in the upcoming hours.

Regarding undefined month, go to config and choose a language. Device then needs internet access to download the language file.

haldi4803, 2024-02-19:
> You most likely need multiplier on kwh set to 0.1 (not uncommon on Kamstrup)

Like that?
[image]

Doesn't seem to work.
[image]
I've downloaded the "meter" file, edited it, and reuploaded it again. So past data is now correct. but it doesn't seem to apply the 0.1x correction.

ArnieO, 2024-02-19:
> I've downloaded the "meter" file, edited it, and reuploaded it again. So past data is now correct. but it doesn't seem to apply the 0.1x correction.

Strange. @gskjold is on vacation this week, he'll have to look into this when he is back. 
Does it behave as expected if you revert back to v2.2.28? (make a new config file first, to save your historic values).

haldi4803, 2024-02-19:
Okay... it's strange... setting it to  0.011
[image]
seems to work.
[image]

dayplot 5 1708358402 48075 0 0 22 19 21 20 21 24 22 22 21 20 20 20 20 21 24 5750 9360 13660 13650 13200 13540 6770 5320

DO you need a reboot after setting/changing the Value? Because it doesn't automatically.

ArnieO, 2024-02-20:
> Okay... it's strange... setting it to 0.011 seems to work.

Setting it to 0.011 works, but not 0.01?

haldi4803, 2024-02-20:
I've set it to 0.1 and rebooted, it seems to work now!
Does it need a reboot to actually accept the value?

ArnieO, 2024-02-20:
> I've set it to 0.1 and rebooted, it seems to work now! Does it need a reboot to actually accept the value?

Good question - and if needed, the firmware should do it. 
@gskjold will have to look into this before we release v2.3.

Thank you for contributing to the testing! 👍

closed 2024-03-21 by gskjold