UtilitechAS/amsreader-firmware #253: SAGEMCOM T211

christianritella, 2022-03-24:
Any support for Swedish SAGEMCOM T211?
https://www.ellevio.se/globalassets/content/el/elmatare-produktblad-b2c/ellevio_produktblad_fas3_t211_web2.pdf

gskjold, 2022-03-24:
It sort of depends on the data format, but most likely. Sagemcom have been tested successfully in other countries. Best way to be certain is to find a technical document explaining the data. Or another Swede user who have already tested it.

christianritella, 2022-03-24:
Hi, it is RJ12.
I have installed the latest software on a esp8266.
Cabling have been done like this;
VIN = Black + Blue
G = Brown + Red
RX = Yellow

christianritella, 2022-03-24:
Yes, a dev board.
Like this photo but without the capacitors.
Do I have to get a non dev board?
[image]

ArnieO, 2022-03-24:
No, it should work with a dev board if you just connect and configure correctly.
It would be better if you refer to pin numbers rather than colours:
[image]

Most important are:
- Pin 2 must be connected to pin 1 to receive data.
- The data output line from the meter is open collector - so you need to add a pullup  resistor. 

And remember to set data rate to 9600 baud, and you might need to invert the data stream - and try both parities. 
This is where you do those changes:
[image]

ArnieO, 2022-03-24:
Sorry; that should be 115 200 baud!

gskjold, 2022-03-24:
You can still use this board, but you cannot connect data to RX pin like you have done. Connect it to GPIO 13 (D7) instead an go to System -> GPIO and change HAN pin to UART2 (GPIO13). This is the setup most people with NodeMCU have had most success with as I understand it.

closed 2022-03-30 by gskjold

christianritella, 2022-04-20:
I have connected RX to D7 and HAN=UART2/GPIO13.
Baudrate 115200 Parity 8N1
Have tried invert on/off;
Invert off: (W) (readHanPort) Unknown data format 40
invert on: (W) (readHanPort) Unknown data format 3F
Does this indicate that it is receiving data?

I have no resistors connected.

Any other suggestions?

ArnieO, 2022-04-21:
> Any other suggestions?

Please try a Tenet debug:
System/Debugging: Select Telnet and Verbose mode
Open a command window on your PC, and write "telnet <IP-address of device>"
You may need to activate telnet, as it is by default disabled on Windows (how to do that: Google). 
Then paste here what you get.