UtilitechAS/amsreader-firmware #254: Add support for current transformer

jmossige, 2022-03-25:
Use of han2mqtt with current transformers results in incorrect power reading - voltage is correct, but all power values ​​are too low as the current used in AMS is low due to the use of an external current transformer.

Possibility to use a factor of values ​​related to current.
In my installation I have a current transformer 200/5 and have to multiply active and reactive power (import / export) and the current by a factor of 40 (200/5) to get real data.

External current transformer is most often used in intakes above 63A - as the meters are not designed for higher currents.

tronde-ams, 2022-03-25:
Have you tried Configuration -> meter -> multipliers?

jmossige, 2022-03-25:
> Have you tried Configuration -> meter -> multipliers?

Not an option i have found, i'm on ams2mqtt v2.0.11.  kamstrup 400V/TN ...

jmossige, 2022-03-25:
ok, found something in the beta. i'll try that. thanks!

closed 2022-03-25 by jmossige