UtilitechAS/amsreader-firmware #257: Entso-E configuration gets prices from 24h in the future

thomaspalmdk, 2022-03-30:
**Describe the bug**
Using the Entso-E API I'm expecting to see the prices of electricity from this hour, not same hour 24h ahead of time.
Reading the mq-queue from Amsreader I get the following 30/3-2022 17:59-ish, time range 17-18:
[{"id":"yy:yy:yy:yy:yy:yy","prices":{"0":0.1880,"1":0.2000,"2":0.2295,.....}}]()
When I look in the transparency platform on the Entso-E homepage, these prices are the ones for 31/3-2022 time range 17-18.
Is it possible to fetch data from today, so I can see live updated prices?

gskjold, 2022-03-30:
What you are describing is not the intended behaviour, you should get the price for current hour as "0" in your json

gskjold, 2022-03-30:
Looking into it now, it looks like a timing issue in the code.

thomaspalmdk, 2022-03-30:
I just did some manual get-requests using the Entso-E API, and I do see todays prices in the result, when providing a periodStart and periodEnd with todays date.
It seems the request from AMSReader might be incorrect. Is it possible to debug/see this request?

gskjold, 2022-03-30:
Enable debugger and set to verbose mode

gskjold, 2022-03-30:
Found the problem, it is using the wrong hour now, after switching to CEST

thomaspalmdk, 2022-03-30:
Great work :-)

gskjold, 2022-03-30:
v2.1.0 is coming out tonight, fix included there

closed 2022-03-30 by gskjold